srsran/srsRAN_4G #543: eNB are not Visible on COTS UE

Mopoma95, 2020-09-04:
Dear All,

I want to use SRSLTE for RAN and Open5gs for Core
Both are cnfigured and eNB is connected to SRSeNB 

But i can't see the network on COTS UE!
I used UHD 3.15 LTS and the master SRSLTE

Any help would be appritiated
I also changed the Gain and earfcn to 1600,3100,3400 but notting changed! still cant see!
for gain i tested the defauld 80,40 and 110, 90 and 60,70 but still no chance

Thanks
 Best regards

Mopoma95, 2020-09-04:
Also,
All make test items has passed successfully

Mopoma95, 2020-09-04:
Even tried it with SRSEPC but still the network is not visible!

Any help?

andrepuschmann, 2020-09-04:
Have you checked https://docs.srslte.com/en/latest/app_notes/source/cots_ue/source/index.html

Mopoma95, 2020-09-04:
Dear @andrepuschmann ,

Yes, I did check it.

The note is few days back i tested with the same configuration and everything was Ok, I could see the network!
But Now even with the same UE the network is not visible!

One other note is that i ALSO tried with UHD-3.15.LTS and SRSLTE tag Version 19__12 while make test i faced with NAS child abort and just one of 715 test failed because of that!

Then Again I tested with UHD-3.15.LTS and SRSLTE master branch there was no error on make test but i can't see the Network on Samsung UE!

Do u have any idea what would be problem?

P.S the UE support band 7 and EPC is connected to eNB!

Thanks
Best regards

andrepuschmann, 2020-09-14:
Can you tell which phone you use?

Mopoma95, 2020-09-14:
@andrepuschmann 
Thanks for your replay

Sure. I used different UEs. including: Samsung galaxy Tab S2, iphone 6s, Samsung A20 and samsung j4!
But in all cases the above mentioned problem is the same!

I used A PC with bellow configuration:
Core i7 3770 3400 MHz, 8GB of RAM, with Ubuntu 18.04 linux 5.4.0.47-generic#51-18.04-1 64bit!

I also tried it in my Laptop with the exact config but no chance! my laptop has almost the same config:
Core i7 4700 2.4GHz up to 3.4Ghz, 8GB RAM with the exact ubuntu version!

I will try in next 24 hours with these configs:
Core i7 6700 3.4Ghz, 16GB of RAM, Linux 5.4.0-42-generic #46~18.04.1-Ubuntu SMP Fri Jul 10 07:21:24 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

andrepuschmann, 2020-09-14:
You are not sharing your logs and configs. It might be related to the bandwidth and EARFCN you are using. Try EARFCN `3350` and use a cell bandwidth of 50 PRB.

Mopoma95, 2020-09-15:
@andrepuschmann 
I mentioned it in this issue [#544](https://github.com/srsLTE/srsLTE/issues/544)

andrepuschmann, 2020-09-15:
If issue #544 is the same, then please close one ticket. Also you are not sharing any logs or PCAPs, just stdout output. Please enable log level info in the eNB and share the logs for an successful run and one which isn't.

andrepuschmann, 2020-10-27:
Closing issue due to missing feedback from author.

closed 2020-10-27 by andrepuschmann