srsran/srsRAN_4G #544: Master clock rate and sample rate

Mopoma95, 2020-09-07:
Hi everyone,

I'm using SRSLTE for Testing 4G network.

I tested UHD driver with these:  master/UHD-3.15.LTS/UHD-3.9-LTS  and Also tested SRSLTE with these : Master/20-04/19-12 branches or Tags but I could'nt see the Network on COTS UE, I'm using USRP B210 and it is healthy!( tested it with OAI ) but in anycase I could'nt managed to see the Network! (9 different tests)

So I tryed to change the freq and gain but still no chance!

Now I want to change the Master clock rate, Clock rate and Samplelling Rate! But when i put master-clock-rate from 23.04e6 to 30.72e6 , still have the same Master clock rate in 

> [INFO] [B200] Detected Device: B210
> [INFO] [B200] Operating over USB 3.
> [INFO] [B200] Initialize CODEC control...
> [INFO] [B200] Initialize Radio control...
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Performing register loopback test... 
> [INFO] [B200] Register loopback test passed
> [INFO] [B200] Asking for clock rate 23.040000 MHz... 
> [INFO] [B200] Actually got clock rate 23.040000 MHz.
> Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
> Setting Sampling frequency 11.52 MHz
> 

part, the master clock rate that i entered is visible on the first part but doesn't changed in the main part! 

> Opening USRP with args: type=b200,master_clock_rate=23.04e6, master_clock_rate=30.72e6 
I also then changed the  lib/src/phy/rf/rf_uhd_imp.c for all USRP B200 and 210 to 30.72e6 but notting changed!

So Now how can i change these parameters?
@andrepuschmann @djanderson @frankist 

Note: I could see the Network in SRSLTE release_19_12 with UHD Master branch but while make test there was 8 errors which all related to phy_dl_test.

Thanks 
Best regards

Mopoma95, 2020-09-07:
Also EPC connected to eNB in all cases!

andrepuschmann, 2020-09-07:
With 19.12 you can see the network with the exact same configs? Can you please double check and send logs for each run if that is the case?

The UE not seeing the network could have many causes. For example, you don't use a GPSDO. Your USRP could be waaay off. Could you try with EARFCN 3350? Some phones don't like the outer edges of the band.

Mopoma95, 2020-09-07:
@andrepuschmann 
Hi,
Yea with 19-12 i can see the network with the exact config files

That was too much works! but i believe that all the logs are the same!!! 
Right Now i have SRSLTE 19-12 with UHD Master branch in Which the Network is Visible

The logs are attached

```

m@n:~/srsLTE/build$ sudo srsenb

Built in Release mode using commit d045213fb on branch HEAD.

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/m/.config/srslte/enb.conf...
Opening 1 RF devices with 1 RF channels...
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-54-gbc585b43
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP with args: type=b200,master_clock_rate=23.04e6
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz
Setting Sampling frequency 11.52 MHz
```

And this is EPC log

```

m@n:~/srsLTE/build$ sudo srsepc

Built in Release mode using commit d045213fb on branch HEAD.

---  Software Radio Systems EPC  ---

Reading configuration file /home/m/.config/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
SCTP Association Shutdown. Association: 6
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
No UEs to be released
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
^CDeleting eNB context. eNB Id: 0x19b
Deleting UE context in HSS. IMSI: 001010123456780
Deleting UE context in HSS. IMSI: 001010123456789
```

These are eNB and EPC logs
Note: I do not try to connect because i'm testing other things right Now!

As u said i do not have GPSDO, But I try to stop and then run again the eNB to undrestand that i dont have GPSDO as you can see there is NO GPSDO log in eNB log file!

I would try again with SRS MASTER branch and Master UHD version and put the log as soon as i can ( maybe in 3 hours)!
and also change earfcn too.

Best regards
Thanks for your quick replay

Mopoma95, 2020-09-07:
I got Something else!

Whenever i get some make test items failed, i can see the Network on UE !!!!
and whenever i get make test all passed! i can't see the network!!!!!!!!!

I get this from testing the same Configuration of SRS 19-12 with UHD Master branch on Laptop!
Both PC are core i7 , 8GB ram, One is PC and the other is LAPTOP! the success in seeing the network was occurred on PC! Both has fresh Ubuntu from the same bootable flash drive with this kernel             
Linux n 5.4.0-45-generic #49~18.04.2-Ubuntu SMP Wed Aug 26 16:29:02 UTC 2020 x86_64 x86_64 x86_64 GNU/LinuxLinux n 5.4.0-45-generic #49~18.04.2-Ubuntu SMP Wed Aug 26 16:29:02 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

Thanks
Best regards

andrepuschmann, 2020-10-27:
No sure what the state is here but I would suggest to try again with 20.10 and reopen if you still encounter issues. Thanks

closed 2020-10-27 by andrepuschmann