srsran/srsRAN_4G #550: socket problem when run on Jetson TX2

jin-github1002, 2020-09-20:
Hi team
I run srsLTE on Jetson TX2. compile and make test are all right. When running srsenb, I got socket problems as follows. Does anyone give any recommendation ? (hareware is LimeSDR USB, driver is SoapySDR. I install srsLTE from the repository)

nvidia@nvidia:~/srsLTE/srsenb$ sudo srsenb enb.conf
[sudo] password for nvidia: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit f02bfe2cf on branch master.

/home/nvidia/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1934

Opening 1 channels in RF device=soapy with args=default
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Soapy has found device #0: default_input=False, default_output=False, device_id=2, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,0, 
Soapy has found device #1: default_input=False, default_output=False, device_id=3, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,1, 
Soapy has found device #2: default_input=False, default_output=False, device_id=4, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,2, 
Soapy has found device #3: default_input=False, default_output=False, device_id=5, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,3, 
Soapy has found device #4: default_input=False, default_output=False, device_id=6, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,4, 
Soapy has found device #5: default_input=False, default_output=False, device_id=7, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,5, 
Soapy has found device #6: default_input=False, default_output=False, device_id=8, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,6, 
Soapy has found device #7: default_input=False, default_output=False, device_id=9, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,7, 
Soapy has found device #8: default_input=False, default_output=False, device_id=10, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,8, 
Soapy has found device #9: default_input=False, default_output=False, device_id=11, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,9, 
Soapy has found device #10: default_input=False, default_output=False, device_id=12, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,10, 
Soapy has found device #11: default_input=False, default_output=False, device_id=13, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,11, 
Soapy has found device #12: default_input=False, default_output=False, device_id=14, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,12, 
Soapy has found device #13: default_input=False, default_output=False, device_id=15, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,13, 
Soapy has found device #14: default_input=False, default_output=False, device_id=16, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,14, 
Soapy has found device #15: default_input=False, default_output=False, device_id=17, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,15, 
Soapy has found device #16: default_input=False, default_output=False, device_id=18, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,16, 
Soapy has found device #17: default_input=False, default_output=False, device_id=19, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,17, 
Soapy has found device #18: default_input=False, default_output=False, device_id=20, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,18, 
Soapy has found device #19: default_input=False, default_output=False, device_id=21, driver=audio, label=hw:tegra-snd-t186ref-mobile-rt565x,19, 
Soapy has found device #20: default_input=False, default_output=False, device_id=22, driver=audio, label=default, 
Soapy has found device #21: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 2.0] 9081C05C42731, media=USB 2.0, module=STREAM, name=LimeSDR-USB, serial=0009081C05C42731, 
Selecting Soapy device: 0
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---

andrepuschmann, 2020-09-21:
Soapy is picking up many audio devices before it actually find the Lime. I suggest you remove the soapy-audio driver `sudo apt-get remove soapysdr-module-audio` and make sure you only have the Lime detected.

andrepuschmann, 2020-10-27:
No response from author. Closing issue.

closed 2020-10-27 by andrepuschmann