srsran/srsRAN_4G #551: I can't connect the srslte app with the phone in some form? PLease help me ?

muratoznn, 2020-09-21:
I want to connect my phone to my srslte application but İt dont connect in no way ? PLease Help me ?   First I installed srslte application to root ubuntu with no problem. I set mnc mcc codes according to my phone sim card. I opened and it started perfect but my phone dont connect to srslte . Why ? How can I connect my phone with imsi code to srslte application automatically ?  I restarted  and restarted my phone again but srslte dont connect to my phone ?  My phone imsi 286039540347361 MNC:286 MCC:03 . I want to connect only my phone to my srslte automatically ? How I can do this ? All application information below
(I want to this : My phone connect to my srslte application automatically and I can use srslte ) 

[enb.conf.txt](https://github.com/srsLTE/srsLTE/files/5258423/enb.conf.txt)
[epc.conf.txt](https://github.com/srsLTE/srsLTE/files/5258424/epc.conf.txt)
[image]
[image]
[image]
[image]

muratoznn, 2020-09-21:
[enb.conf.txt](https://github.com/srsLTE/srsLTE/files/5258449/enb.conf.txt)
[epc.conf.txt](https://github.com/srsLTE/srsLTE/files/5258450/epc.conf.txt)

andrepuschmann, 2020-09-22:
hey, it seems you are using the Lime through Soapy's UHDSoapyDevice. That's not recommended. Better use the Lime through our Soapy driver directly. For that, remove the module for Soapy, i.e. `sudo apt-get remove uhd-soapysdr`

muratoznn, 2020-09-23:
Okay I tried it now . I deleted complete ubuntu and install again . I İnstalled only soapy lime device now. Thank God he finally reacted for the first time. But again it was not connected to my cell phone. İt is giving these errors like Disconnecting rnti=0x4c and RACH:  tti=9261, preamble=26, offset=12, temp_crnti=0x4e     How can I fix these errors and connect my phone to srslte ?
All the mistakes below  . When my phone try to connect it is giving to me these errors ? 

[WARNING] L
LRACH:  tti=861, preamble=7, offset=12, temp_crnti=0x46
RACH:  tti=881, preamble=18, offset=12, temp_crnti=0x47
[WARNING] L
RACH:  tti=901, preamble=28, offset=12, temp_crnti=0x48
RACH:  tti=921, preamble=15, offset=12, temp_crnti=0x49
RACH:  tti=941, preamble=9, offset=12, temp_crnti=0x4a
Disconnecting rnti=0x46.
[WARNING] L
Disconnecting rnti=0x47.
Disconnecting rnti=0x48.
Disconnecting rnti=0x49.
Disconnecting rnti=0x4a.
[WARNING] L
RACH:  tti=2961, preamble=33, offset=12, temp_crnti=0x4b
[WARNING] L
LDisconnecting rnti=0x4b.
RACH:  tti=5041, preamble=33, offset=12, temp_crnti=0x4c
[WARNING] L
[WARNING] L
Disconnecting rnti=0x4c.
RACH:  tti=7161, preamble=26, offset=12, temp_crnti=0x4d
[WARNING] L
Disconnecting rnti=0x4d.
RACH:  tti=9261, preamble=26, offset=12, temp_crnti=0x4e
[WARNING] L
[WARNING] L
Disconnecting rnti=0x4e.
[WARNING] L
[WARNING] L
[WARNING] L
[WARNING] L

muratoznn, 2020-09-24:
I am trying again it is giving to me this error and stopping my phone lte connection . How can I correct this problem and connect to my phone with srslte ?

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit 0ec49a7b1 on branch master.

/root/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 1450

Opening 1 channels in RF device=soapy with args=rxant=LNAH,txant=BAND2
[ERROR] SoapySDR::loadModule(/usr/local/lib/SoapySDR/modules0.6/libLMS7Support.so)
  duplicate entry for lime (/usr/lib/x86_64-linux-gnu/SoapySDR/modules0.6/libLMS7Support.so)
[INFO] [UHD] linux; GNU C++ version 8.3.0; Boost_106700; UHD_3.13.1.0-3build1
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 3.0] 1D4C3893CE8AAC, media=USB 3.0, module=FT601, name=LimeSDR Mini, serial=1D4C3893CE8AAC, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C3893CE8AAC'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M calibration values caching Disable
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
Setting Rx antenna to LNAH
Setting Tx antenna to BAND2
[ERROR] Tx Calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
[ERROR] Rx calibration: MCU error 5 (Loopback signal weak: not connected/insufficient gain?)
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND2

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1830.0 Mhz, UL=1735.0 MHz for cc_idx=0
[WARNING] L
[WARNING] L
[WARNING] L
LRACH:  tti=6421, preamble=11, offset=12, temp_crnti=0x46
[WARNING] L
RACH:  tti=6441, preamble=39, offset=12, temp_crnti=0x47
[WARNING] L
Disconnecting rnti=0x46.
Disconnecting rnti=0x47.
[WARNING] L
RACH:  tti=8341, preamble=18, offset=12, temp_crnti=0x48
[WARNING] L
Disconnecting rnti=0x48.
[WARNING] L
RACH:  tti=21, preamble=47, offset=12, temp_crnti=0x49
[WARNING] L
LLDisconnecting rnti=0x49.
[WARNING] L
RACH:  tti=1941, preamble=44, offset=12, temp_crnti=0x4a

muratoznn, 2020-09-25:
PLease Help ? How can I connect to my phone srsLTE . İt is only stopping mobile phones connections but not connect

muratoznn, 2020-10-02:
I fixed the limesdr driver problem myself. Install Fedora linux. Then install the limesdr drivers. Then you steal the driver application packages inside the apt / archieves folder. Then install these application packages on ubuntu and the device is running smoothly. I fixed it .

closed 2020-10-02 by muratoznn

andrepuschmann, 2020-10-02:
Hey, glad you could solve the issue. I am not fully understanding the approach but it seems that you installed another version of the driver. If so, could you let us know which git hash this driver had? In theory we should be able to just check out and install this particular version without the Fedora detour.

muratoznn, 2020-10-05:
This is the logic here. Application driver software for an operating system does not work in different Linux distributions. Applications used in Linux are free and open source. Limesdr driver software works very healthy in fedora operating system. But the ubuntu limesdr driver software is causing problems. Kali does not work at all on the linux operating system. In the Linux mint operating system, gnu radio causes problems. I installed fedora on a usb disk. Then I installed the limesdr driver software. Later, I completely got the deb packages that the operating system loaded into its repository. Then I installed ubuntu 20.04 on linux operating system with sudo dpkg command. Later when I was installing srslte it thinks soapy software is installed. Then when I typed the srsepc srsenb commands, I was very happy. Because the limesdr driver software was working without any problems. But epc gives an error. My mobile phone is disconnected from lte. How can I run epc. Can you help me? I opened new issue for this