srsran/srsRAN_4G #552: srsEnb bladeRF Tx TX failed: Operation timed out

pat-ford, 2020-09-24:
## Issue Description ##
Built here and also tried apt-get

when I run srsenb ( :~$ sudo srsenb ) I get the following output: 
`pat@worker:~$ sudo srsenb 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit f02bfe2cf on branch master.

Opening 1 channels in RF device=bladeRF with args=default
Opening bladeRF...
Set RX sampling rate 1.92 Mhz, filter BW: 1.92 Mhz

==== eNodeB started ===
Type <t> to view trace
Starting plot for worker_id=0
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Set RX sampling rate 11.52 Mhz, filter BW: 9.22 Mhz
Setting manual TX/RX offset to 27 samples
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
set TX frequency to 2684999998
set RX frequency to 2564999998
/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

^CStopping ..
^C^C/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

/home/pat/tools/srsLTE/lib/src/phy/rf/rf_blade_imp.c.519: TX failed: Operation timed out

Couldn't stop after 5s. Forcing exit.
Killed
`
I get the same with both prebuilt and built here versions.  If I leave it running it takes a while but it will dump. Happens with both stock config file and altered.
Backtrace 
`--- command='srsenb' version=20.04.2 signal=11 date='24/09/2020 17:14:44' ---
	srsenb(+0x11c392) [0x55a850fe3392]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f5a03f61210]
	srsenb(+0x85067) [0x55a850f4c067]
	srsenb(+0x129a59) [0x55a850ff0a59]
	srsenb(+0x67293) [0x55a850f2e293]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f5a04836609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f5a0403d293]

--- command='srsenb' version=20.04.2 signal=11 date='24/09/2020 17:14:44' ---
	srsenb(+0x11c392) [0x55a850fe3392]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f5a03f61210]
	srsenb(+0x85067) [0x55a850f4c067]
	srsenb(+0x129a59) [0x55a850ff0a59]
	srsenb(+0x67293) [0x55a850f2e293]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f5a04836609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f5a0403d293]

--- command='srsenb' version=20.04.2 signal=11 date='24/09/2020 17:14:44' ---
	srsenb(+0x11c392) [0x55a850fe3392]
[... 323 lines cut ...]
  FPGA loaded:              yes
  Flash size:               32 Mbit
  USB bus:                  3
  USB address:              10
  USB speed:                Hi-Speed
  Backend:                  libusb
  Instance:                 0

bladeRF> --help
`

Bladerf A4 w latest fw
Occurs with both ppa and local builds from most recent git

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2020-09-25:
This is a duplicate of #504 I believe. Ball on this is with Nuand. Closing therefore.

closed 2020-09-25 by andrepuschmann

pat-ford, 2020-09-25:
The error is coming from a different area of code than #504, this is a tx timeout not rx.