srsran/srsRAN_4G #555: Help someone on behalf of humanity please? I fixed the Limesdr Driver problem myself but now I have a Srsepc and sim card connection problem

muratoznn, 2020-10-02:
Help someone on behalf of humanity please? I fixed the Limesdr Driver problem myself but now I have a Srsepc and sim card connection problem

My user_db.cvs file is this 
I am taking this errors 
How Can I connect my phone to srslte ?

---  Software Radio Systems EPC  ---

Reading configuration file /root/.config/srslte/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf286, MNC: 0xff03
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:286, MNC:03, PLMN: 8582704
S1 Setup Request - TAC 0, B-PLMN 0
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0xdfae4dd1
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 11110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 286039540347361
User not found at HSS. IMSI: 286039540347361
User not found. IMSI 286039540347361
Received UE Context Release Request. MME-UE S1AP Id 1
UE is not ECM connected. No need to release S1-U. MME UE S1AP Id 1
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0xdfae4dd1
Attach request -- eNB-UE S1AP Id: 2
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 11110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 2
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 286039540347361
User not found at HSS. IMSI: 286039540347361
User not found. IMSI 286039540347361
Received UE Context Release Request. MME-UE S1AP Id 2
UE is not ECM connected. No need to release S1-U. MME UE S1AP Id 2
SCTP Association Shutdown. Association: 8
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
Releasing UE ECM context. UE-MME S1AP Id: 1
Releasing UE ECM context. UE-MME S1AP Id: 2
^CStopping ..

---  exiting  ---

user_db.cvs file 

#                                                                                           
# .csv to store UE's information in HSS                                                     
# Kept in the following format: "Name,Auth,IMSI,Key,OP_Type,OP,AMF,SQN,QCI,IP_alloc"      
#                                                                                           
# Name:     Human readable name to help distinguish UE's. Ignored by the HSS                
# IMSI:     UE's IMSI value                                                                 
# Auth:     Authentication algorithm used by the UE. Valid algorithms are XOR               
#           (xor) and MILENAGE (mil)                                                        
# Key:      UE's key, where other keys are derived from. Stored in hexadecimal              
# OP_Type:  Operator's code type, either OP or OPc                                          
# OP/OPc:   Operator Code/Cyphered Operator Code, stored in hexadecimal                     
# AMF:      Authentication management field, stored in hexadecimal                          
# SQN:      UE's Sequence number for freshness of the authentication                        
# QCI:      QoS Class Identifier for the UE's default bearer.                               
# IP_alloc: IP allocation stratagy for the SPGW.                                            
#           With 'dynamic' the SPGW will automatically allocate IPs                         
#           With a valid IPv4 (e.g. '172.16.0.2') the UE will have a statically assigned IP.
#                                                                                           
# Note: Lines starting by '#' are ignored and will be overwritten

Jackrabrett, 2020-10-15:
Is 286039540347361 your phones IMSI?

alvasMan, 2020-10-15:
The EPC cannot find the UE. See this line:
```
UL NAS: Received Identity Response
ID Response -- IMSI: 286039540347361
User not found at HSS. IMSI: 286039540347361
User not found. IMSI 286039540347361
```

I cannot see the actual `user_db.csv` due to the poor formatting, but my guess is that the IMSI does not exist there.
Make sure that you have that IMSI configured in the user_db.csv. You can see how to configure this on this link: 
https://docs.srslte.com/en/latest/srsepc/source/2_epc_getstarted.html#configuration

andrepuschmann, 2020-10-27:
I believe the issue is solved. Closing ticket.

closed 2020-10-27 by andrepuschmann

fatafalihhilmi, 2022-09-01:
@alvasMan  the link can't open