srsran/srsRAN_4G #556: Hacktoberfest 2020

djanderson, 2020-10-03:
I don't get as much time to hack on open source projects as I'd like, but I'd like to make an effort this month for Hacktoberfest. I'd love to hack on srsLTE, either bug squashing or adding new features. It'd be awesome if project maintainers could label some existing issues and/or add some non-trivial enhancement issues and label them. Maybe tag this issue so they get indexed here?

My background is in core network architecture, LTE protocols, and SDRs, so don't hold back :)

cloudguru-bb, 2020-10-03:
Douglas,

This is a really good idea.  Things are starting to move really quickly
with Enterprise adoption of 5G becoming real and OpenRAN declared as
mandatory by some MNOs globally.

Feel free to PM me if you want to collaborate on this.

Brian

djanderson, 2020-10-03:
It's unclear to me whether they intend to extend this project with 5G support in the future. The Rel14 "stepping-stone" type changes like CUPS S/P-GW might be useful future-proofing, but doesn't really add any user-visible enhancements.

Definitely not a Hacktoberfest-level for one person but if it was split up it could be doable.

andrepuschmann, 2020-10-05:
Hey guys,
thanks a ton for bringing this up and showing interest in working on srsLTE. We've been observing Hacktoberfest for a while but never activity participated so far. That being said, we'd love to have people working on the project and, of course, encourage the community to submit PR, both for code and for documentation. For the latter, please also have a look at the docs repo https://github.com/srsLTE/srsLTE_docs
Right now there are quite a few open issues from the community but none that we created and flagged explicitly as suitable for such hackathons. Please give us some time and we will come up with a list of issues that qualify for this sort of effort. In the meanwhile, if you have some specific topic you'd like to work on please don't hesitate to reach out if you want to clarify anything before getting your hands dirty.

Thanks again

andrepuschmann, 2020-10-08:
Hi again,
we've created a new Wiki page [here](https://github.com/srsLTE/srsLTE/wiki/Possible-contributions) that lists possible contributions areas of contributions that we think also qualify for Hacktoberfest. Please have a look and let us know what you think.

Thanks

andrepuschmann, 2020-10-18:
Hey, quick information to those who are planning to contribute (or have already contributed) to srsLTE as part of the Hacktoberfest challenge. Our current development process requires us to cherry-pick public PRs onto our internal development branch in order to run our CI on it. If those runs are ok, we merge the commit manually and close the public PR. Of course, authorship and copyright of the commit remain with the author. I just wanted to flag that here. I hope this procedure doesn't unqualify for the Hacktoberfest challenge.
Thanks

djanderson, 2020-10-19:
Hi Andre, thanks for bringing this up.

According to the rules:

> PRs count if:
Submitted during the month of October AND
Submitted in a public repo AND (
  The PR is labelled as hacktoberfest-accepted by a maintainer OR
  Submitted in a repo with the hacktoberfest topic AND (
    The PR is merged OR
    The PR has been approved
  )
) 

To be honest I don't know what "accepting" a PR is, but it seems like for srsLTE, it might work to do that and add the label `hacktoberfest-accepted`?

Thanks for your work in setting this up and I look forward to digging in more now that I've got the project setup for development 🙂

andrepuschmann, 2020-10-19:
Excellent. It sounds like this should do the trick, yes. I've added the tag to two recent PRs. Let me know if there are any questions or concerns raised at the end of the month from the organizers side.

djanderson, 2020-11-02:
Looks like the label worked fine. Thanks again for setting this up!

closed 2020-11-02 by djanderson

andrepuschmann, 2020-11-02:
Good to hear that. Thanks again for your contributions