srsran/srsRAN_4G #562: LimeSDR USB  + srsENB fails to start

grax004, 2020-10-12:
## Issue Description ##
I've read numerous guides, attempted to install on multiple versions and releases of all required components yet nothing seems to allow me to get the srsLTE enb to work with the LimeSDR USB. 

Every time I attempt to spin up the eNodeB, I get the same issues surrounding the calibration of the LimeSDR USB. 

I have included the errors I see and versions required to reproduce. 

## Setup Details ##
Ubuntu 20.04.1 LTS = ( `Linux version 5.4.0-48-generic (buildd@lcy01-amd64-010) (gcc version 9.3.0 (Ubuntu 9.3.0-10ubuntu2)) #52-Ubuntu SMP Thu Sep 10 10:58:49 UTC 2020` )
LimeSuite = ( `v20.07.2` )
SoapySDR = ( `soapy-sdr-0.7.2` ) 
srsGUI = ( `release_2_0_qt5` )
srsLTE = ( `release_20_04_2` ) 

The above versions are my most recent attempt however, I've attempted with a number of different combinations with no successful `srsenb` ever spinning up and allowing a UE to connect. 

## Expected Behavior ##
When the startup command is executed:
```
sudo srsenb
```
I expect the `srsenb` to configure itself and begin RX / TX successfully with the LimeSDR USB. 

## Actual Behaviour ##
After executing the startup command, the following output is provided by `srsenb`: 
```
root@cellular-VirtualBox:~/.config/srslte# sudo srsenb 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srslte/enb.conf...

Built in Release mode using commit 0ec49a7b1 on branch heads/release_20_04_2.

Opening 1 channels in RF device=default with args=default
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 2.0] 9072C02881916, media=USB 2.0, module=FX3, name=LimeSDR-USB, serial=0009072C02881916, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR-USB [USB 2.0] 9072C02881916'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAL
Tx antenna set to BAND1
connect(): Connection refused
Failed to initiate SCTP socket. Attempting reconnection in 10 seconds

==== eNodeB started ===
Type <t> to view trace
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=0)
[ERROR] SetFrequencyCGEN(-nan MHz) failed
[ERROR] Tune Rx Filter: MCU error 2 (CGEN tune failed)
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=0)
[ERROR] SetFrequencyCGEN(92.16 MHz) failed
[ERROR] Tune Tx Filter: MCU error 2 (CGEN tune failed)
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=0)
[ERROR] SetFrequencyCGEN(61.44 MHz) failed
[ERROR] setSampleRate(Tx, 0, 11.52 MHz) Failed
setSampleRate Tx fail: SoapyLMS7::setSampleRate() failed

Warning TX/RX time offset for sampling rate -1 KHz not calibrated. Using interpolated value

Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0
/home/cellular/CellularSetup/srsLTE/lib/src/phy/rf/rf_soapy_imp.c.983: Couldn't write all samples after 100 trials.
```

## Steps to reproduce the problem ##
Compile the versions provided above, connect LimeSDR USB, execute `sudo srsenb`. 

## Additional Information ##
I am new to this space and there could be things I'm missing here however, I've followed a number of guides and know people are using the LimeSDR USB successfully with srsLTE making it very frustrating to see mine not working at all ! 

srsLTE also makes a note of the LimeSDR being supported however as of right now, my setup does not work. 

Hopefully someone can give me a hand here, would love to get this setup working ! 

Cheers

grax004, 2020-10-12:
I have done some testing and downgraded srsLTE to `release_19_12`. When using this version, it seems to spin up successfully ( no errors ) however, I cannot see the network on the device ( SIM has been programmed correctly I believe ). Is there again, something I am missing with the antenna configuration and getting it to be seen on the device? 

Here's the output of the startup on `release_19_12`: 
```
Built in Release mode using commit d045213fb on branch HEAD.                                          
                                                                                                      
---  Software Radio Systems LTE eNodeB  ---                                                           
                                                                                                      
Reading configuration file enb.conf...                                                          
Opening 1 RF devices with 1 RF channels...                                                            
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 2.0] 9072C02881916, me$
ia=USB 2.0, module=FX3, name=LimeSDR-USB, serial=0009072C02881916,                                    
[INFO] Make connection: 'LimeSDR-USB [USB 2.0] 9072C02881916'                                         
[INFO] Reference clock 30.72 MHz                                                                      
[INFO] Device name: LimeSDR-USB                                                                       
[INFO] Reference: 30.72 MHz                                                                           
[INFO] LMS7002M register cache: Disabled                                                              
Setting up Rx stream with 1 channel(s)                                                                
Setting up Tx stream with 1 channel(s)                                                                
[INFO] RX LPF configured                                                                              
[INFO] RX LPF configured                                                                              
Set Rx bandwidth to 2.50 MHz                                                                          
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter s
et to 2.5 MHz. Preemphasis filter not active                                                          
[INFO] TX LPF configured                                                                              
Set Tx bandwidth to 5.00 MHz                                                                          
[INFO] Filter calibrated. Filter order-2nd, set to 50 MHz                                             
[INFO] TX LPF configured                                                                              
Set Tx bandwidth to 50.00 MHz                                                                         
Available device sensors:                                                                             
 - clock_locked                                                                                       
 - lms7_temp                                                                                          
Available sensors for Rx channel 0:                                                                   
 - lo_locked                                                                                          
Setting Rx antenna to LNAH                                                                            
Setting Tx antenna to BAND2                                                                           
[INFO] Tx calibration finished                                                                        
[INFO] Rx calibration finished                                                                        
State of gain elements for Rx channel 0 (AGC not supported):                                          
 - TIA: 9.00 dB                                                                                       
 - LNA: 30.00 dB                                                                                      
 - PGA: -4.00 dB                                                                                      
State of gain elements for Tx channel 0 (AGC not supported):                                          
 - PAD: 45.00 dB                                                                                      
 - IAMP: 0.00 dB       
Rx antenna set to LNAH                                                                               
Tx antenna set to BAND2                                                                               
                                                                                                      
Warning burst preamble is not calibrated for device lime. Set a value manually                        
                                                                                                      
Setting frequency: DL=2129.0 Mhz, UL=1729.0 MHz                                                       
Setting Sampling frequency 11.52 MHz                                                                  
                                                                                                      
==== eNodeB started ===                                                                               
[... cut ...]
RF status: O=0, U=0, L=2                                                                              
RF status: O=0, U=0, L=3                                                                              
RF status: O=0, U=0, L=2                                                                              
RF status: O=0, U=0, L=3                                                                              
^CStopping srsENB... Press Ctrl+C 9 more times to force stop                              
```

andrepuschmann, 2020-10-20:
Check your USB connection. Soapy reports USB2, you need USB3. I assume the issue is due to that, closing therefore. Please feel free to reopen if it still exists.

closed 2020-10-20 by andrepuschmann

tyc85, 2020-10-31:
Hi @andrepuschmann, I would like to have a quick follow-up question on this since I faced the same issue when upgrading the release: is this a change on Soapy API calls on srsLTE that it strictly requires USB3 in order for srsenb to work (we noticed srsue does not complain)? It looks like LimeSDR complaints according to the logs:
TuneVCO(CGEN) - failed to lock (cmphl!=0)

What I don't fully understand is when I test API call manually setSampleRate it doesn't report the same problem. 

I would be happy to help see if there's a way to make it work on both USB2 and USB3 if you can show me a few pointers. 

Thanks in advanced!

andrepuschmann, 2020-10-31:
USB2 would work as long as you don't exceed the available bandwidth, but it's certainly not recommended.

tyc85, 2020-10-31:
Hi @andrepuschmann, thanks for the quick response. I wonder whether there's other things I missed. I tried sampling rate of 5.76 MHz (nrb=25) but also see the same error:
....
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(491.52 MHz) failed
....
[ERROR] setSampleRate(Rx, 0, 5.76 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed

9600, 2020-11-17:
Just to note that USB 2.0 won't provide sufficient power for a LimeSDR-USB and you may run into various issues that are ultimately related to brownout, if you are using this and with no external power supply.

tyc85, 2020-11-20:
Hi Andre and 9600, thanks for pointing out the bus bandwidth and power supply issue. I understand the downside of USB 2 and are solving it in parallel. The issue on setting sampling rate for lime SDR remains for USB3 as well. Retrying works and I am sending a PR for more details. Thanks