srsran/srsRAN_4G #563: iPhone Connects to Network but Samsung Wont

matthew-william-lock, 2020-10-12:
Hi there, I am sorry to open an issue here but I am at a loss.

I'm running the srsEnb using a **USRP B210** and transmitting using EARFCN 1325.
My OS is **Ubuntu 18.04** and **UHD drivers are 3.15.0**
I have configured the USIMs correctly as far as I am aware.

I have tested the network with an iPhone 6 and iPhone 8 and both devices are able to successfully detect and attach onto the network. Both devices show that transmission is correctly occurring over EARFCN 1325

I am unfortunately not able to reproduce this with a Samsung device as it simply cannot detect the network. I have tested with a Samsung S7, Samsung S9 and a Samsung S10.

As noted in the documentation UE are expected to work in different ways, but I can't seem to understand why none of the Samsung devices are able to detect the network. With the iPhones working fine I image the network is correctly deployed. Any suggestions would be appreciated. I have already followed the following advice tips : [1](http://www.softwareradiosystems.com/pipermail/srslte-users/2018-June/001662.html), [2](https://docs.srslte.com/en/latest/srsue/source/3_ue_trouble.html)

andrepuschmann, 2020-10-13:
Are you using a GPSDO or external reference with your B210? If not, that could be an issue. Some phones can be very picky.

matthew-william-lock, 2020-10-13:
Hi @andrepuschmann. I unfortunately do not have access to a GPSDO. Is there potentially any way of working around this?

andrepuschmann, 2020-10-14:
You somehow need to calibrate your eNB transmitter. If you have another SDR you could use this to sync to a local GSM basestation and use this as reference for your actual USRP that you wish to calibrate. Check kalibrate-rtl and friends.

nomidaepapi, 2021-12-21:
Hi, 

Maybe try EARFCNs such as: 1300, 1800, 1900, 9260, 800. Helped me.

closed 2023-04-14 by ismagom