srsran/srsRAN_4G #565: Can’t see Ethernet port traffic

Jackrabrett, 2020-10-13:
I am unable to view/export the traffic coming into my UE via my eNb/epc. I can ssh into my UE and see the data that way but it is very laggy and not quite what I'm looking for. Does anyone know how to bridge an Ethernet port to the established LTE network?

andrepuschmann, 2020-10-14:
It's impossible to help you with your issue if you don't provide sufficient details on your setup. Use the template, think about each and every field, and fill it accordingly. I am closing the issue. Please feel free to reopen.

closed 2020-10-14 by andrepuschmann

Jackrabrett, 2020-10-15:
I am using two Intel NUC's as my UE and eNb/epc they are both running on Ubuntu 18.04.5. I am using two Ettus USRP B210's to communicate between the NUC's. Communication via ping and ssh is solid, I have IP Forwarding enabled in "/proc/sys/net/ipv4/ip_forward". The data that I want to be able to access via the base station (eNb/epc) is connected to a UE device through the Ethernet port called eno1, the IP address: 192.168.1.8, my UE is connected to the base station through the usrp device called tun_srsue running a POINTTOPOINT connection to the base station with the static IP address of 172.16.0.2. The base station's up address is 172.16.0.1, from there I can see the connection, ping the UE, ssh into the UE, and even use ssh -X Firefox to open up a web browser and type in 192.168.1.8 and see the data the presents itself in an html form. What isn't working is being able to traffic the data though the UE to my basestaion directly. Ive been trying to create bridges and tunnels but have been unsecessful. Is there anything else needed?