srsran/srsRAN_4G #567: Ethernet traffic not visible on UE via eNb

Jackrabrett, 2020-10-15:
I am using two Intel NUC's as my UE and eNb/epc they are both running on Ubuntu 18.04.5. I am using two Ettus USRP B210's to communicate between the NUC's. Communication via ping and ssh is solid, I have IP Forwarding enabled in "/proc/sys/net/ipv4/ip_forward". The data that I want to be able to access via the base station (eNb/epc), and eventually from a 2nd UE, is connected to a UE device through the Ethernet port called eno1, the IP address: 192.168.1.8, my UE is connected to the base station through the rf connection called tun_srsue running a POINTTOPOINT connection to the base station with the static IP address of 172.16.0.2. The base station's up address is 172.16.0.1, from there I can see the connection, ping the UE, ssh into the UE, and even use ssh -X Firefox to open up a web browser and type in 192.168.1.8 and see the data the presents itself in an html form. What isn't working is being able to traffic the data though the UE to my basestaion directly. Ive been trying to create bridges and tunnels but have been unsecessful. Is there anything else needed?

Jackrabrett, 2020-10-17:
"sudo ip route add 'IP_ADDR_TO' via 'IP_ADDR_FROM'" seems to work. But what I want to do is have access to ip addresses further along the path. I tried achieving this by doing some sort of via via thing but was unsuccessful, it says this is done using "nexthop". I'm not sure of the correct way to input it though, does anyone have experience using this command?

andrepuschmann, 2020-10-20:
I am sorry but this is a Linux networking question and not at all related to srsLTE. Closing therefore. Better ask on stackoverflow or other more appropriate places.

closed 2020-10-20 by andrepuschmann

triple-it, 2020-11-03:
So you want to ip route *over* the LTE network?
e.g.
NUC_client --> NUC_ue1 --[lte ran] --> NUC_enb_epc  ----[NAT] ---> 8.8.8.8/32 ?

A quick test on my side, says that is indeed not possible. But I am unsure why.
A simple "ip add route 8.8.8.8/32 via NUC_ue1" on the NUC_client should be enough if the NUC_ue1 default routing is ok.
But it isn't...

You could of course resort to NATting, but I understand you have a requirement of plain ip routing. That currently does not work over the LTE (tun devices). I don't know why precisely yet..

Would things like RIPv2 or BGP routing over the LTE ran work?

triple-it, 2020-11-03:
From NUC_ue --> NUC_enb_epc   not problem routing whatsoever. (Without NATTING!)
From NUC_enb_epc --> NUC_ue, i cannot ip route however...