srsran/srsRAN_4G #569: setFrequencyCGEN(368.64 MHz) failed LimeSDR with 20.04 version

CassianoDev, 2020-10-19:
## Issue Description ##
I configured the eNb in ubuntu 18.04 and everything is in compliance, except when I start it

## Setup Details ##
LimeSDR USB 3.0, Ubuntu 18.04, Intel Core I7 4790, 32GB RAM, NVMe

## Expected Behavior ##
Start eNb successfully!

## Actual Behaviour ##
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file ./enb.conf...

Built in Release mode using 20.04.0.

/root/srsLTE-release_20_04/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 3400

Opening 1 channels in RF device=default with args=default
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9081C05C20B12, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009081C05C20B12, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9081C05C20B12'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAL
Tx antenna set to BAND1
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed

==== eNodeB started ===
Type <t> to view trace
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0

SoapySDR info:
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.7.1-myriadrf1~ubuntu18.04
API Version: v0.7.1
ABI Version: v0.7

## Additional Information ##
Work with versions <= 19.12

andrepuschmann, 2020-10-20:
Hey, this is likely a version mismatch between the SoapyLMS  driver and srsLTE. In the 20.04 release we've removed some calls inside the Soapy lib that are now part of the SoapyLMS driver. You need to make sure to run a recent version, perhaps best is to use the current Git head. If the problem remains, please file a bug on the LimeSuite repo. Thanks

closed 2020-10-20 by andrepuschmann

rafis, 2020-11-12:
After upgrading SoapySDR from v0.7.1 to v0.8.0 the errors
```
[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed
```
are disappeared. Now everything works!

breakthrough55, 2023-06-07:
hello rafis! I meet the same problem.
`[ERROR] TuneVCO(CGEN) - failed to lock (cmphl!=3)
[ERROR] SetFrequencyCGEN(368.64 MHz) failed
[INFO] RX LPF configured
[ERROR] setSampleRate(Rx, 0, 11.52 MHz) Failed
setSampleRate Rx fail: SoapyLMS7::setSampleRate() failed`
 My configuration is Ubuntu18.04 and srsLTE v20.04.2. I updated SoapySDR to 0.8.1 according to your requirements，but there is new problem 
`eoor:conflicting types for ‘SoapySDRRange’
eoor:conflicting types for ‘SoapySDRKwargs’
......`
@rafis @andrepuschmann