srsran/srsRAN_4G #570: Disconnecting rnti UE connection problem LimeSDR USB

CassianoDev, 2020-10-20:
## Issue Description ##
I have an eNb in version 20.04, Latest Lime Drivers...
Everything works very well, the EPC communicates with eNb and the UE is authenticated!
But a bad problem happens, there is a constant disconnection from the UE...
I've tried to adjust:
time_adv_nsamples, rrc_inactivity_timer and max_prach_offset_us.
Despite the improvement, the disconnection occurs at some point!
The signal does not fall, but stops surfing the internet.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
LimeSDR UBS 3.0, Core i74790,32GBRAM,NVMe,Ubuntu 18.04

## Expected Behavior ##
Disconnection is minimized as much as possible.

## Actual Behaviour ##
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file ./enb.conf...

Built in Release mode using commit 0ec49a7b1 on branch master.

/root/srsLTE/srsenb/src/enb_cfg_parser.cc.883: Force DL EARFCN for cell PCI=1 to 6400

Opening 1 channels in RF device=default with args=default
Soapy has found device #0: addr=1d50:6108, driver=lime, label=LimeSDR-USB [USB 3.0] 9081C05C20B12, media=USB 3.0, module=FX3, name=LimeSDR-USB, serial=0009081C05C20B12, 
Selecting Soapy device: 0
[INFO] Make connection: 'LimeSDR-USB [USB 3.0] 9081C05C20B12'
[INFO] Reference clock 30.72 MHz
[INFO] Device name: LimeSDR-USB
[INFO] Reference: 30.72 MHz
[INFO] LMS7002M register cache: Disabled
Setting up Rx stream with 1 channel(s)
Setting up Tx stream with 1 channel(s)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for Rx channel 0: 
 - lo_locked
[INFO] Tx calibration finished
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - PAD: 45.00 dB
 - IAMP: 0.00 dB
Rx antenna set to LNAL
Tx antenna set to BAND1

==== eNodeB started ===
Type <t> to view trace
Trying to start a plot but plots are disabled (ENABLE_GUI constant in sf_worker.cc)
[INFO] RX LPF configured
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 11.52 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Setting manual TX/RX offset to 50 samples
Setting frequency: DL=816.0 Mhz, UL=857.0 MHz for cc_idx=0
LRACH:  tti=5261, preamble=41, offset=5, temp_crnti=0x46
User 0x46 connected
RACH:  tti=1291, preamble=5, offset=5, temp_crnti=0x47
Disconnecting rnti=0x47.
User 0x46 connected
RACH:  tti=6111, preamble=22, offset=5, temp_crnti=0x48
Disconnecting rnti=0x48.
Disconnecting rnti=0x46.
RACH:  tti=6271, preamble=20, offset=5, temp_crnti=0x49
User 0x49 connected
RACH:  tti=91, preamble=14, offset=5, temp_crnti=0x4a
Disconnecting rnti=0x4a.
Disconnecting rnti=0x49.
RACH:  tti=211, preamble=26, offset=5, temp_crnti=0x4b
User 0x4b connected
RACH:  tti=1831, preamble=15, offset=5, temp_crnti=0x4c
Disconnecting rnti=0x4c.
User 0x4b connected
RACH:  tti=7191, preamble=16, offset=5, temp_crnti=0x4d
Disconnecting rnti=0x4d.
RACH:  tti=7681, preamble=28, offset=5, temp_crnti=0x4e
Disconnecting rnti=0x4e.
RACH:  tti=8301, preamble=10, offset=5, temp_crnti=0x4f
Disconnecting rnti=0x4f.
Disconnecting rnti=0x4b.
RACH:  tti=8461, preamble=19, offset=5, temp_crnti=0x50
User 0x50 connected
RACH:  tti=8271, preamble=50, offset=5, temp_crnti=0x51
Disconnecting rnti=0x51.
RACH:  tti=8421, preamble=3, offset=5, temp_crnti=0x52
Disconnecting rnti=0x52.
RACH:  tti=8551, preamble=20, offset=5, temp_crnti=0x53
Disconnecting rnti=0x53.
RACH:  tti=8741, preamble=44, offset=5, temp_crnti=0x54
Disconnecting rnti=0x54.
RACH:  tti=8871, preamble=1, offset=5, temp_crnti=0x55
Disconnecting rnti=0x55.
RACH:  tti=9101, preamble=21, offset=5, temp_crnti=0x56
Disconnecting rnti=0x56.
RACH:  tti=9261, preamble=29, offset=5, temp_crnti=0x57
Disconnecting rnti=0x57.
RACH:  tti=9421, preamble=26, offset=5, temp_crnti=0x58
Disconnecting rnti=0x58.
RACH:  tti=9581, preamble=34, offset=5, temp_crnti=0x59
Disconnecting rnti=0x59.
RACH:  tti=9671, preamble=26, offset=5, temp_crnti=0x5a
Disconnecting rnti=0x5a.
RACH:  tti=9851, preamble=13, offset=5, temp_crnti=0x5b
Disconnecting rnti=0x5b.
RACH:  tti=9991, preamble=50, offset=5, temp_crnti=0x5c
Disconnecting rnti=0x5c.
RACH:  tti=10141, preamble=28, offset=5, temp_crnti=0x5d
Disconnecting rnti=0x5d.
RACH:  tti=101, preamble=6, offset=5, temp_crnti=0x5e
Disconnecting rnti=0x5e.
RACH:  tti=301, preamble=25, offset=5, temp_crnti=0x5f
Disconnecting rnti=0x5f.
RACH:  tti=461, preamble=35, offset=5, temp_crnti=0x60
Disconnecting rnti=0x60.
RACH:  tti=621, preamble=14, offset=5, temp_crnti=0x61
Disconnecting rnti=0x61.
RACH:  tti=781, preamble=33, offset=5, temp_crnti=0x62
Disconnecting rnti=0x62.
RACH:  tti=911, preamble=12, offset=5, temp_crnti=0x63
Disconnecting rnti=0x63.
RACH:  tti=1101, preamble=16, offset=5, temp_crnti=0x64
Disconnecting rnti=0x64.
RACH:  tti=1261, preamble=20, offset=5, temp_crnti=0x65
Disconnecting rnti=0x65.
RACH:  tti=1831, preamble=30, offset=5, temp_crnti=0x66
Disconnecting rnti=0x66.
Disconnecting rnti=0x50.
RACH:  tti=2001, preamble=11, offset=5, temp_crnti=0x67
User 0x67 connected
RACH:  tti=7981, preamble=39, offset=5, temp_crnti=0x68
Disconnecting rnti=0x68.
Disconnecting rnti=0x67.
RACH:  tti=8141, preamble=29, offset=5, temp_crnti=0x69
User 0x69 connected
Disconnecting rnti=0x69.
^CStopping ..
---  exiting  ---

## Steps to reproduce the problem ##
Run srsenb on LimeSDR USB3.0 and UE attached 

## Additional Information ##
Other versions like 19.12 and earlier also experience the same problem.

andrepuschmann, 2020-10-20:
Could you try with the 'next' branch? I see some lates on the console. Try to make sure you don't have anything else running that slows down your machine

CassianoDev, 2020-10-21:
The Computer is idle in processes when I am running srsenb...
CPU1:30%,CPU2:15%,CPU3:~3%,CPU4:~3%,RAM:5.37 of 32GB
I cannot understand the reason for the disconnection issue!

CassianoDev, 2020-10-21:
> Could you try with the 'next' branch? I see some lates on the console. Try to make sure you don't have anything else running that slows down your machine

The Computer is idle in processes when I am running srsenb...
CPU1:30%,CPU2:15%,CPU3:~3%,CPU4:~3%,RAM:5.37 of 32GB
I cannot understand the reason for the disconnection issue!

andrepuschmann, 2020-10-21:
Try `next` and send eNB logs in info mode. Thanks

CassianoDev, 2020-10-21:
> Try `next` and send eNB logs in info mode. Thanks
I'am running in PHY LOG INFO...
[Note that I only have 1UE attached]
The relevant informations that occurred at the time of detached was:
13:49:33.101221 [MAC ] [W] [ 2552] SCHED: discarding TB 0 pid=0, tti=2552, maximum number of retx exceeded (4) 
13:49:34.401193 [MAC ] [W] [ 3852] SCHED: discarding TB 0 pid=4, tti=3852, maximum number of retx exceeded (4)
13:49:34.401201 [PHY1] [I] [ 3852] PUSCH: cc=0, rnti=0x46, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=1, crc=KO, avg_iter=10.0, snr=1.7 dB, epre=28.2 dBfs, ta=-0.2 us, t=112 us
13:49:34.401216 [PHY1] [I] [ 3852] PHICH: rnti=0x46, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3856
13:49:34.405094 [PHY1] [I] [ 3856] PUSCH: cc=0, rnti=0x46, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=2, crc=KO, avg_iter=10.0, snr=1.2 dB, epre=28.2 dBfs, ta=-0.2 us, t=115 us
13:49:34.405121 [PHY1] [I] [ 3856] PHICH: rnti=0x46, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3860
13:49:34.409045 [PHY1] [I] [ 3860] PUCCH: cc=0; rnti=0x46, f=1, n_pucch=0, dmrs_corr=0.528, corr=0.584, sr=yes
13:49:34.409118 [PHY1] [I] [ 3860] PDCCH: cc=0, f=0, cce= 8, L=1, riv=347, mcs=0, rv=0, ndi=1, f_h=n/a, cqi=no, tpc_pusch=1, dmrs_cs=0, tti_tx_dl=3864
13:49:34.460231 [PHY0] [I] [ 3910] PRACH: cc=0, 0/1, preamble=36, offset=5.1 us, peak2avg=615.6, max_offset=30.0 us
13:49:34.461080 [PHY0] [I] [ 3912] PDCCH: cc=0, f=1A, cce= 0, L=2, riv=100, pid=0, mcs={1}, ndi={0}, tpc_pucch=0, tti_tx_dl=3916
13:49:34.461109 [PHY0] [I] [ 3912] PDSCH: cc=0, rnti=0x2, nof_prb=3, nof_re=450, tbs={11}, mod={2}, rv={0}, tti_tx_dl=3916
13:49:34.471413 [PHY0] [I] [ 3922] PUSCH: cc=0, rnti=0x47, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=0, crc=KO, avg_iter=10.0, snr=-1.3 dB, epre=16.0 dBfs, ta=1.3 us, t=75 us
13:49:34.471461 [PHY0] [I] [ 3922] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3926
13:49:34.479045 [PHY0] [I] [ 3930] PUSCH: cc=0, rnti=0x47, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=2, crc=KO, avg_iter=10.0, snr=-0.3 dB, epre=15.3 dBfs, ta=-0.0 us, t=38 us
13:49:34.479068 [PHY0] [I] [ 3930] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3934
13:49:34.487201 [PHY0] [I] [ 3938] PUSCH: cc=0, rnti=0x47, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=3, crc=KO, avg_iter=10.0, snr=-0.5 dB, epre=15.1 dBfs, ta=0.9 us, t=42 us
13:49:34.487227 [PHY0] [I] [ 3938] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3942
13:49:34.495020 [PHY0] [I] [ 3946] PUSCH: cc=0, rnti=0x47, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=1, crc=OK, avg_iter=1.0, snr=2.3 dB, epre=21.3 dBfs, ta=0.4 us, t=27 us
13:49:34.495062 [PHY0] [I] [ 3946] PHICH: rnti=0x47, hi=1, I_lowest=2, n_dmrs=0, tti_tx_dl=3950
13:49:34.505229 [PHY0] [I] [ 3956] PDCCH: cc=0, f=1, cce= 0, L=3, rbg=0x10000, pid=0, mcs={6}, ndi={1}, tpc_pucch=1, tti_tx_dl=3960
13:49:34.505251 [PHY0] [I] [ 3956] PDSCH: cc=0, rnti=0x47, nof_prb=3, nof_re=450, tbs={32}, mod={2}, rv={0}, tti_tx_dl=3960
13:49:34.509146 [PHY0] [I] [ 3960] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.046, corr=0.000, sr=no
13:49:34.513030 [PHY0] [I] [ 3964] PUCCH: cc=0; rnti=0x47, f=1a, n_pucch=0, dmrs_corr=0.147, corr=0.000, ack=0
13:49:34.528950 [PHY0] [I] [ 3980] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.999, corr=0.998, sr=yes
13:49:34.529004 [PHY0] [I] [ 3980] PDCCH: cc=0, f=0, cce= 0, L=3, riv=347, mcs=0, rv=0, ndi=1, f_h=n/a, cqi=no, tpc_pusch=1, dmrs_cs=0, tti_tx_dl=3984
13:49:34.537466 [PHY0] [I] [ 3988] PUSCH: cc=0, rnti=0x47, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=0, crc=KO, avg_iter=10.0, snr=1.7 dB, epre=28.5 dBfs, ta=0.2 us, t=265 us
13:49:34.537503 [PHY0] [I] [ 3988] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=3992
13:49:34.545376 [PHY0] [I] [ 3996] PUSCH: cc=0, rnti=0x47, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=2, crc=KO, avg_iter=10.0, snr=0.9 dB, epre=28.5 dBfs, ta=0.1 us, t=262 us
13:49:34.545432 [PHY0] [I] [ 3996] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=4000
13:49:34.549239 [PHY0] [I] [ 4000] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.131, corr=0.000, sr=no
13:49:34.553458 [PHY0] [I] [ 4004] PUSCH: cc=0, rnti=0x47, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=3, crc=KO, avg_iter=10.0, snr=1.2 dB, epre=28.4 dBfs, ta=-0.4 us, t=290 us
13:49:34.553498 [PHY0] [I] [ 4004] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=4008
13:49:34.561520 [MAC ] [W] [ 4012] SCHED: discarding TB 0 pid=4, tti=4012, maximum number of retx exceeded (4)
13:49:34.561538 [PHY0] [I] [ 4012] PUSCH: cc=0, rnti=0x47, rb=(2,46), nof_re=6480, tbs=157, mod=2, rv=1, crc=KO, avg_iter=10.0, snr=1.2 dB, epre=28.4 dBfs, ta=0.3 us, t=248 us

after a while...

13:49:34.841232 [PHY0] [I] [ 4292] PHICH: rnti=0x47, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=4296
13:49:34.849133 [PHY0] [I] [ 4300] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.126, corr=0.000, sr=no
13:49:34.869059 [PHY0] [I] [ 4320] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.240, corr=-0.066, sr=no
13:49:34.889239 [PHY0] [I] [ 4340] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.216, corr=-0.074, sr=no
13:49:34.909064 [PHY0] [I] [ 4360] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.095, corr=0.000, sr=no
13:49:34.929246 [PHY0] [I] [ 4380] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.070, corr=0.000, sr=no
13:49:34.949097 [PHY0] [I] [ 4400] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.254, corr=0.063, sr=no
13:49:34.969290 [PHY0] [I] [ 4420] PUCCH: cc=0; rnti=0x47, f=1, n_pucch=0, dmrs_corr=0.331, corr=0.006, sr=no
[... cut ...]
13:49:38.539407 [PHY0] [I] [ 7990] PHICH: rnti=0x4d, hi=0, I_lowest=2, n_dmrs=0, tti_tx_dl=7994
13:49:38.547004 [PHY0] [I] [ 7998] PUSCH: cc=0, rnti=0x4d, rb=(2,4), nof_re=432, tbs=7, mod=2, rv=3, crc=OK, avg_iter=1.0, snr=3.1 dB, epre=22.3 dBfs, ta=0.2 us, t=15 us
13:49:38.547032 [PHY0] [I] [ 7998] PHICH: rnti=0x4d, hi=1, I_lowest=2, n_dmrs=0, tti_tx_dl=8002
13:49:38.549136 [PHY0] [I] [ 8000] PUCCH: cc=0; rnti=0x4c, f=1, n_pucch=0, dmrs_corr=0.077, corr=0.000, sr=no

Full log here:
[ue.zip](https://github.com/srsLTE/srsLTE/files/5416438/ue.zip)

CassianoDev, 2020-10-21:
> Try `next` and send eNB logs in info mode. Thanks

Log most relevant of RF:
14:08:12.158385 [ENB ] [I] Built in Release mode using commit 0ec49a7b1 on branch master.
14:08:12.205508 [RF  ] [I] Configuring physical DL channel 0 with band-pass filter (0.0, 0.0)
14:08:12.205517 [RF  ] [I] Configuring physical UL channel 0 with band-pass filter (0.0, 0.0)
14:08:15.728745 [RF  ] [I] Mapping RF channel 0 to logical carrier 0 on f_tx=816.0 MHz
14:08:15.764171 [RF  ] [I] Mapping RF channel 0 to logical carrier 0 on f_rx=857.0 MHz
14:08:24.023736 [MAC ] [W] [ 8212] SCHED: discarding TB 0 pid=4, tti=8212, maximum number of retx exceeded (4)
14:08:24.883849 [MAC ] [W] [ 9072] SCHED: discarding TB 0 pid=0, tti=9072, maximum number of retx exceeded (4)
14:08:46.963623 [MAC ] [W] [  432] SCHED: discarding TB 0 pid=0, tti=432, maximum number of retx exceeded (4)
14:08:47.003648 [MAC ] [W] [  472] SCHED: discarding TB 0 pid=0, tti=472, maximum number of retx exceeded (4)
14:08:47.043618 [MAC ] [W] [  512] SCHED: discarding TB 0 pid=0, tti=512, maximum number of retx exceeded (4)
14:08:47.083668 [MAC ] [W] [  552] SCHED: discarding TB 0 pid=0, tti=552, maximum number of retx exceeded (4)
14:17:09.563140 [MAC ] [W] [ 1272] SCHED: discarding TB 0 pid=0, tti=1272, maximum number of retx exceeded (4)
14:23:35.299811 [MAC ] [W] [ 8128] SCHED: discarding TB 0 pid=0, tti=8128, maximum number of retx exceeded (4)
14:23:36.116464 [MAC ] [W] [ 8945] SCHED: discarding TB 0 pid=1, tti=8945, maximum number of retx exceeded (4)
14:23:36.123607 [MAC ] [W] [ 8952] SCHED: discarding TB 0 pid=0, tti=8952, maximum number of retx exceeded (4)
14:23:36.163568 [MAC ] [W] [ 8992] SCHED: discarding TB 0 pid=0, tti=8992, maximum number of retx exceeded (4)
14:23:36.203421 [MAC ] [W] [ 9032] SCHED: discarding TB 0 pid=0, tti=9032, maximum number of retx exceeded (4)
14:23:36.243383 [MAC ] [W] [ 9072] SCHED: discarding TB 0 pid=0, tti=9072, maximum number of retx exceeded (4)
14:23:36.283398 [MAC ] [W] [ 9112] SCHED: discarding TB 0 pid=0, tti=9112, maximum number of retx exceeded (4)
14:23:36.323327 [MAC ] [W] [ 9152] SCHED: discarding TB 0 pid=0, tti=9152, maximum number of retx exceeded (4)
14:23:36.363351 [MAC ] [W] [ 9192] SCHED: discarding TB 0 pid=0, tti=9192, maximum number of retx exceeded (4)
14:23:36.496499 [MAC ] [W] [ 9325] SCHED: discarding TB 0 pid=5, tti=9325, maximum number of retx exceeded (4)
14:23:36.556325 [MAC ] [W] [ 9385] SCHED: discarding TB 0 pid=1, tti=9385, maximum number of retx exceeded (4)

14:24:45.438861 [RLC ] [W] Dropping first 105 B of SN 202 due to lost start segment
             0000: 85 10 fb 80 18 e1 cf 6f 73 63 b9 5f 52 ee af 62 
             0010: 8d 3b c1 b6 00 34 88 08 8d 0f 72 e9 52 7e 67 86 
14:24:54.777294 [RLC ] [W] Lost PDU SN: 990
14:24:54.777307 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777309 [RLC ] [W] Dropping PDU 991 due to lost start segment
14:24:54.777330 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777334 [RLC ] [W] Dropping PDU 992 due to lost start segment
14:24:54.777336 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777338 [RLC ] [W] Dropping PDU 993 due to lost start segment
14:24:54.777340 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777345 [RLC ] [W] Dropping PDU 994 due to lost start segment
14:24:54.777347 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777350 [RLC ] [W] Dropping PDU 995 due to lost start segment
14:24:54.777354 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777356 [RLC ] [W] Dropping PDU 996 due to lost start segment
14:24:54.777357 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777362 [RLC ] [W] Dropping PDU 997 due to lost start segment
14:24:54.777363 [RLC ] [W] PDU SN=990 lost, stop reassambling SDU (vr_ur_in_rx_sdu=989)
14:24:54.777367 [RLC ] [W] Dropping first 116 B of SN 998 due to lost start segment

14:26:16.443347 [MAC ] [W] [ 5432] SCHED: discarding TB 0 pid=0, tti=5432, maximum number of retx exceeded (4)
14:26:16.483376 [MAC ] [W] [ 5472] SCHED: discarding TB 0 pid=0, tti=5472, maximum number of retx exceeded (4)
14:26:17.883190 [MAC ] [W] [ 6872] SCHED: discarding TB 0 pid=0, tti=6872, maximum number of retx exceeded (4)
14:28:07.963412 [MAC ] [W] [ 4312] SCHED: discarding TB 0 pid=0, tti=4312, maximum number of retx exceeded (4)
14:28:08.003491 [MAC ] [W] [ 4352] SCHED: discarding TB 0 pid=0, tti=4352, maximum number of retx exceeded (4)
14:28:08.043479 [MAC ] [W] [ 4392] SCHED: discarding TB 0 pid=0, tti=4392, maximum number of retx exceeded (4)
14:28:08.083231 [MAC ] [W] [ 4432] SCHED: discarding TB 0 pid=0, tti=4432, maximum number of retx exceeded (4)
14:28:08.123548 [MAC ] [W] [ 4472] SCHED: discarding TB 0 pid=0, tti=4472, maximum number of retx exceeded (4)
14:28:08.163251 [MAC ] [W] [ 4512] SCHED: discarding TB 0 pid=0, tti=4512, maximum number of retx exceeded (4)
14:28:08.203267 [MAC ] [W] [ 4552] SCHED: discarding TB 0 pid=0, tti=4552, maximum number of retx exceeded (4)
14:29:07.723328 [MAC ] [W] [ 2632] SCHED: discarding TB 0 pid=0, tti=2632, maximum number of retx exceeded (4)
14:29:07.803203 [MAC ] [W] [ 2712] SCHED: discarding TB 0 pid=0, tti=2712, maximum number of retx exceeded (4)
14:29:07.843378 [MAC ] [W] [ 2752] SCHED: discarding TB 0 pid=0, tti=2752, maximum number of retx exceeded (4)
14:29:07.883405 [MAC ] [W] [ 2792] SCHED: discarding TB 0 pid=0, tti=2792, maximum number of retx exceeded (4)

Full RF log
[ue_rf.log](https://github.com/srsLTE/srsLTE/files/5416633/ue_rf.log)

andrepuschmann, 2020-10-21:
Your UL SNR is super low. Try to check your gain/antenna settings, etc.

CassianoDev, 2020-10-21:
> Your UL SNR is super low. Try to check your gain/antenna settings, etc.

how can i view the UE SNR live?

andrepuschmann, 2020-10-21:
Hit `t+Enter` on the eNB console.

CassianoDev, 2020-10-21:
> Hit `t+Enter` on the eNB console.

------DL--------------------------------UL------------------------------------
rnti cqi  ri mcs brate   ok  nok  (%)  snr  phr mcs brate   ok  nok  (%)   bsr
  46  14   0  19  287k   43    0   0% 1.18 27.80.10  115k   98  241  71%   0.0
  46  14   0  13  190k   55    0   0% 4.00 20.90.10  209k  173    0   0%   0.0
  46  15   0  13  5.2k    6    0   0% 2.17 30.40.10   33k   27    0   0%   0.0
  46  14   0 0.0     0    0    0   0%    0 31.0   0  5.0k    4  141  97%   0.0
  46  15   0 0.0     0    0    0   0%    0  0.0   0     0    0  200 100%   0.0
  46  15   0 0.0     0    0    0   0%    0  0.0   0     0    0  200 100%   0.0
  46  13   0 0.0     0    0    0   0% 0.74 29.20.10   11k    9  627  99% 2.41k
Disconnecting rnti=0x46.
RACH:  tti=2061, preamble=24, offset=0, temp_crnti=0x47
  47  15   0 1.5   855    5    0   0% 1.40 29.0 2.2  8.9k    6   65  92%   0.0
RACH:  tti=3001, preamble=23, offset=0, temp_crnti=0x48
Disconnecting rnti=0x48.
Disconnecting rnti=0x47.
RACH:  tti=3131, preamble=34, offset=0, temp_crnti=0x49
User 0x49 connected
  49  12   0  14  3.3M  187    0   0% 2.54 28.0 0.5  223k  121    4 3.2%   0.0
  49  12   0  21   16M  764    0   0% 2.49 28.00.10  184k  162    0   0%   0.0
  49  13   0  24   17M  660    0   0% 2.60 28.10.10  254k  218    0   0%   0.0

The problem is really SNR?

andrepuschmann, 2020-10-27:
Yes, the issue is UL. The connection is very bad so the eNB releases the user eventually. Please check your antenna and gain settings, etc and make sure the UL is stable. Otherwise you'll run into issue like this.

closed 2020-10-27 by andrepuschmann