srsran/srsRAN_4G #578: Reading configuration file enb.conf... Opening 1 RF devices with 1 RF channels... [INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-release [INFO] [LOGGING] Fastpath logging disabled at runtime. Opening USRP with args: type=b200,master_clock_rate=23.04e6 [INFO] [B200] Detected Device: B210 [INFO] [B200] Operating over USB 3. [INFO] [B200] Initialize CODEC control... [INFO] [B200] Initialize Radio control... [INFO] [B200] Performing register loopback test...  [INFO] [B200] Register loopback test passed [INFO] [B200] Performing register loopback test...  [INFO] [B200] Register loopback test passed [INFO] [B200] Asking for clock rate 23.040000 MHz...  [INFO] [B200] Actually got clock rate 23.040000 MHz. Setting frequency: DL=889.0 Mhz, UL=844.0 MHz Setting Sampling frequency 11.52 MHz Failed to initiate SCTP socket. Attempting reconnection in 10 seconds

mimo123-code, 2020-10-28:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mimo123-code, 2020-10-28:
unable to start my open5gs core and srs lte

andrepuschmann, 2020-10-28:
Please provide sufficient content, including logs, etc. And also format the issue title and description correctly. Then feel free to reopen. Thanks

closed 2020-10-28 by andrepuschmann