srsran/srsRAN_4G #579: Send to invalid argument

tomchean, 2020-10-28:
#347  Issue Description ##
When I run epc and enb at same machine, everything works well, but when I run epc at different host, I can successfully initialize sctp, but after UE connected, srsenb show "send to : invalid argument", I am not sure if the config file is wrong.

enb host's ip is 192.168.50.8
epc host's ip is 192.168.50.180

and  my enb.conf looks like following
[enb]
mme_addr = 192.168.50.180
gtp_bind_addr = 127.0.1.1
s1c_bind_addr = 192.168.50.8
n_prb = 50

[image]

herlesupreeth, 2020-10-28:
I believe `gtp_bind_addr = 127.0.1.1` should be set to 192.168.50.8 as well

tomchean, 2020-10-28:
Really thanks, I fix it by setting  gtp_bind_addr = 192.168.50.8

closed 2020-10-28 by tomchean