srsran/srsRAN_4G #583: How to enable QAM256 in downlink eNb?

CassianoDev, 2020-10-31:
I've tried a few things but I can't!
is this possible?

andrepuschmann, 2020-10-31:
If you file an issue, please use the issue template and provide all relevant info. That's what it is there for. In this example, what phone are you using? Does it have QAM256 support?

CassianoDev, 2020-10-31:
> If you file an issue, please use the issue template and provide all relevant info. That's what it is there for. In this example, what phone are you using? Does it have QAM256 support?

Sorry I was in a hurry!
edited!

My phone is Xiaomi Note8

andrepuschmann, 2020-10-31:
> > If you file an issue, please use the issue template and provide all relevant info. That's what it is there for. In this example, what phone are you using? Does it have QAM256 support?
> 
> Sorry I was in a hurry!
> edited!
> 
> My phone is Xiaomi Note8, is supported 256QAM!

Again, please follow the issue template. Send a eNB PCAP and a log in info if possible.

CassianoDev, 2020-11-02:
> Again, please follow the issue template. Send a eNB PCAP and a log in info if possible.

Ok, thanks!

Files log and pcap annexed

CassianoDev, 2020-11-02:
> Ok, thanks!
>
> Files log and pcap annexed

[enb.log](https://github.com/srsLTE/srsLTE/files/5474708/enb.log)

[pcap.zip](https://github.com/srsLTE/srsLTE/files/5474712/pcap.zip)

CassianoDev, 2020-11-02:
> [enb.log](https://github.com/srsLTE/srsLTE/files/5474708/enb.log)
>
> [pcap.zip](https://github.com/srsLTE/srsLTE/files/5474712/pcap.zip)

andrepuschmann, 2020-11-02:
Thanks for the logs, we investigate.

CassianoDev, 2020-11-02:
> Thanks for the logs, we investigate.

Thank you so much for your help!

After fine uplink adjustments, LimeSDR USB is up 48h+ 2x2@10.

cbalint13, 2020-11-03:
> > Thanks for the logs, we investigate.
> 
> Thank you so much for your help!
> 
> After fine uplink adjustments, LimeSDR USB is up 48h+ 2x2@10.

@CassianoDev ,

What do you mean by fine uplink adjusment ?

CassianoDev, 2020-11-03:
> > > Thanks for the logs, we investigate.
> > 
> > 
> > Thank you so much for your help!
> > After fine uplink adjustments, LimeSDR USB is up 48h+ 2x2@10.
> 
> @CassianoDev ,
> 
> What do you mean by fine uplink adjusment ?

RX Gain, I was getting a lot of noise in RX (my antennas not good).

Adjust of UE PUSCH pwr for compensate the low gain and get high SNR.

Antenna height adjustment for frequency used (857 MHz) ARFCN 6400.
I don't have good experiences at high frequencies!

CassianoDev, 2020-11-03:
> Thanks for the logs, we investigate.

Hey...
256QAM works fine for me now at 20.10!
Is it possible to use 64QAM on the uplink?

thank you!

andrepuschmann, 2020-11-03:
> 256QAM works fine for me now at 20.10 release!

Ahhh, you tested with 20.04.2, just saw you were using `0ec49a7b1`. I didn't see that. Of course, it wasn't working with that version.

> Is it possible to use 64QAM on the uplink?

Not yet, but will be added.

Closing issue.

closed 2020-11-03 by andrepuschmann

joeldelosangeles, 2022-10-10:
Hi,  Is 64QAM UL now supported in the latest version 22.04?