srsran/srsRAN_4G #584: IP PREFIX route back towards UE

triple-it, 2020-11-03:
Would/could we expect to be able to route back an IP PREFIX towards an UE?
Without natting, without tunneling, e.g. no gre, ipinip, or vxlan.

I would have some usecases for that..
But it seems I cannot get it to work..

From NUC_ue --> NUC_enb_epc   no problem routing whatsoever. (Without NATTING!)
From NUC_enb_epc --> NUC_ue, i cannot ip route back over the UE however...

Steps to reproduce:

```
enb/epc:
======
root@# nohup srsepc  &
root@# nohup srsenb  &

ue:
===
root@# /usr/local/bin/srsepc_if_masq.sh eno1
Masquerading Interface eno1
root@#
root@# ping -c 1 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=118 time=2.44 ms

root@#  nohup srsue &  tail -F nohup.out | grep IP & 
Network attach successful. IP: 172.16.0.3

enb/epc:
======
root@#ip route add 8.8.8.8/32 via 172.16.0.3
root@#tcpdump -ni srs_spgw_sgi icmp &   ping -c 4 8.8.8.8

PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on srs_spgw_sgi, link-type RAW (Raw IP), capture size 262144 bytes
12:18:00.267712 IP 172.16.0.1 > 8.8.8.8: ICMP echo request, id 94, seq 2, length 64
12:18:01.291696 IP 172.16.0.1 > 8.8.8.8: ICMP echo request, id 94, seq 3, length 64
12:18:02.315690 IP 172.16.0.1 > 8.8.8.8: ICMP echo request, id 94, seq 4, length 64

ue:
==
Stays empty..
root# tcpdump -ni tun_srsue icmp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun_srsue, link-type RAW (Raw IP), capture size 262144 bytes
<void>
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel
```

triple-it, 2020-11-04:
I have created a workaround by adding an ipinip tunnel for traffic back from enb_epc --> ue

enb_epc:
=======
root@# ip route del 192.168.1.0/24 via 172.16.0.11 dev srs_spgw_sgi
root@# ip route add 192.168.1.0/24 via 172.16.0.11 dev tunl0
Error: Nexthop has invalid gateway.
root@# ip route add 192.168.1.0/24 via 172.16.0.11 dev tunl0  onlink

ue:
===
Traffic from UE --> enb_epc works RAW...

root@# tcpdump -ni tun_srsue icmp
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on tun_srsue, link-type RAW (Raw IP), capture size 262144 bytes
17:35:59.579203 IP 192.168.1.2 > 1.1.1.1: ICMP echo request, id 21727, seq 440, length 64
17:36:00.581107 IP 192.168.1.2 > 1.1.1.1: ICMP echo request, id 21727, seq 441, length 64
17:36:01.583002 IP 192.168.1.2 > 1.1.1.1: ICMP echo request, id 21727, seq 442, length 64
^C

enb_epc:
======
Traffic back from enb towards UE, I have to put into in a ipinip tunnel to make it work.

root@# tcpdump -ni srs_spgw_sgi
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on srs_spgw_sgi, link-type RAW (Raw IP), capture size 262144 bytes
17:32:39.163194 IP 192.168.1.2 > 1.1.1.1: ICMP echo request, id 21727, seq 240, length 64
17:32:39.173050 IP 172.16.0.1 > 172.16.0.11: IP 1.1.1.1 > 192.168.1.2: ICMP echo reply, id 21727, seq 240, length 64 (ipip-proto-4)
17:32:40.163215 IP 192.168.1.2 > 1.1.1.1: ICMP echo request, id 21727, seq 241, length 64
17:32:40.172513 IP 172.16.0.1 > 172.16.0.11: IP 1.1.1.1 > 192.168.1.2: ICMP echo reply, id 21727, seq 241, length 64 (ipip-proto-4)
^C

Then traffic flows *BEHIND* the ue works as expected.

root@triplemobile15:~# ping -c 1 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
64 bytes from 1.1.1.1: icmp_seq=1 ttl=54 time=20.3 ms

--- 1.1.1.1 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 20.369/20.369/20.369/0.000 ms
root@triplemobile15:~#

The question though is why this workaround is needed?

Why cannot I do a:
ip route del 192.168.1.0/24 via 172.16.0.11 dev tunl0 onlink
and a direct
ip route add 192.168.1.0/24 via 172.16.0.11 dev srs_spgw_sgi  [ onlink ]    <---- ????? Should be possible, I think

ismagom, 2023-04-14:
This is a general IP routing question.

closed 2023-04-14 by ismagom

triple-it, 2023-04-14:
As this was quite some time back i dont have everything crystal clear anymore..
But I will try to summarize, maybe it is helpful for some reader stumbling on this.

What i do remember is that we ended up patching a mobile core (open5gs/smf) for pushing the routed packets back to the right UE mac layer.
For UE Huawei dongels we also needed custom firmware that supported the ROUTE_BEHIND_MS feature, as Huawei calls the ip prefix routing support over the UE...

As stated here openRAN does not need patching and this normal routing should "just work".
So I did not get it to work, but maybe someone else will, hopefully.