srsran/srsRAN_4G #585: srsepc - TEID or UE IP pool incorrect rotation management

ECTPavloR, 2020-11-06:
## Issue Description ##
Hello! 
Using a few UEs under kind of load test with connecting/disconnect behavior for ~2h. At some point srsepc eating up an IP pool and continue allocating 0.0.0.0 addresses, while in fact only 25 UEs on the air.

## Setup Details ##
srslte 20.04
25UEs (based on Quectel modems)
BladeRF frontend

default epc config (just PLMN adjusted to my SIM cards)
[mme]
mme_code = 0x1a
mme_group = 0x0001
tac = 0x0007
mcc = 255
mnc = 17
mme_bind_addr = 127.0.1.100

[spgw]
gtpu_bind_addr   = 127.0.1.100
sgi_if_addr      = 172.16.0.1
sgi_if_name      = srs_spgw_sgi
max_paging_queue = 100

## Expected Behavior ##
cleaning IP pool from unused contexts and IP addresses re-allocation from declared /24 pool.

## Actual Behaviour ##
... 
Creating Session Response -- MME control TEID: 253
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 253
SPGW: Allocated User TEID 253
SPGW: Allocate UE IP 172.16.0.254
...
Creating Session Response -- MME control TEID: 254
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 254
SPGW: Allocated User TEID 254
SPGW: Allocate UE IP 0.0.0.0
...
Creating Session Response -- MME control TEID: 255
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 255
SPGW: Allocated User TEID 255
SPGW: Allocate UE IP 0.0.0.0
...
## Steps to reproduce the problem ##
Running few UEs with connect/disconnect behavior for a long time. In our case 25-26 UEs for ~2h to use the whole TEID and/or UE IP pool reaching TEID value >253. Could be used less # of UEs but for a longer time to consume the pull

## Additional Information ##
entire log is below. It looks good for all the other connections before TEID 253
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x4cdc6ad5
Attach request -- eNB-UE S1AP Id: 120
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 255172020080003
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 255172020080003
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 255172020080003
Sending ESM information request
UL NAS: Received ESM Information Response
ESM Info: APN internet
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 255172020080003
Creating Session Response -- MME control TEID: 253
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 253
SPGW: Allocated User TEID 253
SPGW: Allocate UE IP 172.16.0.254
Received Create Session Response
Create Session Response -- SPGW control TEID 253
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.254 to IMSI 255172020080003
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x3960003; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 255172020080003
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x4cdc6adc
Attach request -- eNB-UE S1AP Id: 121
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 255172020080002
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 255172020080002
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 255172020080002
Sending ESM information request
UL NAS: Received ESM Information Response
ESM Info: APN internet
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 255172020080002
Creating Session Response -- MME control TEID: 254
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 254
SPGW: Allocated User TEID 254
SPGW: Allocate UE IP 0.0.0.0
Received Create Session Response
Create Session Response -- SPGW control TEID 254
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 0.0.0.0 to IMSI 255172020080002
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x3970003; eNB GTP-U Address 127.0.1.1
UL NAS: Detach Request
Detach request -- IMSI 255172020080002
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_DELETE_SESSION_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 467
UE is not ECM connected. No need to release S1-U. MME UE S1AP Id 467
UE Context Release Completed.
Received UE Context Release Request. MME-UE S1AP Id 453
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Received UE Context Release Complete. MME-UE S1AP Id 453
There are active E-RABs, send release access bearers request
UE Context Release Completed.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x4cdc6ada
Attach request -- eNB-UE S1AP Id: 122
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 1
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 255172020080021
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 255172020080021
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 255172020080021
Sending ESM information request
UL NAS: Received ESM Information Response
ESM Info: APN internet
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 255172020080021
Creating Session Response -- MME control TEID: 255
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 255
SPGW: Allocated User TEID 255
SPGW: Allocate UE IP 0.0.0.0
Received Create Session Response
Create Session Response -- SPGW control TEID 255
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 0.0.0.0 to IMSI 255172020080021
Adding attach accept to Initial Context Setup Request
Sent Initial Context Setup Request. E-RAB id 5
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x3980003; eNB GTP-U Address 127.0.1.1
UL NAS: Detach Request

ECTPavloR, 2020-11-06:
P.S. Workaround: set static IPs in user_db.csv

andrepuschmann, 2020-11-07:
Thanks for reporting this, we'll look into it. Meanwhile, your approach to set static IPs circumvents the issue, as you've already noticed.

closed 2023-04-14 by ismagom