srsran/srsRAN_4G #586: Does it support new BladeRF 2.0 micro xA9/xA4?

K7cl, 2020-11-08:
## Issue Description ##
Does anybody try NEW bladeRF 2.0 micro? Does it totally support? I am considering in buy x115 or xA9
Thanks!

fllay, 2020-11-08:
I have tried. It is woking ok with bladeRF 2.0 micro.

fllay, 2020-11-08:
BTW, My system setup is in the following:

x86 pc with ubuntu 18.04
SDR: bladeRF 2.0 micro
UE: Samsung galaxy J7
SIM: Chinese blank SIM from Aliexpess. 

The only problem I have is that I could not find device_args name to enable bias_tee for both PA and LNA. I can use bladerf_cli to enable them but when I use srsenb, it resets all the setting. If anyone know what is the device_args name to enable bias tee, please let me know. I have a pretty reliable connection with Samsung Galaxy J7 with thin few meters. I will try to power PA and LNA using external power supply and see how much the range improvement.

K7cl, 2020-11-08:
> BTW, My system setup is in the following:
>
> x86 pc with ubuntu 18.04
> SDR: bladeRF 2.0 micro
> UE: Samsung galaxy J7
> SIM: Chinese blank SIM from Aliexpess.
>
> The only problem I have is that I could not [...]

Thanks. Did you try x115 or x40? Is device_args ok in those BladeRF 1.0? By the way, are you using xA9 or xA4?

fllay, 2020-11-08:
I am using xA4. I don't have bladeRF version 1.0 so I could not test it.

K7cl, 2020-11-09:
> I am using xA4. I don't have bladeRF version 1.0 so I could not test it.

OK, Thanks

ECTPavloR, 2020-11-09:
BladeRF2.0 xA4 - tested. Up to 20UE handing well.

fllay, 2020-11-10:
ECTPavloR , Did you get bias tee of BladeRF working? I tried to use device_args but I could not find any argument to enable bias tee.

ECTPavloR, 2020-11-11:
> ECTPavloR , Did you get bias tee of BladeRF working? I tried to use device_args but I could not find any argument to enable bias tee.

I'm running it in default without external amplifiers (unless I didn't get the purpose)

nomidaepapi, 2021-12-21:
Hi,

Read this link: https://nuand.com/forums/viewtopic.php?t=9863

It helped me. Just create a new conf file as the link says with the commands you need. So when your blade powers up, the configurations are read on power up.

sonium-labs, 2022-03-01:
Do you recommend the bias tee when using BladeRF with LTE signals? I've been trying to sync to a real eNB using my BladeRF 2.0 micro xA5, and have been able to acquire the MIB but not sync to SFN. Does anyone have any experience in attempting this? Thanks in advance!

safari-srslte, 2022-06-13:
i have a question, is it possible to connect/detect srslte on our phone when you are using actual sim of some network provider (as we didn't know much about the parameters of actual sim)

closed 2023-04-14 by ismagom

mrahman2-vt, 2023-04-14:
Can anyone help me how to remove my email from srslte, sick of getting
emails, unsubscribe several emails trends, still getting lots of emails,
seems like never stopping.