srsran/srsRAN_4G #590: eNB crashing for subsequent VoLTE calls

herlesupreeth, 2020-11-12:
## Issue Description ##
eNB crashes with the following logs, when placing subsequent calls from the same two UEs after the first call
```

WARNING: Found orphan containers (udr, upf, fhoss, mysql, pcscf, mme, dns, sgwu, ausf, mongo, hss, smf, pcrf, icscf, nrf, scscf, sgwc, amf, udm, rtpengine) for this project. If you removed or renamed this service in your compose file, you can run this command with the --remove-orphans flag to clean it up.
Creating srsenb ... done
Attaching to srsenb
srsenb    | ---  Software Radio Systems LTE eNodeB  ---
srsenb    | 
srsenb    | Reading configuration file /etc/srslte/enb.conf...
srsenb    | 
srsenb    | Built in Release mode using commit fd5cde525 on branch HEAD.
srsenb    | 
srsenb    | Opening 1 channels in RF device=default with args=clock=external
srsenb    | [INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-release
srsenb    | [INFO] [LOGGING] Fastpath logging disabled at runtime.
srsenb    | [INFO] [B200] Detected Device: B210
srsenb    | [INFO] [B200] Operating over USB 3.
srsenb    | [INFO] [B200] Initialize CODEC control...
srsenb    | [INFO] [B200] Initialize Radio control...
srsenb    | [INFO] [B200] Performing register loopback test... 
srsenb    | [INFO] [B200] Register loopback test passed
srsenb    | [INFO] [B200] Performing register loopback test... 
srsenb    | [INFO] [B200] Register loopback test passed
srsenb    | [INFO] [B200] Asking for clock rate 23.040000 MHz... 
srsenb    | [INFO] [B200] Actually got clock rate 23.040000 MHz.
srsenb    | Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
srsenb    | Setting frequency: DL=2660.0 Mhz, UL=2540.0 MHz for cc_idx=0
srsenb    | 
srsenb    | ==== eNodeB started ===
srsenb    | Type <t> to view trace
srsenb    | Closing stdin thread.
srsenb    | RACH:  tti=6981, cc=0, preamble=36, offset=0, temp_crnti=0x46
srsenb    | User 0x46 connected
srsenb    | User 0x46 connected
srsenb    | RACH:  tti=5621, cc=0, preamble=16, offset=0, temp_crnti=0x47
srsenb    | User 0x47 connected
srsenb    | User 0x47 connected
srsenb    | User 0x47 connected
srsenb    | User 0x46 connected
srsenb    | terminate called after throwing an instance of 'std::bad_alloc'
srsenb    |   what():  std::bad_alloc
srsenb    | pure virtual method called
srsenb    | terminate called recursively
srsenb    | pure virtual method called
srsenb    | terminate called recursively
srsenb    | terminate called recursively

```

## Setup Details ##
Running srsLTE enB (release_20_10) + open5gs EPC (tag 2.0.21) + kamailio IMS all in docker setup - https://github.com/herlesupreeth/docker_open5gs

SDR - USRP B210
UEs with sysmocom USIM/ISIM - Samsung Tab 6 lite and iPhone X

## Expected Behavior ##
Should not crash

## Actual Behaviour ##
eNB crashes the moment i attempt second call after disconnecting the first call

## Steps to reproduce the problem ##
- Run the EPC
- Run the srsLTE
- Attach the two UEs to IMS
- UEs successfully register with IMS
- Call one UE from another
- Disconnect the call
- Place another call
- (if by any chance you dont see the above eNB crash, try placing subsequent calls multiple times)

Please let me know if you need further information

andrepuschmann, 2020-11-18:
Hey Herle, thanks for flagging the issue. Could you perhaps try to run the same experiment with `gdb` and send a backtrace of that? Try to compile with `-DCMAKE_BUILD_MODE=RelWithDebInfo` and reduce the number of PRBs if its too slow with gdb. Thanks

herlesupreeth, 2020-11-23:
Hello @andrepuschmann, Unfortunately I am unable to reproduce this error. I will close this issue for now and re-open if i face it again

closed 2020-11-23 by herlesupreeth