srsran/srsRAN_4G #591: COTS US, Sumsung S9+, reattach failed, IMSI 000000000000000

jin-github1002, 2020-11-13:
## Issue Description ##
Hi , team
I use Sumsung S9+ to connect to srsLTE (version 19.12). Platform is Jetson TX2. ENB and EPC are all on a machine. When UE was trying to attach again after released ( not sure why released),  IMSI 000000000000000 was saw. Of course UE could not attach successfully.

## Setup Details ##
I use commercial UE to connect to srsLTE. I found that it could be connected srsLTE for a little while, then released. When UE was trying to attached again, it could not attach at all. I could not find the releasing reason and failed reason for re attach. 

## Expected Behavior ##
I hope to set up a test that two COTS UE connected to srsLTE. enb and epc could be on either a machine or two machines. Now I can not set up the UE being online for over 1 minutes. Test can not be done more. 

## Actual Behaviour ##
I just saw IMS. The exact reason for ue not being online for a long time needs to be found. 

## Additional Information ##
The output on epc are following. If you need more information, please reply this issue. Any help will be appreciated. 

Built in Release mode using 19.12.0.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01, PLMN: 61712
S1 Setup Request - TAC 7, B-PLMN 0
S1 Setup Request - Paging DRX 2
Sending S1 Setup Response
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x45b78740
Attach request -- eNB-UE S1AP Id: 1
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 19
PDN Connectivity Request -- ESM Information Transfer requested: true
UL NAS: Received Identity Response
ID Response -- IMSI: 001010123456780
Downlink NAS: Sent Authentication Request
UL NAS: Received Authentication Response
Authentication Response -- IMSI 001010123456780
UE Authentication Accepted.
Generating KeNB with UL NAS COUNT: 0
Downlink NAS: Sending NAS Security Mode Command.
UL NAS: Received Security Mode Complete
Security Mode Command Complete -- IMSI: 001010123456780
Sending ESM information request
UL NAS: Received ESM Information Response
ESM Info: APN anritsu
Getting subscription information -- QCI 7
Sending Create Session Request.
Creating Session Response -- IMSI: 1010123456780
Creating Session Response -- MME control TEID: 1
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_CREATE_SESSION_REQUEST
SPGW: Allocated Ctrl TEID 1
SPGW: Allocated User TEID 1
SPGW: Allocate UE IP 172.16.0.2
Received Create Session Response
Create Session Response -- SPGW control TEID 1
Create Session Response -- SPGW S1-U Address: 127.0.1.100
SPGW Allocated IP 172.16.0.2 to IMSI 001010123456780
Adding attach accept to Initial Context Setup Request
Initial Context Setup Request -- eNB UE S1AP Id 1, MME UE S1AP Id 1
Initial Context Setup Request -- E-RAB id 5
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
Initial Context Setup Request -- QCI 7 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x4c0003; eNB GTP-U Address 127.0.1.1
UL NAS: Received Attach Complete
Unpacked Attached Complete Message. IMSI 1010123456780
Unpacked Activate Default EPS Bearer message. EPS Bearer id 5
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Sending EMM Information
Received UE Context Release Request. MME-UE S1AP Id 1
There are active E-RABs, send release access bearers request
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_RELEASE_ACCESS_BEARERS_REQUEST
UE is ECM IDLE.
Received UE Context Release Complete. MME-UE S1AP Id 1
UE is not ECM connected. No need to release S1-U. MME UE S1AP Id 1
UE Context Release Completed.
Found UE for Downlink Notification 
MME Ctr TEID 0x1, IMSI: 001010123456780
Initial UE message: NAS Message Type Unknown
Received Initial UE message -- Service Request
Service request -- S-TMSI 0x6ed30a0a
Service request -- eNB UE S1AP Id 2
Service Request -- Short MAC valid
Service Request -- User is ECM DISCONNECTED
UE previously assigned IP: 172.16.0.2
Generating KeNB with UL NAS COUNT: 3
UE Ctr TEID 0
Initial Context Setup Request -- eNB UE S1AP Id 2, MME UE S1AP Id 2
Initial Context Setup Request -- E-RAB id 5
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
Initial Context Setup Request -- S1-U TEID 0x1. IP 127.0.1.100 
Initial Context Setup Request -- QCI 7 
Received Initial Context Setup Response
E-RAB Context Setup. E-RAB id 5
E-RAB Context -- eNB TEID 0x4f0003; eNB GTP-U Address 127.0.1.1
Initial Context Setup Response triggered from Service Request.
Sending Modify Bearer Request.
Received GTP-C PDU. Message type: GTPC_MSG_TYPE_MODIFY_BEARER_REQUEST
Modify Bearer Request received after Downling Data Notification was sent
Initial UE message: NAS Message Type Unknown
Received Initial UE message -- Service Request
Service request -- S-TMSI 0x6ed30a0a
Service request -- eNB UE S1AP Id 3
Service Request -- Short MAC invalid
Service Request -- Short MAC invalid. Sending service reject.
Received UE Context Release Complete. MME-UE S1AP Id 2
No UE context to release found. MME-UE S1AP Id: 2
Initial UE message: NAS Message Type Unknown
Received Initial UE message -- Service Request
Service request -- S-TMSI 0x6ed30a0a
Service request -- eNB UE S1AP Id 4
UE is not EMM-Registered.
Initial UE message: NAS Message Type Unknown
Received Initial UE message -- Service Request
Service request -- S-TMSI 0x6ed30a0a
Service request -- eNB UE S1AP Id 5
UE is not EMM-Registered.
Received UE Context Release Request. MME-UE S1AP Id 4
No UE context to release found. MME-UE S1AP Id: 4
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x6ed30a0a
Attach request -- eNB-UE S1AP Id: 6
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 20
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Found UE context. IMSI: 000000000000000, old eNB UE S1ap Id 3, old MME UE S1AP Id 3
GUTI Attach request NAS integrity failed.
RE-starting authentication procedure.
User not found at HSS. IMSI: 000000000000000
User not found. IMSI 000000000000000
Received UE Context Release Request. MME-UE S1AP Id 5
No UE context to release found. MME-UE S1AP Id: 5
Received UE Context Release Request. MME-UE S1AP Id 3
No UE context to release found. MME-UE S1AP Id: 3
Received UE Context Release Request. MME-UE S1AP Id 524290
No UE context to release found. MME-UE S1AP Id: 524290
Initial UE message: LIBLTE_MME_MSG_TYPE_ATTACH_REQUEST
Received Initial UE message -- Attach Request
Attach request -- M-TMSI: 0x6ed30a0a
Attach request -- eNB-UE S1AP Id: 7
Attach request -- Attach type: 2
Attach Request -- UE Network Capabilities EEA: 11110000
Attach Request -- UE Network Capabilities EIA: 01110000
Attach Request -- MS Network Capabilities Present: true
PDN Connectivity Request -- EPS Bearer Identity requested: 0
PDN Connectivity Request -- Procedure Transaction Id: 20
PDN Connectivity Request -- ESM Information Transfer requested: true
Attach Request -- Found previously attach UE.
Found UE context. IMSI: 000000000000000, old eNB UE S1ap Id 6, old MME UE S1AP Id 7
GUTI Attach request NAS integrity failed.
RE-starting authentication procedure.
User not found at HSS. IMSI: 000000000000000
User not found. IMSI 000000000000000
Received UE Context Release Request. MME-UE S1AP Id 524290
No UE context to release found. MME-UE S1AP Id: 524290

andrepuschmann, 2020-11-16:
Thanks for reporting the issue. Is this happening reproducibly with your setup? Could you perhaps try the 20.10 release and send PCAPs and logs along, thanks.

jin-github1002, 2020-11-17:
@andrepuschmann Thank you for your reply.
The first confusing is: It always happens that IMSI is 00000.... when reattaching whatever UE is sumsung S9+ and iphone7. I have to let ue return to the network by setting it to flight mode once or more. The second confusing is : It is much more possible for iphone to attach successfully than sumsung.  Even though attaching successfully, sumsung can not be online for more than one minute. Then epc received a ue release. 
I have tried srsLTE version 20.10, but it is much serious than 19.12. Then I have to return to 19.12 to test. Do you need me to reproduct this with 20.10? Can I use 19.12 to get the logs?

andrepuschmann, 2020-11-17:
We recommend to use 20.10, yes. IMHO the issue should be fixed there. What exactly is making you use 19.12? If there is another issue, we might need to fix that in 20.10.

jin-github1002, 2020-11-18:
@andrepuschmann Two weeks ago, I tried to run 20.10 with LimeSDR USB to implement a LTE on two Jetson TX2. UE is iphone or sumsung S9+. I use the dl_earfcn = 1934, prb=15 and tx_gain =60, rx_gain=40 ( it is from the online tutorial of srsLTE).   I failed to attach the network. But It work with 19.12. I have logs on TX2. Do you need the logs?
I have another question: why cant I attach successfully every time? It means ue often can not attach successfully. Is there any other .conf that I need to set? I think the setting above is more stable on desk PC.

andrepuschmann, 2020-11-18:
We are mixing issues here. The "IMSI 0000" issue is supposedly fixed in 20.10. If not, please provide logs/PCAP for i. I will therefore close this ticket.

The UE not attaching is a different issue. Note that the Jetson TX2 (actually any ARM board) isn't officially supported by us. It might work or might not work. The performance on those platforms is very limited compared to x86 and there are many knobs that affect the performance. For beginners we definitely suggest x86.

closed 2020-11-18 by andrepuschmann