srsran/srsRAN_4G #595: srsue fails to find cells when nof_antennas = 2 and mulltiple dl_earfcn's are configured

nedsana, 2020-11-23:
## Issue Description ##
Hi, I am facing a problem with cell detection, when I use 2 antennas and configure multiple channels in "dl_earfcn" entry.
I detected it by listening to the nearby operator cells.

Below I will place full config file. Here I will only list the significant, in my opinion, settings and will give some explanations:

**_Experiment 1 ue.conf:_**
- dl_earfcn = 1400,1425,1499
- nof_antennas = 2

**_Expected behavior:_** my personal phone is connected to cells from **_Band 3, dl_earfcn 1425_,** so I was expecting the SDR to be able to detect some or all of these cells (in my case they were 420, 434, 499). The rest dl_earfcns 1400 and 1499 are not used by any operator, so no cells are expected there.

**_Actual behavior:_** no cell were detected and on the second pass through the dl_earfcn list I started getting error 
[ERROR] SetNCOFrequency(index = 0) - Frequency(4.95 MHz) out of range [0-3.84) MHz.

**_Experiment 2 ue.conf:_**
- dl_earfcn = 1275,1425,1575
- nof_antennas = 2

**_Expected behavior:_** my personal phone is connected to cells from **_Band 3, dl_earfcn 1425_,**. The other dl_earfcns 1275 and 1575 are used by other operators. So I was expecting the SDR to be able to detect cells on any of the channels.

**_Actual behavior:_** no cells were detected

**_Experiment 3 ue.conf:_**
- dl_earfcn = 1425,1400,1575
- nof_antennas = 2

**_Expected behavior:_** my personal phone is connected to cells from **_Band 3, dl_earfcn 1425_,**. 1575 is used by other operators and 1400 is not used. So I was expecting the SDR to be able to detect cells on 1425 and 1575.

**_Actual behavior:_** cells were detected on 1425, as I would expect. Nothing was detected for 1400, as expected. And nothing was detected for 1575, not as I would expect. And after ~5 seconds I start getting:
[ERROR] SetNCOFrequency(index = 0) - Frequency(15 MHz) out of range [0-7.68) MHz

**Two more observations:**
- if I run the above examples with **nof_antennas = 1**, the result is just as expected - cells are found for channels 1425,1275,1575.
- if I run the above examples with **nof_antennas = 2** and do not use a list of values in dl_earfcn, but set a single channel, like
dl_earfcn=1425 or dl_earfcn=1575 or dl_earfcn=1275, cells are detected correctly.

I use the following command line for the experiments
/srsue ~/.config/srslte/ue.conf --rf.device_args="rxant=LNAH,rxant=LNAH" --rf.rx_gain=40

## Setup Details ##
Hardware:
- LimeSDR-USB v1.4, 
- CPU: Intel(R) Core(TM) i7-4710MQ CPU @ 2.50GHz, 8 cores, 
- 2xLNA amps for the input signal

Software:
- SoapySDR, branch "soapy-sdr-0.7.2", from 12.01.2020
- LimeSuite, branch "stable", v20.07.02
- srsLTE, branch "master", tag "rel_20_10"
- OS Ubuntu 20.04

I am attaching a zip file with the logs and the configuration for the above tests.
[DlEarfcnTests.zip](https://github.com/srsLTE/srsLTE/files/5581984/DlEarfcnTests.zip)

Please share any comments,
Greetings.

andrepuschmann, 2020-11-26:
Thanks for reporting this. It indeed seems to be a bug on our side. We'll investigate.

nedsana, 2020-12-01:
Any news on this one?