srsran/srsRAN_4G #597: simulate multiple UEs to the same ENodeB by zeroMQ

chunxiao369, 2020-11-25:
Hi,

I would like to ask if it is possible to simulate multiple UEs to the same ENodeB by zeroMQ ?

Thanks.

yagoda, 2020-11-25:
Hey,

In order to have multiple UEs attaching with zeromq, you need some broker
to aggregate the signals, we will be adding a broker in the next release.

If you need something more immediate, you can use gnuradio to make a
broker. see this app note :
https://docs.srslte.com/en/latest/app_notes/source/handover/source/index.html
.

Regards,
Justin.
____
Justin Tallon Ph.D.

Software Radio Systems (SRS)
http://www.softwareradiosystems.com

+353-86-067-0753 | justin.tallon@softwareradiosystems.com

closed 2020-12-10 by andrepuschmann

laf0rge, 2024-09-24:
> we will be adding a broker in the next release.

Has that happened?  This was four years ago. Do you have any pointers to said broker for multiple UE behind a single zeromq eNB?