srsran/srsRAN_4G #598: make test fail

fyy919, 2020-11-25:
Issue Description
When building srsLTE, I execute the command make test, but log shows the error message: 
99% tests passed, 1 tests failed out of 791

Label Time Summary:
long      = 134.03 sec (253 tests)
phy       = 134.03 sec (253 tests)
srsenb    =   6.24 sec (5 tests)
srsue     =   1.02 sec (1 test)

Total Test time (real) = 245.71 sec

The following tests FAILED:
	773 - nas_test (SEGFAULT)
Errors while running CTest
Makefile:149: recipe for target 'test' failed
make: *** [test] Error 8

 Setup Details 
Operation System: Ubuntu 16.04 LTS
RF front-end: USRP B210
library and driver versions: UHD 3.15.0

Expected Behavior
make test successful

Actual Behaviour 
~/srsLTE/build$ sudo ./srsue/test/upper/nas_test

17:10:06.715511 [NAS ] [I] New state DEREGISTERED
17:10:06.715604 [NAS ] [I] DL lcid PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
17:10:06.715613 [NAS ] [I] DL lcid Decrypted PDU
             0000: 07 52 01 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 
             0010: 9b 62 5e 10 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 
             0020: 57 92 52 b8 
17:10:06.715614 [NAS ] [I] Received Authentication Request
17:10:06.715616 [NAS ] [I] MCC=61441, MNC=65281
17:10:06.715618 [NAS ] [D] Authentication request RAND
             0000: 0c 63 a8 54 13 e6 a4 ce d9 86 fb e5 ce 9b 62 5e 
17:10:06.715623 [NAS ] [D] Authentication request AUTN
             0000: 67 57 b3 c2 b9 70 90 01 0c 72 8a 67 57 92 52 b8 
17:10:06.715640 [NAS ] [I] Network authentication successful
17:10:06.715642 [NAS ] [D] Not applying security for PDU. No context configured.
17:10:06.715643 [NAS ] [I] Sending Authentication Response
17:10:06.715648 [NAS ] [I] Generated k_asme:
             0000: 8d 42 d1 bb 80 4b b4 53 cc 00 6d ed 4b 82 dd 55 
             0010: de 66 8b 2b 61 99 73 5d 2d ae 72 fa 24 70 97 ed 
17:10:06.715651 [NAS ] [I] DL lcid PDU
             0000: 37 4e fd 57 11 00 07 5d 02 01 02 f0 70 c1 
17:10:06.715654 [NAS ] [I] DL lcid Decrypted PDU
             0000: 37 4e fd 57 11 00 07 5d 02 01 02 f0 70 c1 
17:10:06.715657 [NAS ] [I] Received Security Mode Command ksi: 1, eea: EEA0, eia: 128-EIA2
17:10:06.715663 [NAS ] [I] NAS encryption key - k_nas_enc
             0000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
             0010: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
17:10:06.715667 [NAS ] [I] NAS integrity key - k_nas_int
             0000: ff 79 be dd c2 b9 39 eb 0b 2f cc 30 ef 6d 00 37 
             0010: 5b ae 51 b3 c0 26 2a 09 2e 2f b3 e5 d3 bc 0f 44 
17:10:06.715669 [NAS ] [D] Generating integrity check. integ_algo:2, count_dl:0, lcid:1
17:10:06.715681 [NAS ] [I] Integrity check ok. Local: count=0, Received: count=0  [4e fd 57 11]
17:10:06.715687 [NAS ] [I] Sending Security Mode Complete nas_current_ctxt.tx_count=0, RB=lcid
17:10:06.739535 [NAS ] [I] New state DEREGISTERED
17:10:06.739552 [USIM] [I] Read Home PLMN Id=00101
17:10:06.739571 [NAS ] [E] Empty EIA list. Select at least one EIA algorithm.
17:10:06.739572 [NAS ] [E] Empty EEA list. Select at least one EEA algorithm.
17:10:06.739575 [NAS ] [I] Skip reading context from file.
Found PLMN:  Id=00101, TAC=65535
[... 133 lines cut ...]
17:10:06.782447 [NAS ] [I] Traffic Flow Template: TFT OP code 0x1, Filter list size 1, Parameter list size 0
17:10:06.782449 [NAS ] [D] Not applying security for PDU. No context configured.
17:10:06.782452 [NAS ] [I] Sending Activate Dedicated EPS Bearer context accept (eps_bearer_id=6, proc_id=0)
             0000: 62 00 c6 
17:10:06.782455 [NAS ] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
17:10:06.782456 [NAS ] [I] Integrity check ok. Local: count=0, Received: count=0  [00 00 00 00]
17:10:06.782457 [NAS ] [I] Update local count to received value 0
17:10:06.782461 [NAS ] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
17:10:06.782462 [NAS ] [I] Received Deactivate EPS bearer context request (eps_bearer_id=6, proc_id=0, cause=0x24)
17:10:06.782466 [NAS ] [I] Removed EPS bearer context (eps_bearer_id=6)
17:10:06.782468 [NAS ] [D] Not applying security for PDU. No context configured.
17:10:06.782470 [NAS ] [I] Sending Deactivate EPS Bearer context accept (eps_bearer_id=6, proc_id=0)
             0000: 62 00 ce 
17:10:06.782474 [NAS ] [I] DL lcid PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
17:10:06.782479 [NAS ] [I] Integrity check ok. Local: count=0, Received: count=0  [00 00 00 00]
17:10:06.782480 [NAS ] [I] Update local count to received value 0
17:10:06.782485 [NAS ] [I] DL lcid Decrypted PDU
             0000: 27 00 00 00 00 00 62 00 cd 24 
17:10:06.782486 [NAS ] [I] Received Deactivate EPS bearer context request (eps_bearer_id=6, proc_id=0, cause=0x24)
17:10:06.782486 [NAS ] [E] EPS bearer doesn't exist (eps_bearer_id=6)
nas_test: ../nptl/pthread_mutex_lock.c:81: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.
Aborted (core dumped)

andrepuschmann, 2020-11-26:
Hey, this issue is a race between two threads in the testing code. We've seen this regression internally as well. Could you give https://github.com/srsLTE/srsLTE/tree/nas_test_fix a try to see if that fixes it for you? In any case, the failing test shouldn't impact the NAS functionality at all.

fyy919, 2020-11-26:
Thanks for your reply!
I have solved this issue.
发件人: Andre Puschmann
发送时间: 2020年11月26日 16:54
收件人: srsLTE/srsLTE
抄送: fyy919; Author
主题: Re: [srsLTE/srsLTE] make test fail (#598)

Hey, this issue is a race between two threads in the testing code. We've seen this regression internally as well. Could you give https://github.com/srsLTE/srsLTE/tree/nas_test_fix a try to see if that fixes it for you? In any case, the failing test shouldn't impact the NAS functionality at all.

andrepuschmann, 2020-11-26:
Ok, good. The fix is part of the next release.

closed 2020-11-26 by andrepuschmann