srsran/srsRAN_4G #599: Timer T301 expired: Going to RRC IDLE

pengyuan-zhou, 2020-11-28:
Hi,

I met with this error at the last step of s1ap handover according to the [note](https://docs.srslte.com/en/latest/app_notes/source/handover/source/index.html#id4)
```
Received HO command to target PCell=6, NCC=2
Timer t304 expired: Handover failed
HO unsuccessful
RRC Connection Reestablishment to PCI=6, EARFCN=2850 (Cause: "handoverFailure")
Timer T301 expired: Going to RRC IDLE
RRC IDLE
```
Tried slowly change the cell gain yet didn't solve it.
Appreciate suggestions :)

andrepuschmann, 2020-12-10:
Manual execution of the cell gain changes can be a bit challenging. I'd suggest you reread the appnote and try it again, try multiple times, faster and slower. I am closing the issue for now, please reopen (with more details) if it still doesn't work for you.

closed 2020-12-10 by andrepuschmann

kaposnick, 2021-02-11:
Hi, 

I experience the same error as @pengyuan-zhou. I have modified the python script (exported from GNU Radio Companion script from the appnote), so that every 120 seconds the cell gains of the 2 eNBs are inversed. I did it this way to prevent manual execution. I am using the zmq driver.

I tried to follow strictly the appnotes:
1. Set the gain0 to 0.5 and wait 5s
2. Gradually increase the gain1 by 0.1, waiting 3s each for each iteration, to 0.6
3. Set the gain1 to 1.0 and wait 5s
4. Set the gain0 to 0

However it still fails. @andrepuschmann have you any suggestion regarding?

frankist, 2021-02-11:
Can you share the enb and ue logs?

kaposnick, 2021-02-12:
Hi @frankist, 

I think there is some timing issues but after a lot of executions I found a combination that succeeds.

1. Set the gain0 to 0.5
2. Gradually increase the gain1 by 0.1, waiting 4s for each iteration, to 1.0
3. Set the gai0 to 0.0

For my setup, with 3s of interval it was pretty unstable. Sometimes the handover succeeded, some others not.

frankist, 2021-02-12:
two thing to take into account:
- make sure the eNB receives the measurementReport from the UE (you can tweak the sensitivity of the UE to sending reports by changing the a3 thresholds in the rr.cfg file)
- the eNB has to be able to send back to the UE the HandoverCommand. If the DL channel degrades too quickly the UE won't receive it.

In the upcoming release, we have several optimizations for the handover to get more smooth

kaposnick, 2021-02-12:
The fact is that the Source eNB receives the measurement report, triggers the Handover and finally sends the Handover command to the UE. However, the UE never starts the procedure with the Target eNB. I will try to reproduce this and as soon as I have logs/pcaps I will upload them.

pengyuan-zhou, 2021-02-16:
Hi @frankist , do you mind telling when the next release would be published? 
Thank you :)