srsran/srsRAN_4G #600: srsLTE cracshed.

Cipher731, 2020-12-01:
Running with UHD 4.0 and N310. How to solve it?

--- command='srsue' version=20.10.1 signal=11 date='01/12/2020 03:29:21' ---
        srsue(+0x17f67f) [0x55dd38a1167f]
        /lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f3762b7b210]
        /usr/local/lib/libsrslte_rf.so.0(+0x2e3cd) [0x7f37633153cd]
        /usr/local/lib/libsrslte_rf.so.0(rf_uhd_open_multi+0x94e) [0x7f37632fb9ae]
        /usr/local/lib/libsrslte_rf.so.0(srslte_rf_open_devname+0x6d) [0x7f37632f700d]
        srsue(+0x26deee) [0x55dd38affeee]
        srsue(+0x2708cc) [0x55dd38b028cc]
        srsue(+0x76ead) [0x55dd38908ead]
        srsue(+0x6127c) [0x55dd388f327c]
        /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7f3762b5c0b3]
        srsue(_start+0x2e) [0x55dd388f48fe]

andrepuschmann, 2020-12-01:
Please provide full details, console output, stack trace, etc? Have you compiled from source?

Cipher731, 2020-12-01:
Solved it by falling back to 3.15.0.0

closed 2020-12-01 by Cipher731