srsran/srsRAN_4G #605: High Packet Loss noticed during iperf DL Throughput test

Vipin2021, 2020-12-10:
## Issue Description ##
High Packet Loss noticed during iperf DL Throughput test
Configuration : 15Mhz, FDD, 1 Antenna, Only 1 Serving Cell, No external/internal interference
Step 1: RSRP Level _ -69dBm , no packet loss, DL Throughput 50Mbps on iperf
Step 2 : RSRP Level _ -75dBm, 60% Packet Loss observed, DL Throughput reduced to 16Mbps
Step 3 : RSRP Level_ -82dBm, 80% Packet Loss , DL Throughput reduced to 7 Mbps

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
Network Configuration : 15Mhz, FDD, 1 Antenna, Only 1 Serving Cell, No external/internal interference
Hardware : X300, UBX40, UHD 3.9.7

## Expected Behavior ##
[What you expect to happen]
0% Packet Loss

## Actual Behaviour ##
[What happens instead e.g. error message]
Packet loss is increasing with RSRP reduction

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Attenuate  RSRP and observe impact on packet loss and throughput

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

xavierarteaga, 2020-12-10:
Hi, I would really like to help you but you need to be a little more accurate.

Could you provide the following?
- UE/eNb console trace
- UE/eNb log files with all levels to info
- PCAP if a higher layer could be involved in it

Vipin2021, 2020-12-10:
Hi Xavier,

Thanks a lot for your kind reply.

Please find below links for the UE/PCAP/Measurement Log files.

https://we.tl/t-faEigftWlh

Best Regards,

Vipin

xavierarteaga, 2020-12-10:
First of all, use `rf.freq_offset` to compensate for the -3620 Hz offset the UE receives.

I made a graph Bit rate vs SNR:
[image]

As the signal power drops, the quality decreases, and consequently the bit rate decreases. The BLER is around 10% which I believe is fine. 

I guess you are using iperf, is the DL rate in the srsue console matching the iperf server rate?

Vipin2021, 2020-12-10:
Hi Xavier,

We were using iperf,

Query : is the DL rate in the srsue console matching the iperf server rate? : It was matching.
Will change rf.freq_offset to -3620Hz and keep you posted.

Best Regards,

Vipin

[Screenshot from 2020-12-10 15-49-51]<https://user-images.githubusercontent.com/29545975/101787629-940ec700-3aff-11eb-84cc-ec1058a581c5.png>

Vipin2021, 2020-12-11:
Hi Xavier,

Tried with Freq Offset -3620Hz but issue remains same, please need your kind support over here.

Br,

Vipin

xavierarteaga, 2020-12-11:
The DL SNR Vs bit rate makes sense to me.

The RSRP looks high for such low SNR. Is the parameter `rx_gain_offset` set correctly? Did you use any calibrated equipment?

Is the AGC enabled? You could try tunning the Rx gain manually to see what is the best performance.

Is the X300 calibrated correctly?

Vipin2021, 2020-12-11:
Hi Xavier,

Did you use any calibrated equipment? : No, we use another LTE UE Measurement to calibrate rx_gain_offset.

Measured RSRP on another UE and tried to match with srsUE.

AGC Enabled : Can you please share more details about it ?

Is the X300 calibrated correctly ? : No such calibration done from our side.

You could try tunning the Rx gain manually to see what is the best performance : We will try and let you know.

Vipin2021, 2020-12-11:
Hi Xavier,

Tried to tune “Rx Gain” but it didn’t impact RSRP effectively and couldn’t reach desired RSRP value in any case.

RSRP
Rx Gain Offset
Rx Gain
-94
95
38
-61
62
38
-61
62
71
-51
62
5
-61
62
200
-61
62
400