srsran/srsRAN_4G #606: srsenb can not decode S1AP PAGING

t1412, 2020-12-10:
## Issue Description ##
[Describe the issue in detail]
We connect srsenb with REAL MME. 
MME send S1AP PAGING with IMSI. After received this message, srsenb can not decode it.
enb.log
09:56:56.115863 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/asn1_utils.cc][1023] Decoding failure.
09:56:56.115865 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][32924] Decoding failure.
09:56:56.115882 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][37408] Decoding failure.
09:56:56.115884 [ASN1] [E] This method only supports unpacking up to 32 bits
09:56:56.115885 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][336] Decoding failure.
09:56:56.115887 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][50173] Decoding failure.
09:56:56.115889 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][50346] Decoding failure.
09:56:56.115891 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][56679] Decoding failure.
09:56:56.115893 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][57796] Decoding failure.
09:56:56.115894 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/s1ap_asn1.cc][58730] Decoding failure.
09:56:56.115924 [S1AP] [E] Failed to unpack received PDU

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2020-12-10:
Hey, thanks for reporting this Please send a S1AP PCAP with the message.

t1412, 2020-12-10:
[enb_s1ap.zip](https://github.com/srsLTE/srsLTE/files/5672295/enb_s1ap.zip)

andrepuschmann, 2020-12-10:
Thanks, we'll be looking into it.

frankist, 2020-12-11:
I confirm the issue with handling constrained-size octet strings. I provide a fix in the branch `pr_asn1_fix`. Let me know if it solves the problem for you.

t1412, 2020-12-14:
Thanks, I'll verify it soon

t1412, 2020-12-15:
Hi @frankist , 

I verified that fixed code is done for decoding, but i got issue when encoding paging message which is sent to UE. 

10:51:27.805458 [RRC ] [I] Assembled paging for ue_id=723, tti=199 
10:51:27.805465 [ASN1] [E] The condition lb <= n <= ub (0 <= 84 <= 9) was not met 
10:51:27.805468 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/include/srslte/asn1/asn1_utils.h][981] Encoding failure. 
10:51:27.805470 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][78920] Encoding failure. 
10:51:27.805525 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][79003] Encoding failure. 
10:51:27.805529 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/include/srslte/asn1/asn1_utils.h][991] Encoding failure. 
10:51:27.805531 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][79076] Encoding failure. 
10:51:27.805533 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][79242] Encoding failure. 
10:51:27.805534 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][79205] Encoding failure. 
10:51:27.805536 [ASN1] [E] [/home/tml/workspace/srsLTE/lib/src/asn1/rrc_asn1.cc][79254] Encoding failure. 
10:51:27.805538 [RRC ] [E] Failed to pack PCCH 

Please help to Please help to resolve. 
Content of S1AP PAGING is same as attached pcap.

t1412, 2020-12-15:
For more information:

In add_paging_id function: 
Test with IMSI = 452040038647507 
Current code: 
if (ue_paging_id.type().value == asn1::s1ap::ue_paging_id_c::types_opts::imsi) { 
paging_elem.ue_id.set_imsi(); 
paging_elem.ue_id.imsi().resize(ue_paging_id.imsi().size()); 
memcpy(paging_elem.ue_id.imsi().data(), ue_paging_id.imsi().data(), ue_paging_id.imsi().size()); 
rrc_log->console("Warning IMSI paging not tested\n"); 

With IMSI octet receive from MME: 
ue_paging_id.imsi().data() = 54 02 04 30 68 74 05 f7 
ue_paging_id.imsi().size() = 8 
=> Encoding fail 

When I change code (test purpose) to: 
if (ue_paging_id.type().value == asn1::s1ap::ue_paging_id_c::types_opts::imsi) { 
paging_elem.ue_id.set_imsi(); 
uint8_t test_data[15] = {4,5,2,0,4,0,0,3,8,6,4,7,5,0,7}; 
paging_elem.ue_id.imsi().resize(15); 
memcpy(paging_elem.ue_id.imsi().data(), test_data, 15); 
rrc_log->console("Warning IMSI paging not tested\n"); 
=> Endcoding Success 
11:15:47.507577 [RRC ] [I] Assembled paging for ue_id=723, tti=199 
11:15:47.507593 [RRC ] [I] Assembling PCCH payload with 1 UE identities, payload_len=10 bytes, nbits=80 
11:15:47.507644 [RRC ] [D] PCCH-Message - Tx paging (10 B) 
0000: 40 19 45 20 40 03 86 47 50 70 
11:15:47.507667 [RRC ] [D] Content: 
[ 
{ 
"PCCH-Message": { 
"message": { 
"c1": { 
"paging": { 
"pagingRecordList": [ 
{ 
"ue-Identity": { 
"imsi": [ 
4, 
5, 
2, 
0, 
4, 
0, 
0, 
3, 
8, 
6, 
4, 
7, 
5, 
0, 
7 
] 
}, 
"cn-Domain": "ps" 
} 
] 
} 
} 
} 
} 
} 
]

mynameismikemike, 2021-05-09:
@frankist I think I am seeing a similar issue with the S1SetupResponse even using the pr_asn1_fix branch :/

---cut--
00:50:14.509585 [S1AP] [I] Received S1AP PDU
             0000: 20 11 00 24 00 00 03 00 3d 40 09 80 07 6d 6d 65
             0010: 5f 73 76 63 00 69 00 0b 00 00 00 f1 10 00 00 0b
00:50:14.509619 [ASN1] [E] Extension of PrintableString not supported
00:50:14.509628 [PHY0] [D] [ 2792] Setting TTI=2794, tx_mutex=2, tx_time=4:0.298741 to worker 0
00:50:14.509632 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][40908] Decoding failure.
00:50:14.509636 [ASN1] [E] This method only supports unpacking up to 32 bits
00:50:14.509642 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][336] Decoding failure.
00:50:14.509661 [PHY0] [D] [ 2794] Worker 0 running
00:50:14.509669 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][51713] Decoding failure.
00:50:14.509675 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][51810] Decoding failure.
00:50:14.509678 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][57374] Decoding failure.
00:50:14.509683 [ASN1] [E] This method only supports unpacking up to 32 bits
00:50:14.509688 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][58539] Decoding failure.
00:50:14.509703 [ASN1] [E] [/home/mike/srs_fix/srsRAN/lib/src/asn1/s1ap_asn1.cc][58732] Decoding failure.
00:50:14.509706 [S1AP] [E] Failed to unpack received PDU
---cut---

Thoughts? (sorry, I'm new to github)

frankist, 2021-05-27:
Hey, have you tried the newer release?

mynameismikemike, 2021-05-27:
Thank you for your response @frankist.  I just attempted with Version 21.4.0:

---cut---
20:33:39.269675 [ENB    ] [I] Using binary srsenb with arguments:
20:33:39.269870 [ENB    ] [I] Built in Release mode using commit 89f16eed2 on branch master.
20:33:41.838901 [COMN   ] [D] [    0] Setting RTO_INFO options on SCTP socket. Association 0, Initial RTO 3000, Minimum RTO 1000, Maximum RTO 6000
20:33:41.838915 [COMN   ] [D] [    0] Setting SCTP_INITMSG options on SCTP socket. Max attempts 3, Max init attempts timeout 5000
20:33:41.899003 [COMN   ] [D] [    0] RxSockets: socket fd=15 has been registered.
20:33:41.899135 [COMN   ] [D] [    0] RxSockets: socket fd=16 has been registered.
20:33:41.902994 [ASN1   ] [E] [    0] Extension of PrintableString not supported
20:33:41.903007 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][47482] Decoding failure.
20:33:41.903009 [ASN1   ] [E] [    0] This method only supports unpacking up to 32 bits
20:33:41.903010 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][200] Decoding failure.
20:33:41.903010 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][59396] Decoding failure.
20:33:41.903011 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][59493] Decoding failure.
20:33:41.903012 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][65396] Decoding failure.
20:33:41.903012 [ASN1   ] [E] [    0] This method only supports unpacking up to 32 bits
20:33:41.903013 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][66623] Decoding failure.
20:33:41.903014 [ASN1   ] [E] [    0] [/home/mike/srsRAN_latest/srsRAN/lib/src/asn1/s1ap.cc][66831] Decoding failure.
20:33:41.903015 [S1AP   ] [E] Failed to unpack received PDU
    0000: 20 11 00 24 00 00 03 00 3d 40 09 80 07 6d 6d 65
    0010: 5f 73 76 63 00 69 00 0b 00 00 00 f1 10 00 00 0b
20:33:47.063625 [S1AP   ] [E] Proc "MME Connection" - S1Setup failed.
---cut---

Attached is the 'enb_s1ap.pcap'.
[enb_s1ap.pcap.txt](https://github.com/srsran/srsRAN/files/6556720/enb_s1ap.pcap.txt)

Thoughts?

Thanks again!

sareek, 2022-10-16:
Have anybody resolved this issue yet. Coz I am also facing something similar,

[image]

andrepuschmann, 2022-10-16:
> Have anybody resolved this issue yet. Coz I am also facing something similar,
> 
> [image]

This is just an unfinished S1 Setup procedure. Check your IPs are right. It has nothing to do with the S1AP paging issue.

@mynameismikemike 
Just reviewing the thread an it seems there are no new PCAP captures after the various ASN1 fixes have been merged. 

> `20:33:41.903015 [S1AP ] [E] Failed to unpack received PDU`
> `0000: 20 11 00 24 00 00 03 00 3d 40 09 80 07 6d 6d 65`
> `0010: 5f 73 76 63 00 69 00 0b 00 00 00 f1 10 00 00 0b`

The issue here is that the logged message is not complete. Its cut after 32 Bytes. 

Since the issue is already quite old I am going to close it assuming that it's not present in the newest ASN1 unpacking code. If it does still exist, please feel free to reopen again and provide full logs or PCAPs.

Thanks

closed 2022-10-16 by andrepuschmann

sareek, 2022-10-17:
> unfinished S1 Setup procedure. Check your IPs are right

@andrepuschmann i believe i have set ip addresses correctly. Also, epc log says sending S1 setup response but enb is not able to connect. 
[image]

Where do i need to make change exactly?
epc config
[image]

enb config
[image]

Thanks

bcgordon3, 2024-04-22:
> > unfinished S1 Setup procedure. Check your IPs are right

Just throwing this out here -- srs does not like an unnamed MME. So if your s1response includes a blank mme name, the s1 setup will fail. The way to fix this is in your EPC, make sure your MME have a name.