srsran/srsRAN_4G #607: Low O/P Tx Power with USRP X300 + UBX40

Vipin2021, 2020-12-11:
## Issue Description ##
[Describe the issue in detail]
As per Ettus, with UBX40 and X300 Tx Output Power should be +20dBm but in our case it is not exceeding more than 10dBm in any case.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
N/w Configuration : 1 Cell
Hardware : USRP X300 + UBX40
UHD :3.9.7

## Expected Behavior ##
[What you expect to happen]
+20dBm Output Power

## Actual Behaviour ##
[What happens instead e.g. error message]
+10dBm Output Power

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Check O/P with RF Power meter and keep on changing your Tx Gain

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

xavierarteaga, 2020-12-11:
1. As UL transmission might be discontinuous, you need to use a vector analyzer for measuring PUCCH/PUSCH peak power and density.
2. The maximum Tx power is the Maximum Tx USRP power minus the crest factor of LTE UL (~7dB)
3. To transmit maximum power use `phy.force_ul_amplitude=0.9`.

Vipin2021, 2020-12-11:
Hi Xavier,
Thanks a lot, please can you let us know where and how to change phy.force_ul_amplitude=0.9.
Best Regards,

Vipin2021, 2020-12-22:
Hi Xavier,

After tuning  force_ul_amplitude=0.9, could able to increase Tx Power from 11dBm to 14dBm.

Thanks a lot...

Please we need to pull it up till +20dBm with external PA.

closed 2023-04-14 by ismagom