srsran/srsRAN_4G #610: Please wants to tune parameter “phy.cfo_is_doppler” to True, can you please suggest where can I get the same ?

Vipin2021, 2021-01-10:
## Issue Description ##
[Describe the issue in detail]
Please wants to tune parameter “phy.cfo_is_doppler” to True, can you please suggest where can I get the same ?
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
X300 + UBX40+ srsLTE 3.9.7 SW
## Expected Behavior ##
[What you expect to happen]
For High Doppler
## Actual Behaviour ##
[What happens instead e.g. error message]
Couldn't find parameter in UE Conf File
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Need to know location of parameter “phy.cfo_is_doppler” to tune 
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2021-01-11:
Hey, just pass it as UE command line parameter `--phy.cfo_is_doppler true`.

Vipin2021, 2021-01-11:
Hi,

After changing this parameter to true UE couldn’t attach.

Best Regards,

ismagom, 2021-01-11:
Please use Github to report issues and our mailing list to ask questions. 

If you are trying to report an issue, follow the template and describe your setup: what eNB you are using and what version, what UE and version, HW in UE and eNB, carrier and doppler frequency, wired/wireless, share configuration files for both eNB and UEs, share also log files, stdout console logs etc 

Otherwise it's impossible to help. Thanks a lot.