srsran/srsRAN_4G #615: Only first 8 bits of TAC passed by the ENB to the EPC

frederickpelchat, 2021-01-23:
## Issue Description ##
The issue is well explained in the following sections as well as the solution.

## Expected Behavior ##
Using whole 16 bits of the TAC number as defined in the specification:

IE/Group 			Name | Presence | Range | IE 			type and reference | Semantics 			description
-- | -- | -- | -- | --
TAC | M |   | OCTET 			STRING (SIZE (2)) |  

## Actual Behaviour ##
Due to a uint8_t cast on the assigned TAC number while looping on every TAC numbers in the S1 request handler, only the lowest 8 bits of the TAC number is used in the comparison. In the default config 0x0007 is defined as the default TAC so the bug is not triggered but when setting the highest part of the TAC the TAC match fails.

in ./srsLTE/srsepc/src/mme/s1ap_mngmt_proc.cc

in s1ap_mngmt_proc::handle_s1_setup_request

  // Check for TAC match
  bool tac_match = false;
  for (uint8_t tac : enb_ctx.tacs) {
    if (m_s1ap->get_tac() == tac) {
      tac_match = true;
      break;
    }
  }

## Steps to reproduce the problem ##
Define highest part of the TAC in rr.conf and launch both the ENB and the EPC. The S1 request will fail saying the TAC does not match.

andrepuschmann, 2021-01-31:
Thanks for reporting the issue, Frederick,

andrepuschmann, 2021-04-23:
Closing this. Fix is merged

closed 2021-04-23 by andrepuschmann