srsran/srsRAN_4G #618: Unit Test and E2E ZMQ Failure when built with AUTO_DETECT_ISA=OFF

beaverandbuffcode, 2021-01-29:
When SRSLTE is configured using -DAUTO_DETECT_ISA=OFF, make test fails most PHY layer unit tests.

Additionally, using end to end LTE network testing using ZMQ driver fails, displaying failure to bring up UE network device.

On examination, failure occurs between srsenb and srsue, as attachment is never successfully made to bring up network.

Errors verified on x86_64 and aarch64; in same environment as failures demonstrated, rebuilding with AUTO_DETECT_ISA=ON both with and without DISABLE_SIMD produces all unit tests passing and successful end to end ZMQ tests.

andrepuschmann, 2021-01-31:
Hey,
the `AUTO_DETECT_ISA` and `DISABLE_SIMD` are quite advanced options and depending on your system architecture they may or may not make sense. We generally do not recommend changing them unless you know what you're doing.

In some cases they are known to not work at all because some parts of the code (e.g. Turbo decoder) are only available in a SIMD-optimized version and won't even run without it, hence all tests failing. For me, running make test after `cmake -DAUTO_DETECT_ISA=False -DDISABLE_SIMD=True ..` fails all PDSCH/PUSCH tests because of that. But that's actually expected.

For your particular case, could you please share detailed CMake/compile output and also the result of `cat /proc/cpuinfo`? Basically follow the issue template and fill all required fields.

Cheers
Andre

closed 2023-04-14 by ismagom

mrahman2-vt, 2023-04-14:
Hey
Could you please remove me from your mailing list?