srsran/srsRAN_4G #622: Sync issue when using USRP X310 with two antennas, one antenna has no such issue.

Vondark, 2021-02-04:
## Issue Description ##
I'm using srsLTE + USRP x310 and b210 to receive LTE FDD DL CSI information. If I use b210 two antennas or X310 one antenna, it works well. If I use x310 two antennas, the output will always be 0+0i. I tried to debug the codes, and I ended up here: https://github.com/srsLTE/srsLTE/blob/45486b6e2c622ab1d47906521662f087816d7d03/lib/src/phy/sync/sync.c#L563
I found that with b210 two antennas or x310 one antenna, the sf_idx is 0,1,2,...,9 in turns, while with X310 two antennas it is always ZERO. I think there must be some bug with the multi-antennas configuration of x310. Could you help me with that? 

-- This is my first time to propose an issue. Sorry for any misbehavior. 

## Setup Details ##
Network configuration: [Intel X520-DA2](http://ark.intel.com/products/39776/Intel-Ethernet-Converged-Network-Adapter-X520-DA2)

Operation System: Ubuntu 16.04.7 LTS

RF front-end: USRP x310 (And USRP 2954r, the same issue)

library and driver versions: I think it is the latest version of srsLTE( Install using git clone https://github.com/srsLTE/srsLTE.git a few days ago).  GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.15.0.0-release. 

## Expected Behavior ##
The [sf_idx](https://github.com/srsLTE/srsLTE/blob/45486b6e2c622ab1d47906521662f087816d7d03/lib/src/phy/sync/sync.c#L563) is 0,1,2,...,9 in turns when using USRP X310 two antennas. 

## Actual Behavior ##
The [sf_idx](https://github.com/srsLTE/srsLTE/blob/45486b6e2c622ab1d47906521662f087816d7d03/lib/src/phy/sync/sync.c#L563) is always ZERO when using USRP X310 two antennas. But it's 0,1,2,...,9 in turns when using USRP B210 two antennas or X310 one antenna. 

## Steps to reproduce the problem ##
RF setup : USRP X310 connect to a PC with Ubuntu 16.04, set the config to two antennas. 

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2021-02-08:
Thanks for reporting the issue. We'll look into it.

Vondark, 2021-03-24:
I found that the cause of this issue is UHD 3.15. When I was testing using benchmark_rate (tuned and optimized as [this](https://kb.ettus.com/USRP_Host_Performance_Tuning_Tips_and_Tricks) and [this](https://kb.ettus.com/Using_Dual_10_Gigabit_Ethernet_on_the_USRP_X300/X310) ), if I use one channel, 200MSps works perfect. But if I use two channels, even 20MSps can not be held. 

I have not tried UHD 4.0.0, Ubuntu 20.04 and dpdk, and I think they may improve the performance, especially dpdk.

closed 2021-03-24 by Vondark