srsran/srsRAN_4G #623: srsLTE crashed

avalori1, 2021-02-05:
srsLTE crashes after fairly long time running OK

## Issue Description ##
crash
## Setup Details ##
~/srsLTE/srsepc$ srsenb --version
--- Software Radio Systems LTE eNodeB ---
Version 20.10.1

But strangely, can't get the version of srsepc?

~/srsLTE/srsepc$ srsepc --version
Built in Release mode using commit 45486b6 on branch master.
--- Software Radio Systems EPC ---
Reading configuration file /home/bts/.config/srslte/epc.conf...

## Expected Behavior ##
keep running
## Actual Behaviour ##
crash

## Steps to reproduce the problem ##
Just wait, runs for a fairly long time but then crash

## Additional Information ##
srsLTE.backtrace.crash

--- command='srsenb' version=20.10.1 signal=11 date='05/02/2021 08:03:33' ---
	srsenb(+0x17bc33) [0x55cc1d3eac33]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7f4138143040]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZSt18_Rb_tree_incrementPSt18_Rb_tree_node_base+0x13) [0x7f4138b54fa3]
	srsenb(+0x186d11) [0x55cc1d3f5d11]
	srsenb(+0x5ea1e) [0x55cc1d2cda1e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f4139deb6db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f413822571f]

--- command='srsenb' version=20.10.1 signal=6 date='05/02/2021 08:03:34' ---
	srsenb(+0x17bc33) [0x55cc1d3eac33]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7f4138143040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7f4138142fb7]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7f4138144921]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8c957) [0x7f4138b37957]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ae6) [0x7f4138b3dae6]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92b21) [0x7f4138b3db21]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x938ff) [0x7f4138b3e8ff]
	srsenb(+0xd6e43) [0x55cc1d345e43]
	srsenb(+0xcc70f) [0x55cc1d33b70f]
	srsenb(+0xc7bf2) [0x55cc1d336bf2]
	srsenb(+0xc7de0) [0x55cc1d336de0]
	srsenb(+0xb76b1) [0x55cc1d3266b1]
	srsenb(+0x826f5) [0x55cc1d2f16f5]
	srsenb(+0x189ba1) [0x55cc1d3f8ba1]
	srsenb(+0x5ea1e) [0x55cc1d2cda1e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f4139deb6db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f413822571f]

--- command='srsenb' version=20.10.1 signal=11 date='05/02/2021 12:51:47' ---
	srsenb(+0x17bc33) [0x56300b261c33]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7f8e47ab7040]
	srsenb(+0xb8868) [0x56300b19e868]
	srsenb(+0x8274c) [0x56300b16874c]
	srsenb(+0x189ba1) [0x56300b26fba1]
	srsenb(+0x5ea1e) [0x56300b144a1e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f8e4975f6db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f8e47b9971f]

avalori1, 2021-02-08:
> Thanks for reporting the issue. Unfortunately, without debug symbols it's
> hard to tell what's going on. Could you build with debug symbols perhaps
> and run again. If it crashes, please share the backtrace.

Dear Andre,
sorry for my ignorance... a novice on this. How can I do that?
Is an option to be added at the make step in this instructions?
https://docs.srslte.com/en/latest/general/source/1_installation.html

Thanks

avalori1, 2021-02-11:
@andrepuschmann would be possible for you to point me in the direction of instruction to do so, so I can collect some more information?

Thanks

andrepuschmann, 2021-02-11:
We work on instructions for that. I hope to get something out shortly.
Quick tip is to use `cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo ..`. Clean up the build folder before that and run a clean build. Then when it crashes, post the `./srsLTE_backtrace.log` file

avalori1, 2021-02-11:
Thanks for your tip!I think I managed to build as instructed. It seems that there is a bit more info coming up but could not see the `./srsLTE_backtrace.log` file

The info I got are:
In the terminal

`RACH:  tti=5331, cc=0, preamble=3, offset=0, temp_crnti=0x46
User 0x46 connected
terminate called after throwing an instance of 'std::length_error'
  what():  basic_string::_M_create
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---`

And in the `srsLTE.backtrace.crash` file
`--- command='srsenb' version=20.10.1 signal=6 date='11/02/2021 21:27:46' ---
	srsenb(+0x162327) [0x55f0252d4327]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7f2c8f8e5040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7f2c8f8e4fb7]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7f2c8f8e6921]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8c957) [0x7f2c902d9957]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ae6) [0x7f2c902dfae6]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92b21) [0x7f2c902dfb21]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92d54) [0x7f2c902dfd54]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8e811) [0x7f2c902db811]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x124799) [0x7f2c90371799]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_mutateEmmPKcm+0x6b) [0x7f2c90371b8b]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm+0x1cb) [0x7f2c9037294b]
	/usr/local/lib/libsrslte_rf.so.0(+0x29f12) [0x7f2c90a0cf12]
	/usr/local/lib/libsrslte_rf.so.0(rf_uhd_send_timed_multi+0x328) [0x7f2c909f8998]
	/usr/local/lib/libsrslte_rf.so.0(srslte_rf_send_timed_multi+0x29) [0x7f2c909f6ab9]
	srsenb(+0x245678) [0x55f0253b7678]
	srsenb(+0x2479f1) [0x55f0253b99f1]
	srsenb(+0x76ae0) [0x55f0251e8ae0]
	srsenb(+0x7f5e6) [0x55f0251f15e6]
	srsenb(+0x16e389) [0x55f0252e0389]
	srsenb(+0x5f59e) [0x55f0251d159e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f2c918216db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f2c8f9c771f]`

avalori1, 2021-02-13:
Dear @andrepuschmann

Please find another example of crash... If you have recommendations on how to gather more useful data, I will be happy to do so.
Not sure if this is of any usefulness, but I noticed that in the terminal I see twice the crash message...
```
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
--- exiting ---
--- exiting ---

```

In the `srsLTE.backtrace.crash` file
```
--- command='srsenb' version=20.10.1 signal=11 date='13/02/2021 16:15:04' ---
	srsenb(+0x162327) [0x561ed2eb6327]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7f2f2123e040]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_submit_transfer+0x1c5) [0x7f2f1b330a95]
	/usr/local/lib/libuhd.so.3.15.0(+0x86922c) [0x7f2f1f29122c]
	/usr/local/lib/libuhd.so.3.15.0(+0x869548) [0x7f2f1f291548]
	/usr/local/lib/libuhd.so.3.15.0(+0x868515) [0x7f2f1f290515]
	/usr/local/lib/libuhd.so.3.15.0(+0x5ad1f3) [0x7f2f1efd51f3]
	/usr/local/lib/libsrslte_rf.so.0(+0x29eee) [0x7f2f22365eee]
	/usr/local/lib/libsrslte_rf.so.0(rf_uhd_send_timed_multi+0x328) [0x7f2f22351998]
	/usr/local/lib/libsrslte_rf.so.0(srslte_rf_send_timed_multi+0x29) [0x7f2f2234fab9]
	srsenb(+0x245678) [0x561ed2f99678]
	srsenb(+0x2479f1) [0x561ed2f9b9f1]
	srsenb(+0x76ae0) [0x561ed2dcaae0]
	srsenb(+0x7f5e6) [0x561ed2dd35e6]
	srsenb(+0x16e389) [0x561ed2ec2389]
	srsenb(+0x5f59e) [0x561ed2db359e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7f2f2317a6db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7f2f2132071f]
```

andrepuschmann, 2021-02-15:
The crash seems to come from the PHY layer interacting with the radio. But it's hard to tell how from the logs/info we have, the backtrace isn't very useful either.
Could you perhaps try the following:
`cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_ASAN=True ..`, again wipe the build folder and compile again. Please send stdout when you run `cmake`. This command enables AddressSanitizer. This will slow down the execution and maybe the eNB doesn't work anymore. But you said the eNB crashed eventually without doing stuff, right? If so, just leave it running like before and see what happens. Thanks

avalori1, 2021-02-15:
Andre,
The one below is the output while running 'cmake' using the option you emailed me. Since it always takes a bit of time to have a crash happening, I send this to you in the meantime, and the result of the crash later...

I am not sure what you mean with "But you said the eNB crashed eventually without doing stuff, right?"... I mean, there was always a UE connected doing some traffic... I did not try to leave it running with no UE connected... anyway, will try to start it and update you.

'''
bts@bts-ThinkPad-L460:~/srsLTE/build$ cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_ASAN=True ..
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_SYSTEM: Linux-5.4.0-65-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Enabling ccache for C
-- Enabling ccache for CXX
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.7
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- Found fftw3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- Found MBEDTLS: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so  
-- Checking for module 'libpcsclite'
--   Found libpcsclite, version 1.8.23
-- Found PCSCLITE: /usr/lib/x86_64-linux-gnu/libpcsclite.so  
-- PCSC LIBRARIES: /usr/lib/x86_64-linux-gnu/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
-- UHD LIBRARIES /usr/local/lib/libuhd.so
-- UHD INCLUDE DIRS /usr/local/include
-- Found UHD: /usr/local/lib/libuhd.so  
-- Performing Test UHD_ENABLE_X300_FW_RESET
-- Performing Test UHD_ENABLE_X300_FW_RESET - Failed
-- Performing Test UHD_ENABLE_RFNOC
-- Performing Test UHD_ENABLE_RFNOC - Success
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC
-- Performing Test UHD_ENABLE_CUSTOM_RFNOC - Failed
[... 58 lines cut ...]
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-EPC command defined
-- Configuring done
-- Generating done
-- Build files have been written to: /home/bts/srsLTE/build
'''

avalori1, 2021-02-15:
@andrepuschmann 

the eNB crashed (fairly quickly) and this time indeed spitted out a lot of info!
I copied as HTML to keep the formatting (although the colour coding was lost).

Regards

<pre> 264 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 265 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 266 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 281 0.10   0 2.0     0    0   89 100%    0  0.0   0     0    0    0   0%   0.0
 283 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 285 3.0   0 0.0     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 286 10.0   0 2.7     0    0   97 100%    0  0.0   0     0    0    0   0%   0.0
 391 0.10   0 2.0     0    0   89 100%    0  0.0   0     0    0    0   0%   0.0
 3b1  11   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 494  12   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
terminate called after throwing an instance of &apos;std::length_error&apos;
  what():  basic_string::_M_create
srsLTE crashed... backtrace saved in &apos;./srsLTE.backtrace.crash&apos;...
---  exiting  ---
=================================================================
<font color="#EF2929"><b>==26751==ERROR: AddressSanitizer: heap-use-after-free on address 0x612000000a20 at pc 0x558355248a37 bp 0x7f4ccc0527a0 sp 0x7f4ccc052790</b></font>
<font color="#729FCF"><b>READ of size 1 at 0x612000000a20 thread T25</b></font>
    #0 0x558355248a36 in srslog::detail::shared_variable&lt;bool&gt;::operator bool() const /home/bts/srsLTE/lib/include/srslte/srslog/detail/support/thread_utils.h:99
    #1 0x558355248a36 in srslog::log_channel::enabled() const /home/bts/srsLTE/lib/include/srslte/srslog/log_channel.h:87
    #2 0x558355248a36 in void srslog::log_channel::operator()&lt;char*&gt;(std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const&amp;, char*&amp;&amp;) /home/bts/srsLTE/lib/include/srslte/srslog/log_channel.h:104
    #3 0x558355248a36 in srslte::srslog_wrapper::log(std::unique_ptr&lt;srslte::logger::log_str, srslte::logger::log_str_deleter&gt;) /home/bts/srsLTE/lib/src/common/logger_srslog_wrapper.cc:29
    #4 0x558355236fdb in srslte::log_filter::all_log(srslte::LOG_LEVEL_ENUM, unsigned int, char*, unsigned char const*, int, bool) /home/bts/srsLTE/lib/src/common/log_filter.cc:138
    #5 0x55835523cf8a in srslte::log_filter::info(char const*, ...) /home/bts/srsLTE/lib/src/common/log_filter.cc:181
    #6 0x5583554ffeb1 in srslte::radio::handle_rf_msg(srslte_rf_error_t) /home/bts/srsLTE/lib/src/radio/radio.cc:957
    #7 0x7f4cddb2ac68 in log_underflow /home/bts/srsLTE/lib/src/phy/rf/rf_uhd_imp.cc:220
    #8 0x7f4cddb2ac68 in async_thread /home/bts/srsLTE/lib/src/phy/rf/rf_uhd_imp.cc:263
    #9 0x7f4cddb3c315 in void* std::__invoke_impl&lt;void*, void* (*)(void*), rf_uhd_handler_t*&gt;(std::__invoke_other, void* (*&amp;&amp;)(void*), rf_uhd_handler_t*&amp;&amp;) /usr/include/c++/7/bits/invoke.h:60
    #10 0x7f4cddb3c315 in std::__invoke_result&lt;void* (*)(void*), rf_uhd_handler_t*&gt;::type std::__invoke&lt;void* (*)(void*), rf_uhd_handler_t*&gt;(void* (*&amp;&amp;)(void*), rf_uhd_handler_t*&amp;&amp;) /usr/include/c++/7/bits/invoke.h:96
    #11 0x7f4cddb3c315 in decltype (__invoke((_S_declval&lt;0ul&gt;)(), (_S_declval&lt;1ul&gt;)())) std::thread::_Invoker&lt;std::tuple&lt;void* (*)(void*), rf_uhd_handler_t*&gt; &gt;::_M_invoke&lt;0ul, 1ul&gt;(std::_Index_tuple&lt;0ul, 1ul&gt;) /usr/include/c++/7/thread:234
    #12 0x7f4cddb3c315 in std::thread::_Invoker&lt;std::tuple&lt;void* (*)(void*), rf_uhd_handler_t*&gt; &gt;::operator()() /usr/include/c++/7/thread:243
    #13 0x7f4cddb3c315 in std::thread::_State_impl&lt;std::thread::_Invoker&lt;std::tuple&lt;void* (*)(void*), rf_uhd_handler_t*&gt; &gt; &gt;::_M_run() /usr/include/c++/7/thread:186
    #14 0x7f4cdd42f6de  (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xbd6de)
    #15 0x7f4cde9fa6da in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76da)
    #16 0x7f4cdcaec71e in __clone (/lib/x86_64-linux-gnu/libc.so.6+0x12171e)

<font color="#8AE234"><b>0x612000000a20 is located 224 bytes inside of 280-byte region [0x612000000940,0x612000000a58)</b></font>
<font color="#AD7FA8"><b>freed by thread T26 (WORKER0) here:</b></font>
    #0 0x7f4cdecf32c0 in operator delete(void*) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe12c0)
    #1 0x5583551dd529 in __gnu_cxx::new_allocator&lt;std::__detail::_Hash_node&lt;std::pair&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const, srslog::log_channel&gt;, true&gt; &gt;::deallocate(std::__detail::_Hash_node&lt;std::pair&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const, srslog::log_channel&gt;, true&gt;*, unsigned long) /usr/include/c++/7/ext/new_allocator.h:125
    #2 0x5583551dd529 in std::allocator_traits&lt;std::allocator&lt;std::__detail::_Hash_node&lt;std::pair&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const, srslog::log_channel&gt;, true&gt; &gt; &gt;::deallocate(std::allocator&lt;std::__detail::_Hash_node&lt;std::pair&lt;std::__cxx11::basic_string&lt;char, std::char_traits&lt;char&gt;, std::allocator&lt;char&gt; &gt; const, srslog::log_channel&gt;, true&gt; &gt;&amp;, std::__detail::_Hash_node&lt;std::pair&lt;std::__cxx11::basi [... cut ...]
  Container overflow:      <font color="#729FCF"><b>fc</b></font>
  Array cookie:            <font color="#EF2929"><b>ac</b></font>
  Intra object redzone:    <font color="#FCE94F"><b>bb</b></font>
  ASan internal:           <font color="#FCE94F"><b>fe</b></font>
  Left alloca redzone:     <font color="#729FCF"><b>ca</b></font>
  Right alloca redzone:    <font color="#729FCF"><b>cb</b></font>
==26751==ABORTING
<font color="#8AE234"><b>bts@bts-ThinkPad-L460</b></font>:<font color="#729FCF"><b>~/srsLTE/build</b></font>$ 
</pre>

andrepuschmann, 2021-02-15:
Thanks for the logs. I have the feeling that there is something else going wrong before the issue that ASAN reports. I think we had a similar problem before internally and fixed that in our tree already. But it would be good to double check and backport (if that's the case) to 20.10. For that I think it would be best is if we could reproduce the issue ourselves. What exactly are you doing before the crash happens? I see the RNTIs of the UEs is quite high, are you attaching many UEs? What system is this running on? Can you please provide the info that we've asked for in the issue template? Thanks

andrepuschmann, 2021-02-15:
A bit of a shot in the blue, but could you perhaps give https://github.com/srsLTE/srsLTE/tree/fix_enb_crash a try?

avalori1, 2021-02-15:
Thanks for the quick feedback.
This is what I have done (I will try to be as detailed as possible).
I am running the versions mentioned above (20.10.1). I start both srsenb and srsepc (without a specific order).
I have a total of 4 SIMs that I programmed and used, I have placed them at different times in 4 different devices, swapping them several times:
- Samsung s5
- IPhone 6s
- Lenovo PC with internal modem
- a U-Blox modem module
At the moment of the latest test (the one that you were guessing might have not worked) there was only one active device  with a programmed SIM in it (the Samsung). One other of the custom SIMs was in the Iphone which was, however, switched off.
The Samsung phone was able to see the network but was unable to connect to it (with all the previous builds the phone was connecting happily and able to surf the internet).

About the system, does this answer to your questions?:

I am using an Ettus B200 as SDR and the below as PC

bts@bts-ThinkPad-L460:~$ sudo neofetch --stdout
root@bts-ThinkPad-L460 
---------------------- 
OS: Ubuntu 18.04.5 LTS x86_64 
Host: 20FVS1A400 ThinkPad L460 
Kernel: 5.4.0-65-generic 
Uptime: 2 days, 5 hours, 27 mins 
Packages: 2404 
Shell: bash 4.4.20 
Resolution: 1920x1080 
WM: GNOME Shell 
WM Theme: Adwaita 
Theme: Adwaita [GTK3] 
Icons: Adwaita [GTK3] 
Terminal: gnome-terminal 
CPU: Intel i5-6200U (4) @ 2.800GHz 
GPU: Intel Skylake GT2 [HD Graphics 520] 
Memory: 2871MiB / 7395MiB

avalori1, 2021-02-16:
@andrepuschmann 

Dear Andre, one more update:

I Run the version `fix_enb_crash` compiled with the `cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_ASAN=True ..`
 option
```
---  Software Radio Systems LTE eNodeB  ---
Reading configuration file /home/bts/.config/srslte/enb.conf...
Built in RelWithDebInfo mode using commit 812f2571d on branch fix_enb_crash.
```
which crashed after about 5 hours with not much info in the stout (compared to all the info that came before)
```
 4fb 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 4fc 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
terminate called recursively
srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---
```
in the `srsLTE.backtrace.crash` there are two sections with the right timestamp:

```
--- command='srsenb' version=20.10.1 signal=6 date='16/02/2021 04:48:45' ---
	srsenb(+0x67f480) [0x55da1841f480]
	srsenb(+0x63c40e) [0x55da183dc40e]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7fd38aec8040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7fd38aec7fb7]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7fd38aec9921]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8c957) [0x7fd38b8bc957]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ae6) [0x7fd38b8c2ae6]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92b21) [0x7fd38b8c2b21]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92d54) [0x7fd38b8c2d54]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x8e811) [0x7fd38b8be811]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x124799) [0x7fd38b954799]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE9_M_mutateEmmPKcm+0x6b) [0x7fd38b954b8b]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(_ZNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEE10_M_replaceEmmPKcm+0x1cb) [0x7fd38b95594b]
	/usr/local/lib/libsrslte_rf.so.0(+0x5b00f) [0x7fd38c02100f]
	/usr/local/lib/libsrslte_rf.so.0(rf_uhd_recv_with_time_multi+0x6ae) [0x7fd38bfee6fe]
	srsenb(+0x90874a) [0x55da186a874a]
	srsenb(+0x90ac47) [0x55da186aac47]
	srsenb(+0x246bdc) [0x55da17fe6bdc]
	srsenb(+0x1aeb1e) [0x55da17f4eb1e]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7fd38ceb86db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fd38afaa71f]

--- command='srsenb' version=20.10.1 signal=6 date='16/02/2021 04:48:47' ---
	srsenb(+0x67f480) [0x55da1841f480]
	srsenb(+0x63c40e) [0x55da183dc40e]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3f040) [0x7fd38aec8040]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7fd38aec7fb7]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7fd38aec9921]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x94c42) [0x7fd38b8c4c42]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92ae6) [0x7fd38b8c2ae6]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92b21) [0x7fd38b8c2b21]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0x92d54) [0x7fd38b8c2d54]
	/usr/local/lib/libuhd.so.3.15.0(+0x2164ae) [0x7fd3884bc4ae]
	/usr/local/lib/libuhd.so.3.15.0(_Z25set_c_global_error_stringRKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE+0x141) [0x7fd388b820f1]
	/usr/local/lib/libsrslte_rf.so.0(+0x5bbb3) [0x7fd38c021bb3]
	/usr/local/lib/libsrslte_rf.so.0(+0x22513) [0x7fd38bfe8513]
	/usr/local/lib/libsrslte_rf.so.0(_ZNSt6thread11_State_implINS_8_InvokerISt5tupleIJPFPvS3_EP16rf_uhd_handler_tEEEEE6_M_runEv+0x36) [0x7fd38bffa316]
	/usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xbd6df) [0x7fd38b8ed6df]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x76db) [0x7fd38ceb86db]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x3f) [0x7fd38afaa71f]
```

andrepuschmann, 2021-02-16:
Ok, I'll try to reproduce here myself. What else do you do during the 5h of the test with the UEs? You do traffic? Attach/detach? I see the RNTIs `4fb` is quite high?
Also, you use srsEPC by default with the default config, right?
Can you perhaps share all configs you use so I can setup the exact same here?

avalori1, 2021-02-16:
@andrepuschmann 

during the 5 hours there was a single UE (Samsung) which was **trying** to connect (but I think never managed to reach the internet). So, probably indeed may (at least attempts) to attach-detach.
To be 100% detailed (although I am pretty sure is not relevant), there was one more of the custom SIMs registered in the `user_db.csv` which was in an IPhone, which was turned off) and one other phone with a commercial SIM in it (maybe it was trying to connect to the eNB even if the SIM was not of course allowed?)
I am attaching the .conf files (removed the `user_db.csv`). In the `user_db.csv`, apart the IMSI and key information, the only difference from the default is that I assigned staic IPs to try to solve the issue #585 which I was facing in the past with several UEs (this test was with only one UE online, as said.
Can you please elaborate a bit more on the RNTIs? I understand that are basically the "UE identifiers"... however, this are many more than the number of allowed IMSIs... Is a RNTI created also for interactions with non allowed UEs? Is there a way to "reset" them? Would it be of any help?

Thanks and regards.

[srslte.zip](https://github.com/srsLTE/srsLTE/files/5988371/srslte.zip)

avalori1, 2021-02-18:
@andrepuschmann 
Hoping to be useful rather than annoying, one more example...
I also saved the files:
```
enb.log
sb.pcap
enb_s1ap.pcap
epc.log
epc.pcap
```
Let me know if you need them (in total are almost 5GB...)

Cheers,
Andrea

<pre> 160 4.5   0 1.1     0    0   96 100%    0  0.0   0     0    0    0   0%   0.0
 163 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 1b7 8.0   00.10     0    0    8 100%    0  0.0   0     0    0    0   0%   0.0
 1b8 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 1b9 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 1ba 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 215 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 216 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 217  11   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 225 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 243  14   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 244 6.5   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 245 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 26d  12   0 0.0     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 26e 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 270 7.0   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 29c 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 2df 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 2f8 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 311 0.10   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 312 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 313 1.0   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 315  14   0 0.8     0    0   93 100%    0  0.0   0     0    0    0   0%   0.0
 316 8.0   00.10     0    0    8 100%    0  0.0   0     0    0    0   0%   0.0
 317 0.10   00.10     0    0    3 100%    0  0.0   0     0    0    0   0%   0.0
 319 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 31a 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 349 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 34a 4.0   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
 34b 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 34c 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 34d 0.10   0 0.0     0    0    0   0%    0  0.0   0     0    0    0   0%   0.0
 34f 3.0   00.10     0    0    4 100%    0  0.0   0     0    0    0   0%   0.0
RACH:  tti=4361, cc=0, preamble=30, offset=1, temp_crnti=0x350
RACH:  tti=4451, cc=0, preamble=5, offset=31, temp_crnti=0x351
RACH:  tti=4471, cc=0, preamble=4, offset=31, temp_crnti=0x352
RACH:  tti=4491, cc=0, preamble=22, offset=31, temp_crnti=0x353
RACH:  tti=4511, cc=0, preamble=8, offset=31, temp_crnti=0x354
Disconnecting rnti=0x351.
RACH:  tti=4531, cc=0, preamble=54, offset=31, temp_crnti=0x355
Disconnecting rnti=0x352.
RACH:  tti=4551, cc=0, preamble=12, offset=31, temp_crnti=0x356
Disconnecting rnti=0x353.
RACH:  tti=4571, cc=0, preamble=10, offset=31, temp_crnti=0x357
Disconnecting rnti=0x354.
RACH:  tti=4591, cc=0, preamble=39, offset=31, temp_crnti=0x358
Disconnecting rnti=0x355.
RACH:  tti=4611, cc=0, preamble=44, offset=31, temp_crnti=0x359
RACH:  tti=4631, cc=0, preamble=29, offset=31, temp_crnti=0x35a
Disconnecting rnti=0x356.
Disconnecting rnti=0x357.
RACH:  tti=4651, cc=0, preamble=51, offset=31, temp_crnti=0x35b
RACH:  tti=4671, cc=0, preamble=10, offset=31, temp_crnti=0x35c
Disconnecting rnti=0x358.
Disconnecting rnti=0x359.
[... 224 lines cut ...]
  0x0fed3d1f9810: <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#AD7FA8"><b>f8</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#AD7FA8"><b>f8</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font>
  0x0fed3d1f9820: <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#AD7FA8"><b>f8</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#AD7FA8"><b>f8</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font>
  0x0fed3d1f9830: <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#AD7FA8"><b>f8</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color="#EF2929"><b>f2</b></font> <font color= [... cut ...]
  Container overflow:      <font color="#729FCF"><b>fc</b></font>
  Array cookie:            <font color="#EF2929"><b>ac</b></font>
  Intra object redzone:    <font color="#FCE94F"><b>bb</b></font>
  ASan internal:           <font color="#FCE94F"><b>fe</b></font>
  Left alloca redzone:     <font color="#729FCF"><b>ca</b></font>
  Right alloca redzone:    <font color="#729FCF"><b>cb</b></font>
==6565==ABORTING
<font color="#8AE234"><b>bts@bts-ThinkPad-L460</b></font>:<font color="#729FCF"><b>~</b></font>$ 
</pre>

andrepuschmann, 2021-02-18:
@avalori1 We've been able to reproduce the issue and work on a fix for it. I'll keep you posted as soon as we have something that you can try. Thanks again for the reports

avalori1, 2021-02-18:
Great, many thanks. Looking forward to try your solution.

andrepuschmann, 2021-02-24:
Hey, we've pushed a new branch to https://github.com/srsLTE/srsLTE/tree/fixing_enb_20_10 that contains a few backported fixes for the eNB. Let me know if they improve the situation for you.

Thanks
Andre

avalori1, 2021-03-15:
Hello, I run some tests and this are the first impressions.
It seems to be way better. I did try to install with the `cmake -DCMAKE_BUILD_TYPE=RelWithDebInfo -DENABLE_ASAN=True ..`
option but after that I was unable to start my node (using a B200).
The Tx and Rx LED on the board were off or flickering... After re-compiling without the option the enode seems working well and with no crash. Will want to also increase the number of UE but will need some time for that.
On the topic, is there a way to put in relationship the rnti shown by the trace of the srsenb with the IMSI?

Thanks

andrepuschmann, 2021-03-15:
That's good to know. Thanks for checking. The ASAN build is not meant for normal operation. It's just for testing, etc. There is 2x performance penalty when having ASAN for all the checks it does. Therefore, you see the flickering LEDs. 

Regarding your other question, no there is no relationship. The RNTI is assigned by the eNB without knowing the IMSI

liumaolin-931, 2021-04-14:
Hi
I have some similar problems and the terminal interface is as follows:

sudo srsepc epc.conf
Built in Release mode using commit 45486b6 on branch master.

--- Software Radio Systems EPC ---

Reading configuration file epc.conf...
HSS Initialized.
Error binding SCTP socket
Error initializing GTP-C

The strange thing is that it can work normally after boot and restart, but when I turn off the current terminal and open a new one, this error will appear. Can someone help me?Thanks!

avalori1, 2021-04-14:
Seems that you still have a srsepc running in the background... did you
start it with & at the end?
Try
$ pidof srsepc
if you get a number means that epc is still running...

and then:
$ sudo kill -9 <PID_you_got_above>

hope it helps

liumaolin-931, 2021-04-15:
thank you very much!The problem is solved！

 start it with &amp; at the end?
 
 $ sudo kill -9 <PID_you_got_above&gt;
 
 hope it helps
 
 
 &gt; Hi
 &gt; I have some similar problems and the terminal interface is as follows:
 &gt;
 &gt; sudo srsepc epc.conf
 &gt; Built in Release mode using commit 45486b6
 &gt; <https://github.com/srsLTE/srsLTE/commit/45486b6e2c622ab1d47906521662f087816d7d03&gt;
 &gt; on branch master.
 &gt;
 &gt; --- Software Radio Systems EPC ---
 &gt;
 &gt; Reading configuration file epc.conf...
 &gt; HSS Initialized.
 &gt; Error binding SCTP socket
 &gt; Error initializing GTP-C
 &gt;
 &gt; The strange thing is that it can work normally after boot and restart, but
 &gt; when I turn off the current terminal and open a new one, this error will
 &gt; appear. Can someone help me?Thanks!
 &gt; <https://github.com/srsLTE/srsLTE/issues/623#issuecomment-819275876&gt;, or
 &gt; unsubscribe

andrepuschmann, 2021-04-23:
Very good. Closing the issue.

closed 2021-04-23 by andrepuschmann