srsran/srsRAN_4G #626: Hi, How much Coverage in meter or KM in srsLTE with USRP B210? is it depend on Bandwidth? or What depends on it?

mahdibhi, 2021-02-16:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

closed 2021-02-16 by andrepuschmann

frankist, 2021-02-16:
This is not an issue.

mahdibhi, 2021-02-16:
Yes, It Is A Question.