srsran/srsRAN_4G #629: Change of channel at runtime

lucabaldesi, 2021-02-18:
Hi,
first, thank you for this project! I am experimenting with eNodeB and I am currently trying to reconfigure at runtime (according to a trigger function I still have to define) the channel frequency (or the earfcn).
So far, I have identified two ways to accomplish this,
 1. Through direct RRC reconfigure messages, following TS 36.331 v.10.22.0; 
 2. Defining multiple cells (sectors) for the same eNodeB (following [1]) and triggering a handover;

The first option, being more straightforward, makes more sense, but I could not find a direct mapping in the code API (rcc_mobility.h). I imagine I may accomplish this by using the function handle_ho_req() as a template.

Questions:
 1. is it possible to change the channel frequency (earfcn) of the eNodeB at runtime? In general, is it possible to change the values of its configuration after it has booted?
 2. can I add/remove new cells (sectors) to a running eNodeB dynamically?
 3. do you have any suggestion on how to implement the reconfiguration message function just to change the frequency/earfcn?

Best,
Luca

[1]  https://docs.srslte.com/en/latest/app_notes/source/handover/source/index.html#handover-appnote

andrepuschmann, 2021-04-23:
Hey,
I am afraid all what your asking for isn't implemented. Cells are statically configured and loaded at startup. They can't be changed at run time. 

Thanks

closed 2021-04-23 by andrepuschmann

SumitEurecom, 2024-04-05:
Can we change the Tx and Rx gains at run time ?

stackprogramer, 2024-11-08:
Did anyone find a solution for changing parameters in real time?

lucabaldesi, 2024-11-12:
Hi,
in the end I implemented an interface that at run time allows the change of the frequency. It is not possible to add/remove a cell but I configured two cells and kept one active while I was reconfiguring the other, then triggered a seamless handover of the clients from one cell to another (as described in the 3GPP standard).

For the technical explanation: https://baldesi.ovh/download/Baldesi2022Charm.pdf
Code is available here: https://github.com/lucabaldesi/srsRAN

stackprogramer, 2024-11-15:
@lucabaldesi Great.  Thanks very much. I will examine it.