srsran/srsRAN_4G #631: epc doesn't start due to the following issue

saichandrad, 2021-03-08:
## Issue Description ##
When I try to start the epc using srsepc command - It shows it couldn't initialize SGi-interface, SPGW-GTP-U, SPGW

## Setup Details ##
Running on ubuntu 16

## Expected Behavior ##
epc start

## Actual Behaviour ##
shows it couldn't initializae - SGi-interface, SPGW-GTP-U, SPGW

## Steps to reproduce the problem ##
run sudo srsepc commnd

[image]

JayakumarCK2, 2021-03-08:
I think you have a suspended srsepc.
Do pidof srsepc : you will get a Jobid
Kill the process, restart epc

matan1008, 2021-03-11:
In my humble opinion you are missing the `sudo` before the `srsepc` command, thus failing to create the TUN device.

liumaolin-931, 2021-04-14:
Hi
I have some similar problems and the terminal interface is as follows:

sudo srsepc epc.conf
Built in Release mode using commit 45486b6e2 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file epc.conf...
HSS Initialized.
Error binding SCTP socket
Error initializing GTP-C

The strange thing is that it can work normally after boot and restart, but when I turn off the current terminal and open a new one, this error will appear. Can someone help me?Thanks!

andrepuschmann, 2021-04-23:
probably another EPC running using the same port.

closed 2021-04-23 by andrepuschmann

EhsaanIlyas456, 2022-08-24:
anyone solved it?

KingEverest, 2022-10-29:
the problem occurs when you run "srsran_install_configs.sh user" without root provilege. To solve the problem run the script with root privileges like as below

'sudo srsran_install_configs.sh user'