srsran/srsRAN_4G #634: srsue TDD can not connect to TDD network

apple68, 2021-03-17:
## Issue Description ##
srsue TDD can not connect to TDD network

## Actual Behaviour ##
[image]
after ue send attach request, it will receive rrcconnect release!!!

EugeneBogush, 2021-04-15:
hi,
which equipment you using?
would you to attach the logs and pcap-file?

apple68, 2021-04-16:
> hi,
> which equipment you using?
> would you to attach the logs and pcap-file?

hello, I use usrpB210.Here are my logs and pcap-file.thank you!!!
[log.zip](https://github.com/srsLTE/srsLTE/files/6322002/log.zip)

andrepuschmann, 2021-04-23:
Logs look pretty ok for the DL. Signal is not great but good enough. I'd say it's a UL issue. The ConnectionSetupComplete isn't going through. Check the antenna, cabling, power, etc.

apple68, 2021-04-27:
> Logs look pretty ok for the DL. Signal is not great but good enough. I'd say it's a UL issue. The ConnectionSetupComplete isn't going through. Check the antenna, cabling, power, etc.

Wow!!! 
Hi, if srsue run in TDD mode, need special equipment or configuration?
Thank you!!!
I have tried various releases, only release_19.06 can attach TDD network.

andrepuschmann, 2021-11-26:
I could well be that the particular config that the network uses only works with `release_19.06` then. We don't test all configs I am afraid.

closed 2021-11-26 by andrepuschmann