srsran/srsRAN_4G #636: srsenb and srsue v20.10.1 crash with UHD4 and X310/E320

dustinmaas, 2021-03-23:
## Issue Description ##
`srsenb` and `srsue` v20.10.1 crash reliably during radio initialization when using UHD4 and an X310 or E320. The crash is triggered when accessing the USRP property tree ([this line](https://github.com/srsLTE/srsLTE/blob/45486b6e2c622ab1d47906521662f087816d7d03/lib/src/phy/rf/rf_uhd_generic.h#L197)). I don't see the same issue when using a B210. Presumably, this is because the path in the property tree exists for B2xx USRPs. Also, the issue doesn't present with the X310/E320 when srsLTE v20.10.1 is built against UHD3.15, so maybe the API for accessing the property tree has changed.

## Setup Details ##
Server-class compute w/ lots of cores and RAM
Ubuntu 18.04
UHD 4.0.0.0
USRP X310 or E320

## Expected Behavior ##
`srsenb` and `srsue` run normally.

## Actual Behaviour ##
`srsenb` and `srsue` crash during radio initialization.

## Steps to reproduce the problem ##
Build srsLTE v20.10.1 against UHD4, connect an X310 or E320, and start `srsenb` or `srsue`.

## Additional Information ##
Here is stdout/stderr for `srsenb`:

```
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /etc/srslte/enb.conf...

Built in RelWithDebInfo mode using commit 45486b6e2 on branch master.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.0-2ubuntu1emulab2
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
ASAN:DEADLYSIGNAL
=================================================================
==9000==ERROR: AddressSanitizer: SEGV on unknown address 0x000000000000 (pc 0x7fc293aab19f bp 0x7ffd97191210 sp 0x7ffd9718fdc0 T0)
==9000==The signal is caused by a READ memory access.
==9000==Hint: address points to the zero page.
    #0 0x7fc293aab19e in rf_uhd_generic::usrp_make(uhd::device_addr_t const&, unsigned int) /local/srsLTE/lib/src/phy/rf/rf_uhd_generic.h:197
    #1 0x7fc293a5528a in rf_uhd_open_multi /local/srsLTE/lib/src/phy/rf/rf_uhd_imp.cc:711
    #2 0x7fc293a4583d in srslte_rf_open_devname /local/srsLTE/lib/src/phy/rf/rf_imp.c:123
    #3 0x563b042a8029 in srslte::radio::open_dev(unsigned int const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) /local/srsLTE/lib/src/radio/radio.cc:453
    #4 0x563b042b2f87 in srslte::radio::init(srslte::rf_args_t const&, srslte::phy_interface_radio*) /local/srsLTE/lib/src/radio/radio.cc:145
    #5 0x563b03b89296 in srsenb::enb::init(srsenb::all_args_t const&, srslte::logger*) /local/srsLTE/srsenb/src/enb.cc:88
    #6 0x563b03b28d7d in main /local/srsLTE/srsenb/src/main.cc:496
    #7 0x7fc292912b96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)
    #8 0x563b03b2c7e9 in _start (/local/srsLTE/build/srsenb/src/srsenb+0x1987e9)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV /local/srsLTE/lib/src/phy/rf/rf_uhd_generic.h:197 in rf_uhd_generic::usrp_make(uhd::device_addr_t const&, unsigned int)
==9000==ABORTING
```

andrepuschmann, 2021-03-23:
Thanks @dustinmaas for flagging the issue. We are currently mainly testing against UHD 3.15 LTS. Before the 20.10 release, I remember testing against UHD 4.0 manually with a B210, but never with a X310.

We try to get a UHD 4.0 installation on one of our test machines with an X310 before the upcoming release, but I can't promise anything earlier than that.

dustinmaas, 2021-05-03:
> hey @dustinmaas, I confirm the issue with the device tree for the X310
> and UHD 4.0. It's now fixed and working reasonably well (there are
> still some warnings printed) with the X310 and UHD4. Here is 2xCA with
> 100 PRB:
> ```
> $  ./srsenb/src/srsenb ../srsenb/enb.conf.example
> --enb_files.rr_config /etc/srsran/rr_2cc.co
> nf  --rf.tx_gain=5  --enb.n_prb=100
> ---  Software Radio Systems LTE eNodeB  ---
>
> Reading configuration file ../srsenb/enb.conf.example...
>
> Built in Release mode using commit bde30bb9b on branch dev.
>
> Opening 2 channels in RF device=default with args=default
> [INFO] [UHD] linux; GNU C++ version 9.3.1 20200408 (Red Hat 9.3.1-2);
> Boost_106900; UHD_4.0.0.0-50-ge520e3ff
> [INFO] [LOGGING] Fastpath logging disabled at runtime.
> [INFO] [MPMD FIND] Found MPM devices, but none are reachable for a UHD
> session. Specify find_all to find all devices.
> Opening USRP channels=2, args: type=x300,master_clock_rate=184.32e6
> [INFO] [UHD RF] RF UHD Generic instance constructed
> [INFO] [X300] X300 initialization sequence...
> [INFO] [X300] Maximum frame size: 8000 bytes.
> [INFO] [X300] Radio 1x clock: 184.32 MHz
> [... 57 lines cut ...]
>    0   0%   0.0
> ```
>
> I am closing the issue therefore. Please reopen if you still have
> problems with the new release.

Thanks for adding a fix Andre. I’ve confirmed that it works in the 
latest release.

joeldelosangeles, 2021-05-25:
Hi Andres,

I'm using srsLTE v20.10.1 built against UHD3.15 using a B210.  Also, I saw in your post that you successfully made an X310 work with PRB=100.   Can the same be said about the E320?  Can we expect a value near 150 Mbps for the downlink if I use an E320?

Hoping for your response.

Thanks,
Joel

andrepuschmann, 2021-05-26:
We've not tested with an E320 ourselves.

joeldelosangeles, 2021-09-22:
Hi Andre,

We are using UHD_3.14.1.1-release and B210.  What is the uplink throughput received at 5 MHz (prb 25)?  How about at 10 MHz (prb 50).  We are only getting up to 2 Mbps or even lower.  Do you have an idea with regards to this issue? 

Thanks,
Joel

andrepuschmann, 2021-09-22:
UL for a single user with 50 PRB should be around 21-23 Mbps. Works well with B210, tested with UHD 3.15 LTS.

joeldelosangeles, 2021-09-23:
> UL for a single user with 50 PRB should be around 21-23 Mbps. Works well with B210, tested with UHD 3.15 LTS.

Hi Andre,

I was able to upgrade to UHD 3.15 LTS and reinstall srsLTE 20.4.1 with B210.  S1AP is connected.  enb_s1ap.pcap shows 2 MIB's but enb.pcap shows no packets.  My phone cant camp and RACH connect/disconnect is not present in t trace.  Attached are enb.log and srsenb output.  

I hope you can help me with this issue.   Additional information during srsLTE make test:

724/739 Test #724: nas_test ...................
.......................***Exception: Child aborted  0.22 sec

[enb.log](https://github.com/srsran/srsRAN/files/7215843/enb.log)
[srsenb.txt](https://github.com/srsran/srsRAN/files/7215847/srsenb.txt)

Thanks,
Joel

andrepuschmann, 2021-09-23:
You run in USB2 mode. That's not gonna work

```
[INFO] [B200] Operating over USB 2.
```

joeldelosangeles, 2021-09-23:
Hi Andre,

Thank you Andre.  It is now working.  This is just an indoor short distance setup.  At max pusch mcs = 16, we were able to get maximum uplink throughput of 13 Mbps iperf UDP.  The throughput at uplink mcs = 28 is unstable.

Would you know the cause if srsenb suddenly stops at 

RF status: O=3, U=4, L=56

although it was able to recover after a few minutes. 

Thanks,
Joel