srsran/srsRAN_4G #639: Problem with "interpolate" method in channel estimates

softdev86, 2021-04-05:
## Issue Description ##
Dear all,
I came across a situation where the snr of the received signal is relatively high, but the percentage of successfully decoded  sib1 packets in the pdsch_ue utility is very low. Specifically, the problem was observed on a cell with 4 ports and SRSLTE_ESTIMATOR_ALG_INTERPOLATE method for channel estimates. The problem is not reproducible when the method for channel estimates is switched to SRSLTE_ESTIMATOR_ALG_AVERAGE. Also the problem is not reproducible if I switch to a cell with one or two ports.

## Setup Details ##
I have been running tests on ubuntu 16, libuhd 3.13, b210 device with srslte 20.10.1 version. Here are the cell parameters:
 - Type:            FDD
 - PCI:             408
 - Nof ports:       4
 - CP:              Normal
 - PRB:             100
 - PHICH Length:    Normal
 - PHICH Resources: 1
 - SFN:             404

## Expected Behavior ##
Example of statistics from pdsh_ue with SRSLTE_ESTIMATOR_ALG_AVERAGE on PCI 408:
          CFO:  -50.70 Hz
         RSRP: +38.5 dBm | +39.9 dBm
          SNR:  +4.7 dB
           TM: 2
           Rb:   0.10 /   0.14 /   0.14 Mbps (net/maximum/processing)
   PDCCH-Miss:  8.58%
   PDSCH-BLER:  0.22%
         TB 0: mcs=5; tbs=144
         TB 1: mcs=0; tbs=0

## Actual Behaviour ##
Example of statistics from pdsh_ue with SRSLTE_ESTIMATOR_ALG_INTERPOLATE on PCI 408:
          CFO: -150.58 Hz
         RSRP: +39.1 dBm | +40.3 dBm
          SNR:  +4.1 dB
           TM: 2
           Rb:   0.01 /   0.14 /   0.00 Mbps (net/maximum/processing)
   PDCCH-Miss: 36.14%
   PDSCH-BLER: 89.34%
         TB 0: mcs=5; tbs=144
         TB 1: mcs=0; tbs=0

## Steps to reproduce the problem ##
I'm not 100% sure, but the problem should be reproducible on any cell with 4 antenna ports and SRSLTE_ESTIMATOR_ALG_INTERPOLATE method for channel estimates on rx side.

## Additional Information ##
Changes in chest_dl.c, interpolate_pilots function solved the problem with SRSLTE_ESTIMATOR_ALG_INTERPOLATE  for me. I attach diff with changes. 
[chest_dl_diff.txt](https://github.com/srsLTE/srsLTE/files/6258412/chest_dl_diff.txt)

Hope it helps.

Best regards

andrepuschmann, 2021-04-23:
Thanks for reporting this. We'll investigate.

xavierarteaga, 2021-04-27:
Many thanks, @softdev86! Your patch will be included in the next release.

closed 2021-04-27 by xavierarteaga