srsran/srsRAN_4G #641: Error building rf_soapy_imp.c.o: conflicting types for ‘SoapySDRRange’

MountainLogic, 2021-04-07:
trying to build for on a fresh ubuntu 20.04 install.  Install preqs and srsgui.  cloned, cd srslte, mkdir build, cd build then:
```

foobar@foobar-pc:~/wip/srslte/build$ cmake ..
-- CMAKE_SYSTEM: Linux-5.8.0-48-generic
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- FFTW3F LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- FFTW3F STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.a
-- FFTW3F INCLUDE DIRS: /usr/include
-- FFT_LIBRARIES: /usr/lib/x86_64-linux-gnu/libfftw3f.so
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.so
-- MBEDTLS STATIC LIBRARIES: /usr/lib/x86_64-linux-gnu/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/include
-- PCSC LIBRARIES: /usr/lib/x86_64-linux-gnu/libpcsclite.so
-- PCSC INCLUDE DIRS: /usr/include/PCSC
-- Building with PCSC support.
-- UHD LIBRARIES /usr/lib/x86_64-linux-gnu/libuhd.so
-- UHD INCLUDE DIRS /usr/include
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- FINDING ZEROMQ.
-- SRSGUI LIBRARIES /usr/local/lib/libsrsgui.so
-- SRSGUI INCLUDE DIRS /usr/local/include
-- Found SRSGUI: /usr/local/lib/libsrsgui.so  
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- SSE4.1 is enabled - target CPU must support it
-- AVX is enabled - target CPU must support it
-- AVX2 is enabled - target CPU must support it
-- FMA is enabled - target CPU must support it
-- CMAKE_C_FLAGS is  -Wno-unused-but-set-variable -Werror=incompatible-pointer-types -Wall -Wno-comment -Wno-write-strings -Wno-unused-result -Wformat -Wmissing-field-initializers -Wtype-limits -std=c99 -fno-strict-aliasing -D_GNU_SOURCE -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -mfma -DLV_HAVE_FMA -Ofast -funroll-loops -fvisibility=hidden -Werror --param large-function-growth=1600
-- CMAKE_CXX_FLAGS is  -Wall -Wno-comment -Wno-reorder -Wno-unused-variable -Wtype-limits -std=c++11 -fno-strict-aliasing -Wno-unused-but-set-variable -march=native -mfpmath=sse -mavx2 -DLV_HAVE_AVX2 -DLV_HAVE_AVX -DLV_HAVE_SSE -fvisibility=hidden -O3 -fno-trapping-math -fno-math-errno -DBUILD_TYPE_RELEASE -Werror
-- Using install prefix: /usr/local
-- Building for version: 20.10.1
--    examples will be installed.
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build command defined
-- Building with srsUE
-- No post-build-UE command defined
-- No post-build command defined
-- Building with srsENB
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
-- Found LibConfig: /usr/lib/x86_64-linux-gnu/libconfig.so
-- static LibConfig path: /usr/lib/x86_64-linux-gnu/libconfig.a
-- Checking for module 'sctp'
--   No package 'sctp' found
-- SCTP LIBRARIES: /usr/lib/x86_64-linux-gnu/libsctp.so
-- SCTP INCLUDE DIRS: /usr/include
-- No post-build-ENB command defined
-- Building with srsEPC
-- Found LibConfig++: /usr/lib/x86_64-linux-gnu/libconfig++.so
-- static LibConfig++ path: /usr/lib/x86_64-linux-gnu/libconfig++.a
[... 1073 lines cut ...]
Scanning dependencies of target sched_grid_test
[ 77%] Building CXX object srsenb/test/mac/CMakeFiles/sched_grid_test.dir/sched_grid_test.cc.o
[ 77%] Linking CXX executable rlc_am_test
[ 77%] Built target rlc_am_test
Scanning dependencies of target s1ap_asn1
[ 77%] Building CXX object lib/src/asn1/CMakeFiles/s1ap_asn1.dir/s1ap_asn1.cc.o
Scanning dependencies of target rrc_asn1
[ 77%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o
[ 77%] Linking CXX executable sched_grid_test
[ 77%] Built target sched_grid_test
[ 77%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1_nbiot.cc.o
[ 78%] Linking CXX executable sched_lc_ch_test
[ 78%] Built target sched_lc_ch_test
[ 78%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1_enum.cc.o
make[1]: *** [CMakeFiles/Makefile2:4721: lib/src/phy/rf/CMakeFiles/srslte_rf.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 78%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1_utils.cc.o
[ 78%] Linking CXX static library libs1ap_asn1.a
[ 78%] Built target s1ap_asn1
[ 79%] Linking CXX static library librrc_asn1.a
[ 79%] Built target rrc_asn1
make: *** [Makefile:163: all] Error 2
foobar@foobar-pc:~/wip/srslte/build$ 

```

andrepuschmann, 2021-04-08:
That's a mismatch of Soapy libs. What version are you using? Soapy? srsLTE?

MountainLogic, 2021-04-10:
scottfoobar@foobar-pc:~/wip/srslte$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal

foobar@foobar-pc:~/wip/srslte$ git branch -a
* master
  remotes/origin/HEAD -> origin/master
  remotes/origin/fix_enb_crash
  remotes/origin/fixing_enb_20_10
  remotes/origin/master
  remotes/origin/nas_test_fix
  remotes/origin/pr_asn1_fix

foobar@foobar-pc::~/wip/srslte$ git status
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	build/

nothing added to commit but untracked files present (use "git add" to track)

foobar@foobar-pc:~$ apt-cache policy soapy libsoapysdr-dev
libsoapysdr-dev:
  Installed: 0.7.2-1
  Candidate: 0.7.2-1
  Version table:
 *** 0.7.2-1 500
        500 http://us.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
        100 /var/lib/dpkg/status
N: Unable to locate package soapy
foobar@foobar-pc:~$ apt-cache policy soapy libsoapysdr0.7
libsoapysdr0.7:
  Installed: 0.7.2-1
  Candidate: 0.7.2-1
  Version table:
 *** 0.7.2-1 500
        500 http://us.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
        100 /var/lib/dpkg/status
N: Unable to locate package soapy
foobar@foobar-pc:~$ soapy_power --version
soapy_power 1.6.1

MountainLogic, 2021-04-10:
I'm really here looking for a tool to measure the xtal error for a PlutoSDR, so if there is a prebuilt for say windows, I measure the offset I'd be happy with that.

MountainLogic, 2021-04-15:
/usr/include/soapySDR/version.h  starts with this:

```
///
/// \file SoapySDR/Version.h
///
/// Utility functions to query version information.
///
/// \copyright
/// Copyright (c) 2014-2018 Josh Blum
/// Copyright (c) 2016-2016 Bastille Networks
/// SPDX-License-Identifier: BSL-1.0
///

#pragma once
#include <SoapySDR/Config.h>

/*!
 * API version number which can be used as a preprocessor check.
 * The format of the version number is encoded as follows:
 * <b>(major << 24) | (minor << 16) | (16 bit increment)</b>.
 * Where the increment can be used to indicate implementation
 * changes, fixes, or API additions within a minor release series.
 *
 * The macro is typically used in an application as follows:
 * \code
 * #if defined(SOAPY_SDR_API_VERSION) && (SOAPY_SDR_API_VERSION >= 0x00001234)
 * // Use a newer feature from the SoapySDR library API
 * #endif
 * \endcode
 */
#define SOAPY_SDR_API_VERSION 0x00070001

/*!
 * ABI Version Information - incremented when the ABI is changed.
 * The ABI version format is <b>version[-extra]</b>.
 * The <i>version</i> comes from the associated library major.minor version.
 * And <i>extra</i> is empty for releases but set on development branches.
 * The ABI should remain constant across patch releases of the library.
 */
#define SOAPY_SDR_ABI_VERSION "0.7"

```

What version is expected

MountainLogic, 2021-04-15:
I'm seeing the error with a clean clone of srsLTE.

in /usr/local/lib I have 0.8, hummm, but removing all of libsoapy-dev and reinstalling seemed to install a lot of 0.7 stuff. hummm?

lrwxrwxrwx  1 root root       18 Apr  6 11:38 libSoapySDR.so -> libSoapySDR.so.0.8
lrwxrwxrwx  1 root root       20 Apr  6 11:38 libSoapySDR.so.0.8 -> libSoapySDR.so.0.8.0
-rw-r--r--  1 root root   632856 Apr  6 11:36 libSoapySDR.so.0.8.0

MountainLogic, 2021-04-15:
SO it turns out I some how have different versions of the libs installed:
home/foobar/wip/SoapySDR/build/lib/libSoapySDR.so
/home/foobar/wip/SoapySDR/build/lib/libSoapySDR.so.0.8
/home/foobar/wip/SoapySDR/build/lib/libSoapySDR.so.0.8.0
/usr/lib/x86_64-linux-gnu/libSoapySDR.so
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7
/usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7.2
/usr/local/lib/libSoapySDR.so
/usr/local/lib/libSoapySDR.so.0.8
/usr/local/lib/libSoapySDR.so.0.8.0

~/wip/SoapySDR/build$ SoapySDRUtil --info
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Lib Version: v0.8.0-g351896d3
API Version: v0.8.0
ABI Version: v0.8
Install root: /usr/local
Search path:  /usr/local/lib/SoapySDR/modules0.8
Module found: /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so (0.1.1-2f346c7)
Module found: /usr/local/lib/SoapySDR/modules0.8/librtlsdrSupport.so   (0.3.2-8277756)
Available factories... plutosdr, rtlsdr
Available converters...
 -  CF32 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS16 -> [CF32, CS16, CS8, CU16, CU8]
 -  CS32 -> [CS32]
 -   CS8 -> [CF32, CS16, CS8, CU16, CU8]
 -  CU16 -> [CF32, CS16, CS8]
 -   CU8 -> [CF32, CS16, CS8]
 -   F32 -> [F32, S16, S8, U16, U8]
 -   S16 -> [F32, S16, S8, U16, U8]
 -   S32 -> [S32]
 -    S8 -> [F32, S16, S8, U16, U8]
 -   U16 -> [F32, S16, S8]
 -    U8 -> [F32, S16, S8]

which version should I keep?

arywz, 2021-04-22:
libsoapysdr-dev provided Types.h, and **SoapySDRRange** is defined in that file. But the definations of SoapySDRRange between SoapySDR.git and libsoapysdr-dev are different. I guess the libsoapysdr-dev is outdated.

Temporary solution I used is **mv /usr/include/SoapySDR /usr/include/SoapySDR.bk** and re-run cmake, and then it should not  apprear that conflict error.

MountainLogic, 2021-04-23:
@arywz thanks for the great advice! I'll investigate.

andrepuschmann, 2021-04-28:
I am closing this issue. It's a build and/or version mismatch thing

closed 2021-04-28 by andrepuschmann

MountainLogic, 2021-04-28:
I uninstalled the latest soapy that I built from source and things got a littler farther.T there was a linking problem with libsoapysdr.so.  It seems that the the link  for the real .so using apt-get Ubuntu 20.04 package is placed in /user/lib/x86_64-linux-gnu and make could not find the .so at link time.   I added a new symbolic link using this:
```
cd  /user/local/lib
sudo ln /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7.2 libSoapySDR.so 

```
and it built like without any further issues.  I'm sure three is a better way to solve this than a symbolic link