srsran/srsRAN_4G #645: srsenb stdout without space

avalori1, 2021-04-14:
## Issue Description ##
The output for srsenb in some cases is missing a space between adiacent metrics. This makes parsing automatically tedious (and possibly even ambiguous).
For example, splitting a line may lead to
['46', '15', '0', '8.0', '392', '1', '0', '0%', '12.4', **'36.010.0'**, '4.3k', '1', '0', '0%', '0.0'] where 36.010.0 may be interpreted as one of the two options: 
1. 36.01 and 0.0
2. 36.0 and 10.0
 
## Setup Details ##
I am running the below, but seems that is happening with all the versions... 
Built in Release mode using commit 45486b6e2 on branch master.

## Expected Behavior ##
Having a space in between the metric whatever are the numerical values 

## Actual Behaviour ##
As described above, in some cases the space is missing

## Steps to reproduce the problem ##
Just run srsenb extensively, with different UE and/or conditions.

## Additional Information ##
Additionally, to even further simplify the parsing, would be good to have unique column names (e.g. brateU and brateD for the two), rather than (or in addition to) having the DL and UL descriptions above, which are easy to read by eye, but less when parsing the text file automatically.

andrepuschmann, 2021-04-23:
Thanks for the reporting. We are aware of the issue and look into it.

andrepuschmann, 2021-11-26:
All this is now available in the official release.

closed 2021-11-26 by andrepuschmann