srsran/srsRAN_4G #650: Unable to build srsLTE on Ubuntu

chainhead, 2021-04-21:
## Issue Description ##
I am unable to build `srsLTE` as it is unable to find `fftw` and `mbedTLS`. I even tried installing `fftw` with `sudo apt-get install -y fftw3`. This is the error message I see.

## Setup Details ##
- AWS EC2 instance running Ubuntu 20.04 LTS (64 bit)

### Installation steps

```bash
sudo apt-get update && sudo apt-get install -y \
build-essential \
libtool \
pkg-config \
m4 \
cmake \
libzmq3-dev
# libzmq
git clone https://github.com/zeromq/libzmq.git
cd libzmq
./autogen.sh
./configure
make
sudo make install
sudo ldconfig
cd
# czmq
git clone https://github.com/zeromq/czmq.git
cd czmq
./autogen.sh
./configure
make
sudo make install
sudo ldconfig
cd
# fftw
wget http://www.fftw.org/fftw-3.3.9.tar.gz
tar -xzf fftw-3.3.9.tar.gz && cd fftw-3.3.9
./configure
make && sudo make install
cd
# mbedTLS
wget https://github.com/ARMmbed/mbedtls/archive/refs/tags/v2.26.0.tar.gz
tar -xzf v2.26.0.tar.gz && cd mbedtls-2.26.0
cmake .
cmake --build .
sudo cmake --install .
cd
# srsLTE
git clone https://github.com/srsLTE/srsLTE.git
cd srsLTE
mkdir build
cd build
cmake ../
make
```

## Expected Behavior ##
- Build of `srsLTE` completes successfully.

## Actual Behaviour ##
The following error message is seen when building `srsLTE`.

```bash
-- CMAKE_SYSTEM: Linux-5.4.0-1038-aws
-- CMAKE_SYSTEM_PROCESSOR: x86_64
-- CMAKE_CXX_COMPILER: /usr/bin/c++
-- Build type not specified: defaulting to Release.
-- Checking for module 'fftw3f >= 3.0'
--   No package 'fftw3f' found
-- FFTW3F LIBRARIES: FFTW3F_LIBRARY-NOTFOUND
-- FFTW3F STATIC LIBRARIES: FFTW3F_STATIC_LIBRARY-NOTFOUND
-- FFTW3F INCLUDE DIRS: /usr/local/include
-- Could NOT find fftw3f (missing: FFTW3F_LIBRARY) 
-- Checking for module 'polarssl'
--   No package 'polarssl' found
-- POLARSSL LIBRARIES: POLARSSL_LIBRARIES-NOTFOUND
-- POLARSSL STATIC LIBRARIES: POLARSSL_STATIC_LIBRARIES-NOTFOUND
-- POLARSSL INCLUDE DIRS: POLARSSL_INCLUDE_DIRS-NOTFOUND
-- Could NOT find POLARSSL (missing: POLARSSL_LIBRARIES POLARSSL_INCLUDE_DIRS) 
-- Checking for module 'mbedtls'
--   No package 'mbedtls' found
-- MBEDTLS LIBRARIES: /usr/local/lib/libmbedcrypto.a
-- MBEDTLS STATIC LIBRARIES: /usr/local/lib/libmbedcrypto.a
-- MBEDTLS INCLUDE DIRS: /usr/local/include
-- Checking for module 'libpcsclite'
--   No package 'libpcsclite' found
-- Could NOT find PCSCLITE (missing: PCSCLITE_LIBRARY PCSCLITE_INCLUDE_DIR) 
-- PCSC LIBRARIES: PCSCLITE_LIBRARY-NOTFOUND
-- PCSC INCLUDE DIRS: PCSCLITE_INCLUDE_DIR-NOTFOUND
-- UHD LIBRARIES UHD_LIBRARIES-NOTFOUND
-- UHD INCLUDE DIRS UHD_INCLUDE_DIRS-NOTFOUND
-- Could NOT find UHD (missing: UHD_LIBRARIES UHD_INCLUDE_DIRS) 
-- Checking for module 'libbladeRF'
--   No package 'libbladeRF' found
-- libbladeRF not found.
-- FINDING SOAPY.
-- Checking for module 'SoapySDR'
--   No package 'SoapySDR' found
-- libSOAPYSDR not found.
-- FINDING ZEROMQ.
-- Could NOT find Boost (missing: Boost_INCLUDE_DIR program_options) (Required is at least version "1.35")
CMake Error at CMakeLists.txt:263 (message):
  Boost required to build srsLTE

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/srsLTE/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/srsLTE/build/CMakeFiles/CMakeError.log".
```

## Steps to reproduce the problem ##
- Run installation steps on Ubuntu-20.04 instance.

## Additional Information ##
- None

matan1008, 2021-04-23:
Try installing `libfftw3-dev`, `libsctp-dev` and `libmbedtls-dev` with apt

andrepuschmann, 2021-04-23:
Yes, you need the dev libs. Just follow the install tutorial. Closing

closed 2021-04-23 by andrepuschmann