srsran/srsRAN_4G #651: [Consult] When will release the 5G NR support feature in syslte

chaolinyi, 2021-04-21:
Hi, guys.

When will we tend to release the 5G NR support feature in syslte?

BR//chaolinyi

kbarlee, 2021-04-21:
https://www.srslte.com/srslte-srsran

chaolinyi, 2021-04-21:
Hi, kbarlee.

Thanks for your reply.

I have read the guide. And 5G NR srsRAN will release on 21.04 release. And the repo here tags show that 20.04-release has released on May 4th,2020.

So 21.04 will release on May 4th, 2021, is right?

I can't wait to pull the new release, and use it. [~smile]

kbarlee, 2021-04-21:
I'm not sure of the exact date, but I'm eagerly waiting too! Sometime very soon for the NSA srsUE, then October time for the srsGNB.

andrepuschmann, 2021-04-22:
We'll release srsRAN 21.04 on April 30th. Stay tuned ;-)

closed 2021-04-22 by andrepuschmann