srsran/srsRAN_4G #656: PHY to initialize

olawhk, 2021-04-30:
## Issue Description ##
I'm running srsUE and srsUE on the same docker container and using zmq for fake RF.  Upon starting eNB which is successfully connected to an external EPC (Open5GS on a different container), I wasn't able t

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2021-04-30:
Hey, please send EPC logs and pcap. It seems a signaling issue on the core side. Thanks

andrepuschmann, 2021-05-14:
Closing this due to user inactivity for two weeks.

closed 2021-05-14 by andrepuschmann

jestrada2006, 2023-02-01:
This is probably due to RRC Inactivity timer.  I had a similar issue and it resolved it.

sundepo, 2023-02-07:
> This is probably due to RRC Inactivity timer. I had a similar issue and it resolved it.

jestrada2006, how did you solve the problem?

jestrada2006, 2023-02-07:
> 

It is located in enb.conf, near the bottom, in the advanced settings section.