srsran/srsRAN_4G #658: eNB crahes for VoLTE call when E-RAB ID is 13 or above

herlesupreeth, 2021-05-02:
## Issue Description ##
eNB crashes with the following logs, when placing 11th call from the same two UEs

```
Opening USRP channels=1, args: recv_frame_size=9232,send_frame_size=9232,type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2660.0 Mhz, UL=2540.0 MHz for cc_idx=0 nof_prb=25

==== eNodeB started ===
Type <t> to view trace
RACH:  tti=4211, cc=0, preamble=36, offset=0, temp_crnti=0x46
User 0x46 connected
User 0x46 connected
RACH:  tti=8431, cc=0, preamble=17, offset=0, temp_crnti=0x47
User 0x47 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x47 connected
User 0x46 connected
User 0x46 connected
User 0x47 connected
User 0x47 connected
User 0x46 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
User 0x46 connected
User 0x47 connected
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
srsRAN crashed... backtrace saved in './srsRAN.backtrace.crash'...
---  exiting  ---
srsRAN crashed... backtrace saved in './srsRAN.backtrace.crash'...
---  exiting  ---
terminate called recursively
srsRAN crashed... backtrace saved in './srsRAN.backtrace.crash'...
---  exiting  ---

```

## Setup Details ##
Running srsRAN enB (master) + open5gs EPC (tag v2.2.7) + kamailio IMS all in docker setup - https://github.com/herlesupreeth/docker_open5gs

SDR - USRP B210
UEs with sysmocom USIM/ISIM - Xiaomi phone and iPhone X

## Expected Behavior ##
Should not crash

## Actual Behaviour ##
eNB crashes the moment i attempt 11th call and works good for all the previous 10 calls

## Steps to reproduce the problem ##

- Run the open5gs EPC + IMS
- Run the srsRAN eNB
- Attach the two UEs to IMS
- UEs successfully register with IMS
- Make ten calls between UEs
- Disconnect the call
- Place another (11th) call

## Additional Information ##
In order to allow E-RAB ID > 12 i.e. LCID > 10, I have done the following modification to srsRAN eNB. No other modification has been done

```
diff --git a/lib/include/srsran/common/common.h b/lib/include/srsran/common/common.h
index c0f5f5598..c341b24c3 100644
--- a/lib/include/srsran/common/common.h
+++ b/lib/include/srsran/common/common.h
@@ -40,8 +40,8 @@
 #define SRSRAN_UE_CATEGORY 4
 
 #define SRSRAN_N_SRB 3
-#define SRSRAN_N_DRB 11
-#define SRSRAN_N_RADIO_BEARERS 14
+#define SRSRAN_N_DRB 8
+#define SRSRAN_N_RADIO_BEARERS 11
 
 #define SRSRAN_N_MCH_LCIDS 32
```

I have attached the logs (INFO level), backtrace (taken with -DCMAKE_BUILD_TYPE=RelWithDebInfo), pcaps below. Please let me know if you need further details.
[log_bt_pcap.zip](https://github.com/srsran/srsRAN/files/6411021/log_bt_pcap.zip)

andrepuschmann, 2021-05-03:
Good catch. Thanks, Supreeth. We'll look into this.

davidrupprecht, 2021-05-03:
Dear @herlesupreeth, thanks for the catch. Can you install `libdw-dev` recompile it, trigger the bug again and send me the stack trace. Thanks a lot.

herlesupreeth, 2021-05-03:
@davidrupprecht sure will do that. Thank you

herlesupreeth, 2021-05-03:
Please find the log, pcaps and backtrace attached. Let me know if you need anything else
[log_bt_pcap2.zip](https://github.com/srsran/srsRAN/files/6414330/log_bt_pcap2.zip)

davidrupprecht, 2021-05-04:
Unfortunately the stacktrace did not use the latest stacktrace printer. Can you run `cmake .. -DCMAKE_BUILD_TYPE=RelWithDebInfo` and see if is building with `Building with backward-cpp support ` or the HAVE_BACKWARD flag is enabled. Thanks.

herlesupreeth, 2021-05-04:
My bad, i installed that library after compilation. Please find the trace after re-compiling. Following are the logs + backtrace. I also have the coredump but its around 120 MB, let me know if you need it.

[srsRAN.backtrace.zip](https://github.com/srsran/srsRAN/files/6420334/srsRAN.backtrace.zip)

```
Creating srsenb ... done
Attaching to srsenb
srsenb    | ---  Software Radio Systems LTE eNodeB  ---
srsenb    | 
srsenb    | Reading configuration file /etc/srsran/enb.conf...
srsenb    | 
srsenb    | Built in RelWithDebInfo mode using commit 3a0e7bb6a on branch crash_fix.
srsenb    | 
srsenb    | Opening 1 channels in RF device=default with args=clock=external,recv_frame_size=9232,send_frame_size=9232
srsenb    | [INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-release
srsenb    | [INFO] [LOGGING] Fastpath logging disabled at runtime.
srsenb    | Opening USRP channels=1, args: recv_frame_size=9232,send_frame_size=9232,type=b200,master_clock_rate=23.04e6
srsenb    | [INFO] [UHD RF] RF UHD Generic instance constructed
srsenb    | [INFO] [B200] Detected Device: B210
srsenb    | [INFO] [B200] Operating over USB 3.
srsenb    | [INFO] [B200] Initialize CODEC control...
srsenb    | [INFO] [B200] Initialize Radio control...
srsenb    | [INFO] [B200] Performing register loopback test... 
srsenb    | [INFO] [B200] Register loopback test passed
srsenb    | [INFO] [B200] Performing register loopback test... 
srsenb    | [INFO] [B200] Register loopback test passed
srsenb    | [INFO] [B200] Asking for clock rate 23.040000 MHz... 
srsenb    | [INFO] [B200] Actually got clock rate 23.040000 MHz.
srsenb    | Setting frequency: DL=2660.0 Mhz, UL=2540.0 MHz for cc_idx=0 nof_prb=25
srsenb    | 
srsenb    | ==== eNodeB started ===
srsenb    | Type <t> to view trace
srsenb    | RACH:  tti=4421, cc=0, preamble=40, offset=0, temp_crnti=0x46
srsenb    | User 0x46 connected
srsenb    | User 0x46 connected
srsenb    | RACH:  tti=9801, cc=0, preamble=30, offset=0, temp_crnti=0x47
srsenb    | User 0x47 connected
srsenb    | User 0x47 connected
srsenb    | User 0x46 connected
srsenb    | User 0x47 connected
srsenb    | User 0x46 connected
srsenb    | User 0x47 connected
srsenb    | RACH:  tti=5891, cc=0, preamble=42, offset=0, temp_crnti=0x48
srsenb    | RACH:  tti=5901, cc=0, preamble=37, offset=0, temp_crnti=0x49
srsenb    | Disconnecting rnti=0x47.
srsenb    | Disconnecting rnti=0x46.
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | User 0x48 connected
srsenb    | User 0x49 connected
srsenb    | terminate called after throwing an instance of 'std::bad_alloc'
srsenb    |   what():  std::bad_alloc
[... 153 lines cut ...]
srsenb    |          77: 		      "can't delete pointer to incomplete type");
srsenb    |       >  78: 	delete __ptr;
srsenb    |          79:       }
srsenb    |          80:     };
srsenb    | #7    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7fc2d07aedf4, in _Unwind_Resume
srsenb    | #6    Object "/lib/x86_64-linux-gnu/libgcc_s.so.1", at 0x7fc2d07ae572, in 
srsenb    | #5    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7fc2d0de64b7, in __gxx_personality_v0
srsenb    | #4    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7fc2d0de5b48, in 
srsenb    | #3    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7fc2d0de6ae5, in 
srsenb    | #2    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.25", at 0x7fc2d0de0956, in 
srsenb    | #1    Source "/build/glibc-S9d2JN/glibc-2.27/stdlib/abort.c", line 79, in abort [0x7fc2d03ed920]
srsenb    | #0    Source "../sysdeps/unix/sysv/linux/raise.c", line 51, in raise [0x7fc2d03ebfb7]
srsenb    | Aborted (Signal sent by tkill() 22 0)
srsenb    | #6    Source "/srsLTE/srsenb/src/stack/mac/sched_carrier.cc", line 457, in __stack_chk_fail [0x5640b3481bfb]
srsenb    |         454:   }
srsenb    |         455: 
srsenb    |         456:   // call DL scheduler metric to fill RB grid
srsenb    |       > 457:   sched_algo->sched_dl_users(*ue_db, tti_result);
srsenb    |         458: }
srsenb    |         459: 
srsenb    |         460: int sched::carrier_sched::alloc_ul_users(sf_sched* tti_sched)
srsenb    | Aborted (core dumped)
srsenb exited with code 134

```

frankist, 2021-05-04:
Hey @herlesupreeth. Can you try to set `MAX_LC = 13` in `lib/include/srsran/interfaces/sched_interface.h` and see if the issue vanishes? If so, we will provide a fix.

herlesupreeth, 2021-05-04:
Hey @frankist , unfortunately, that didnt solve the entire issue. Atleast now it doesn't crash but on the 7th call, UE doesnt receive the call at all, its like as though there is no GTP tunnel.

Please find the logs and pcap attached

In the logs i found this error
```

15:52:53.050892 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/rr_ded.cc][4140] Encoding failure.
15:52:53.050892 [ASN1   ] [E] [    0] [/srsLTE/lib/include/srsran/asn1/asn1_utils.h][1043] Encoding failure.
15:52:53.050893 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/rr_ded.cc][8477] Encoding failure.
15:52:53.050893 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][12341] Encoding failure.
15:52:53.050893 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][13991] Encoding failure.
15:52:53.050893 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][13894] Encoding failure.
15:52:53.050894 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][13841] Encoding failure.
15:52:53.050894 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][14981] Encoding failure.
15:52:53.050894 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][14575] Encoding failure.
15:52:53.050894 [ASN1   ] [E] [    0] [/srsLTE/lib/src/asn1/rrc/dl_dcch_msg.cc][15080] Encoding failure.
15:52:53.050895 [RRC    ] [E] Failed to encode DL-DCCH-Msg
15:52:53.050913 [S1AP   ] [I] Tx S1AP SDU, E_RABSetupResponse
```
[debug_1.zip](https://github.com/srsran/srsRAN/files/6422343/debug_1.zip)

herlesupreeth, 2021-05-06:
I have submitted a PR to fix this issue, it would be great if you could review it and let me you know your feedback.

I have tested its working as well. Please refer to the attached pcap for verify its working
[erab_fix.zip](https://github.com/srsran/srsRAN/files/6432359/erab_fix.zip)

Looking forward to your feedback

andrepuschmann, 2021-05-06:
Thanks for the PR Supreeth. Actually, we have already been working on a fix for it internally.
Besides using two Android phones and doing a VoLTE call between do you know of any other way to trigger the ERAB creation from the EPC side to trigger the bug?

herlesupreeth, 2021-05-06:
I know a way to trigger couple of eRAB creation from open5gs WebUI but not sure whether its able to trigger e-RAB until the e-RAB Id > 12.

Got to open5gs WebUI where you add SIM details

1. Click on "PCC Rules" under that internet APN, there set MBR Downlink/Uplink, GBR Downlink/Uplink to any desired value (for VoLTE call its enough to have value of 64 there)
2. Right under PCC rules there is a small "+" button upon pressing it you get to provide flow description (as shown in screenshot below) 
[image]
3. Here you can provide flow description as followins

Flow Direction* - Downlink
Description - permit out udp from <ANY_IP> <ANY_PORT> to any

Add another flow rule

Flow Direction* - Uplink
Description - permit out udp from <ANY_IP> <ANY_PORT> to any

This will trigger E-RAB Setup for one bearer

4. For triggering further bearer creations, i would suggest to adding further PCC rules with varying QCI and ARP and adding flow description as above for each PCC rule (dont forget to add DRB configuration at eNB for all the QCIs created in WebUI)

herlesupreeth, 2021-05-06:
Just another update

> For triggering further bearer creations, i would suggest to add further PCC rules with varying QCI and ARP and adding flow description as above for each PCC rule (dont forget to add DRB configuration at eNB for all the QCIs created in WebUI)

You can add PCC rules with same QCI and ARP but different flow-descriptions to trigger another e-RAB setup

kbarlee, 2021-05-10:
Hey Andre

I guess you could use https://github.com/open5gs/open5gs/blob/main/misc/db/open5gs-dbctl as an example, and modify to set the subscriber pcc rules via a script. Something along these lines?

```
#!/bin/bash

IMSI=001010000000001
MSISDN=0000000001
KI=465B5CE8B199B49FAA5F0A2EE238A6BC
OPC=E8ED289DEBA952E4283B54E88E6183CA

mongo --eval "db.subscribers.update( { \"imsi\" : \"$IMSI\" },
    { \$setOnInsert:
        {
            \"imsi\" : \"$IMSI\",
            \"subscribed_rau_tau_timer\" : NumberInt(12),
            \"network_access_mode\" : NumberInt(2),
            \"subscriber_status\" : NumberInt(0),
            \"access_restriction_data\" : NumberInt(32),
            \"slice\" :
            [{
                \"sst\" : NumberInt(1),
                \"default_indicator\" : true,
                \"_id\" : new ObjectId(),
                \"session\" :
                [{
                    \"name\" : \"internet\",
                    \"type\" : NumberInt(3),
                    \"_id\" : new ObjectId(),
                    \"pcc_rule\" : [{
                        \"qos\" : { 
                            \"index\" : NumberInt(1),
                            \"gbr\" : {
                                \"uplink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
                                \"downlink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) }
                            },
                            \"mbr\" : {
                                \"uplink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
                                \"downlink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) }
                            },
                            \"arp\" : {
                                \"priority_level\" : NumberInt(2),
                                \"pre_emption_capability\" : NumberInt(2),
                                \"pre_emption_vulnerability\" : NumberInt(2)
                            }
                        },
                        \"flow\" : [
                            {
                                \"direction\" : NumberInt(1),
                                \"description\" : \"permit out udp from any 1-65535 to 45.45.45.45\",
                            },
                            {
                                \"direction\" : NumberInt(2),
                                \"description\" : \"permit out udp from any 1-65535 to 45.45.45.45\",
                            },
                        ]
                    }],
                    \"ambr\" :
                    {
                        \"uplink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
                        \"downlink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
                    },
                    \"qos\" :
                    {
                        \"index\" : NumberInt(9),
                        \"arp\" :
                        {
                            \"priority_level\" : NumberInt(8),
                            \"pre_emption_capability\" : NumberInt(1),
                            \"pre_emption_vulnerability\" : NumberInt(1),
                        },
                    },
                }],
            }],
            \"ambr\" :
            {
                \"uplink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
                \"downlink\" : { \"value\": NumberInt(1), \"unit\" : NumberInt(3) },
            },
            \"security\" :
            {
                \"k\" : \"$KI\",
                \"amf\" : \"8000\",
                \"op\" : null,
                \"opc\" : \"$OPC\",
                \"msisdn\" : [ \"$MSISDN\" ],
                \"schema_version\" : NumberInt(1),
            },
            \"__v\" : 0
        },
    },
    upsert=true);" open5gs
exit 0
```

Cheers
Kenny

andrepuschmann, 2021-05-11:
Thanks @kbarlee for the advice. I am very new to open5gs and its currently giving me some serious headaches. I'll give your proposal a try as soon as we have a reliable CI setup with it running.

Meanwhile, we've collected a few commits that would be nice to test before submission. @herlesupreeth could you perhaps try https://github.com/srsran/srsRAN/tree/release_21_04_final with your setup that led to the eNB crash? It includes a couple of more fixes as well. Unfortunately, I couldn't replicate the issue here locally so it would be nice to make sure it's fixed for you.

Thanks
Andre

herlesupreeth, 2021-05-12:
@andrepuschmann I tried with the branch mentioned, but i have an issue. While I placed the first VoLTE call the call gets connected, then i disconnected the call and tried placing another call but this did not work at all (tried this scenario 4 times, same result). In the pcap you can see that for the 2nd call there is no "Activate deidicated EPS Bearer Context Accept"

Please find the pcap and enb.log attached
[log_pcap.zip](https://github.com/srsran/srsRAN/files/6464912/log_pcap.zip)

frankist, 2021-05-12:
Hey @herlesupreeth . Could you please increase the log `all_hex_limit` to 4096 and send new logs? This way, I can see the entirity of the packets being exchanged between the enb and UE.

frankist, 2021-05-12:
@herlesupreeth could you also rebase your fix on https://github.com/srsran/srsRAN/tree/release_21_04_final. This way it will be easier to include your commit changes.

herlesupreeth, 2021-05-12:
@frankist sure will do

herlesupreeth, 2021-05-12:
@frankist Please find the logs with increased hex limit attached (tested with release_21_04_final branch)

[Uploading enb_log_4096.zip…]()

andrepuschmann, 2021-05-12:
@herlesupreeth We've consolidated a few patches and would like to ask you to test your scenario again. It is included in this "new" branch https://github.com/srsran/srsRAN/tree/release_21_04_erab_fixes Thanks again for your help. Much appreciated!

herlesupreeth, 2021-05-13:
@andrepuschmann Tried the suggested branch. Facing the same issue where I place the first VoLTE call the call gets connected, then i disconnected the call and tried placing another call but this did not work at all. In the pcap you can see that for the 2nd call there is no "Activate dedicated EPS Bearer Context Accept" again
[log_pcap.zip](https://github.com/srsran/srsRAN/files/6471675/log_pcap.zip)

Btw, I had look into this commit https://github.com/srsran/srsRAN/commit/d81788e077a97bf1aedbe48eb8989766e50ec216, i think we cannot have fixed DRB and LCID mappings since the maximum number of DRBs in LTE can go upto 11 where as the LCIDs that can be used for DRBs are 3..10. Please correct me if I am wrong

andrepuschmann, 2021-05-13:
@herlesupreeth Actually the NAS message for the dedicated bearer activation wasn't sent. We'll provide another patch shortly. Thanks again

andrepuschmann, 2021-05-13:
Here https://github.com/srsran/srsRAN/tree/fix_erabid_to_drb_conversions is the new branch (just differs in one commit) for you to try. Let me know how it goes. If it doesn't work, please send also the eNB MAC PCAP along. Thanks

herlesupreeth, 2021-05-13:
@andrepuschmann Thank you. I tested the branch and it works falwlessly

andrepuschmann, 2021-05-14:
Excellent. Thanks for your help and support again @herlesupreeth.

Closing the issue now as the fixes will be included in master soon.

closed 2021-05-14 by andrepuschmann