srsran/srsRAN_4G #662: eNB crashes in srsenb::rrc::ue::parse_ul_dcch

rriggio, 2021-05-08:
## Issue Description ##
The eNB compiled from the latest master crashes during attach or detach. The behavior is random. Sometimes the UE (a nexus 5) is attaching without problems other times the eNB crashes.

## Setup Details ##
I am using ubuntu 20.04.02, an ettus B210mini and a Nexus 5 as UE.

## Expected Behavior ##
The eNB should not crash

## Actual Behaviour ##
The eNB crashes

## Steps to reproduce the problem ##
Compile from master edit the enb.conf to point to the core network and try to attach a UE.

## Additional Information ##
This is the backtraces of the crash:

`#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007ffff7673859 in __GI_abort () at abort.c:79
#2  0x00005555556780da in srsran::optional<unsigned int>::value() [clone .part.0] ()
#3  0x000055555567e7a5 in srsenb::s1ap::ue::send_ulnastransport(std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) ()
#4  0x000055555567e87a in srsenb::s1ap::write_pdu(unsigned short, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) ()
#5  0x000055555570e9d9 in srsenb::rrc::ue::parse_ul_dcch(unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) ()
#6  0x00005555556f4a7a in srsenb::rrc::parse_ul_dcch(unsigned short, unsigned int, std::unique_ptr<srsran::byte_buffer_t, std::default_delete<srsran::byte_buffer_t> >) ()
#7  0x00005555556f6cb4 in srsenb::rrc::tti_clock() ()
#8  0x000055555564f3e4 in srsran::task_details::smallbuffer_table_t<srsenb::enb_stack_lte::tti_clock()::{lambda()#1}, void>::call(void*) const ()
#9  0x000055555564eb4a in srsenb::enb_stack_lte::run_thread() ()
#10 0x00005555556066fd in srsran::thread::thread_function_entry(void*) ()
#11 0x00007ffff7f97609 in start_thread (arg=<optimized out>) at pthread_create.c:477
#12 0x00007ffff7770293 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95`

And this is the crash log:

`--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:42:25' ---
	./srsenb(+0x296a5f) [0x55c57886ca5f]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f61c683a210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f61c683a18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f61c6819859]
	./srsenb(+0x1240da) [0x55c5786fa0da]
	./srsenb(+0x12a7a5) [0x55c5787007a5]
	./srsenb(+0x12a87a) [0x55c57870087a]
	./srsenb(+0x1ba9d9) [0x55c5787909d9]
	./srsenb(+0x1a0a7a) [0x55c578776a7a]
	./srsenb(+0x1a2cb4) [0x55c578778cb4]
	./srsenb(+0xfb3e4) [0x55c5786d13e4]
	./srsenb(+0xfab4a) [0x55c5786d0b4a]
	./srsenb(+0xb26fd) [0x55c5786886fd]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f61c713d609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f61c6916293]

--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:42:26' ---
	./srsenb(+0x296a5f) [0x55c57886ca5f]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f61c683a210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f61c683a18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f61c6819859]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0x9e951) [0x7f61c6bee951]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa47c) [0x7f61c6bfa47c]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa4e7) [0x7f61c6bfa4e7]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x1b338) [0x7f61c6f5e338]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x3dc02) [0x7f61c6f80c02]
	/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0xce) [0x7f61c683e15e]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x122b7) [0x7f61c6f552b7]

--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:42:26' ---
--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:42:53' ---
	./srsenb(+0x296a5f) [0x55acd3411a5f]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f0bb930a210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f0bb930a18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f0bb92e9859]
[... 136 lines cut ...]
	./srsenb(+0x12a87a) [0x55734ad2687a]
	./srsenb(+0x1ba9d9) [0x55734adb69d9]
	./srsenb(+0x1a0a7a) [0x55734ad9ca7a]
	./srsenb(+0x1a2cb4) [0x55734ad9ecb4]
	./srsenb(+0xfb3e4) [0x55734acf73e4]
	./srsenb(+0xfab4a) [0x55734acf6b4a]
	./srsenb(+0xb26fd) [0x55734acae6fd]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f4c55955609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f4c5512e293]

--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:44:48' ---
	./srsenb(+0x296a5f) [0x55734ae92a5f]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f4c55052210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f4c5505218b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f4c55031859]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0x9e951) [0x7f4c55406951]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa47c) [0x7f4c5541247c]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa4e7) [0x7f4c554124e7]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x1b338) [0x7f4c55776338]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x3dc02) [0x7f4c55798c02]
	/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0xce) [0x7f4c5505615e]
	/home/empower/TMP/srsRAN/build/lib/src/phy/rf/libsrsran_rf.so.0(+0x122b7) [0x7f4c5576d2b7]

--- command='./srsenb enb.conf' version=21.04.0 signal=6 date='08/05/2021 07:44:48' ---
`

yagoda, 2021-05-09:
Hey,
Can you send an email to srslte-users@lists.softwareradiosystems.com with the enb.log, epc.log and the console output?

Additionally, can you provide information about the SIM card used?

Regards,
Justin

rriggio, 2021-05-10:
I have added the brand and model of SIM card I am using. I am now having some difficulties in replicating the crash however if it can help, the crash was due to a failed assert at this line in s1ap::ue::send_ulnastransport:

`container.mme_ue_s1ap_id.value                        = ctxt.mme_ue_s1ap_id.value();`

frankist, 2021-05-10:
Thanks @rriggio . It seems that UL NAS Transport is being sent to the MME for a UE whose S1 context is not yet fully created. We will need enb logs to further debug this issue.

frankist, 2021-05-12:
Thanks @rriggio. The branch https://github.com/srsran/srsRAN/tree/release_21_04_final should be able to resolve the crash. Can you confirm?

I noticed based on the enb.log that the EPC is taking a very long time to reply, or even not replying at all, to the InitialUEMessage. Are you using the srsEPC? If so, could you also send the epc.log file?

andrepuschmann, 2021-05-14:
Closing the issue. This should be fixed in master.

closed 2021-05-14 by andrepuschmann