srsran/srsRAN_4G #667: USRP error

shuimoshusheng, 2021-05-14:
when i run sudo srsenb, the log output like this:
[image]

my device is usrp x310, system is ubuntu 18.04.3,linux kernal is  5.0.0-23-generic

andrepuschmann, 2021-05-14:
what version, etc.? provide full details and use the provided template. Don't post pictures but rather screen logs.

shuimoshusheng, 2021-05-14:
> what version, etc.? provide full details and use the provided template. Don't post pictures but rather screen logs.

Run epc and enb device system is ubuntu 18.04.3 ,linux kernal is 5.0.0-23-generic, RF device is USRP X310, and run ue device system is ubuntu 18.04.3,linux kernal is 5.0.0-23-generic, RF device is USRP b210.
==================================================================================
srsepc log is:
==================================================================================
root@ueransim:~# sudo srsepc

Built in Release mode using commit d0c3a4232 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /root/.config/srsran/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01
S1 Setup Request - TAC 7, B-PLMN 0xf110
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response

==================================================================================
srsenb log is:
==================================================================================
root@ueransim:~# sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor

Built in Release mode using commit d0c3a4232 on branch master.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-64-g91a21949
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1298 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1316 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50

==== eNodeB started ===
Type <t> to view trace
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 10.6858 >= 10.6631. Starting new burst...
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #1
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #2
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[... 40 lines cut ...]
Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-64-g91a21949
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Waiting PHY to initialize ... done!
Attaching UE...
.............
...........................................................................................................................................................................................................................................
.......................................................................................................................................................................
................................................................................................................................
....................................................................................................................................................................................................................................
......................................................................
.........................................................................................................................................................................
........................................................................................................................................................................................................................................................................................................................................................................................................................................

xavierarteaga, 2021-05-14:
hey, can you try tweaking the `rf` parameters:
```
# Force base-band sampling rate:
srate=23.04e6

# Force sampling rate at initiation and ethernet frame sizes
device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
```

shuimoshusheng, 2021-05-14:
> hey, can you try tweaking the `rf` parameters:
> 
> ```
> # Force base-band sampling rate:
> srate=23.04e6
> 
> # Force sampling rate at initiation and ethernet frame sizes
> device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
> ```

What command to use to modify the sample rate？

shuimoshusheng, 2021-05-14:
> 嘿，您可以尝试调整`rf`参数：
> 
> ```
> # Force base-band sampling rate:
> srate=23.04e6
> 
> # Force sampling rate at initiation and ethernet frame sizes
> device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
> ```

What command to use to modify the sample rate？

xavierarteaga, 2021-05-14:
You can try appending to `srsenb`:
```
$ sudo srsenb ... --rf.srate=23.04e6 --rf.device_args=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
```

Substitute `...` for your current arguments.

shuimoshusheng, 2021-05-17:
> .

> You can try appending to `srsenb`:
> 
> ```
> $ sudo srsenb ... --rf.srate=23.04e6 --rf.device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
> ```
> 
> Substitute `...` for your current arguments.
===================================================================================
... is the enb configuration file? 

run this command ，log output is：
===================================================================================
root@ueransim:~# sudo srsenb ... --rf.srate=23.04e6 --rf.device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file ......
Failed to read configuration file ... - exiting

shuimoshusheng, 2021-05-17:
when I run enb and epc on usrp b 210, and run ue on usrp x310. The log output is as follows：
========
root@free5gc-compose:~# sudo srsepc

Built in Release mode using commit d0c3a4232 on branch master.

---  Software Radio Systems EPC  ---

Reading configuration file /root/.config/srsran/epc.conf...
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
SPGW GTP-U Initialized.
SPGW S11 Initialized.
SP-GW Initialized.
Received S1 Setup Request.
S1 Setup Request - eNB Name: srsenb01, eNB id: 0x19b
S1 Setup Request - MCC:001, MNC:01
S1 Setup Request - TAC 7, B-PLMN 0xf110
S1 Setup Request - Paging DRX v128
Sending S1 Setup Response
SCTP Association Shutdown. Association: 4
Deleting eNB context. eNB Id: 0x19b
Releasing UEs context
No UEs to be released

========
root@free5gc-compose:~# sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor

Built in Release mode using commit d0c3a4232 on branch master.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-64-g91a21949
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50

==== eNodeB started ===
Type <t> to view trace
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 18.8096 >= 8.63634. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 48.0712 >= 48.0711. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 62.9339 >= 51.2849. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 75.8751 >= 70.0312. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 98.7228 >= 93.9495. Starting new burst...

========
root@ueransim:~# sudo srsue
Reading configuration file /root/.config/srsran/ue.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor

Built in Release mode using commit d0c3a4232 on branch master.

Opening 1 channels in RF device=default with args=default
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-64-g91a21949
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1305 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1315 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Waiting PHY to initialize ... done!
Attaching UE...
....................................................................................................................................
..............................................................................................................................................................................................................
...............................................................................................................................................................................................................................
...................................................................................................................................................................................................................................................................................................................................................................................................

shuimoshusheng, 2021-05-18:
root@ubuntu:~# sudo srsenb --rf.srate=23.04e6 --rf.device_args=type=x300,sampling_rate=23.04e6,send_frame_size=1472,recv_frame_size=1472
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor

Built in Release mode using commit 1c6dd8c4a on branch master.

Opening 1 channels in RF device=default with args=type=x300,sampling_rate=23.04e6,send_frame_size=1472,recv_frame_size=1472
Opening USRP channels=1, args: type=x300,send_frame_size=1472,recv_frame_size=1472,master_clock_rate=184.32e6
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 1472 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
[INFO] [0/DmaFIFO_0] Initializing block control (NOC ID: 0xF1F0D00000000000)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1307 MB/s)
[INFO] [0/DmaFIFO_0] BIST passed (Throughput: 1306 MB/s)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000001)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000000)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000000)
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50

==== eNodeB started ===
Type <t> to view trace
Error - buffer pool is empty
Error - buffer pool is empty
[ERROR] [X300] 192.168.10.2: x300 fw communication failure #1
EnvironmentError: IOError: x300 fw poke32 - reply timed out
[ERROR] [UHD] An unexpected exception was caught in a task loop.The task loop will now exit, things may not work.AssertionError: reply.sequence == request.sequence
  in virtual void x300_ctrl_iface_enet::__poke32(uhd::wb_iface::wb_addr_type, uint32_t)
  at /build/uhd-j2Haqj/uhd-3.15.0.0-1/host/lib/usrp/x300/x300_fw_ctrl.cpp:134

/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 8.62936 >= 8.49617. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 13.6623 >= 10.7067. Starting new burst...
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 18.1433 >= 15.6734. Starting new burst...
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 27.7809 >= 27.7809. Starting new burst...
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 31.3839 >= 30.3252. Starting new burst...
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.309: UHD unhandled event code 32
/root/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc.1353: Error timed out while receiving samples from UHD.

vikramkapoor, 2021-07-01:
I get the following error running on UHD210
 sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /root/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=UHD with args=default
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.0-release
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B200
[INFO] [B200] Loading FPGA image: /usr/share/uhd/images/usrp_b200_fpga.bin...
[INFO] [B200] Operating over USB 2.
[INFO] [B200] Detecting internal GPSDO.... 
[INFO] [GPS] No GPSDO found
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=25

==== eNodeB started ===
Type <t> to view trace
[INFO] [UHD RF] Tx while waiting for EOB, timed out... 11.1173 >= 0. Starting new burst...

zhujiu, 2021-09-27:
> what version, etc.? provide full details and use the provided template. Don't post pictures but rather screen logs.

Hi, @andrepuschmann:

First, I met the same problem with @shuimoshusheng ,I think the command should be like this.
`sudo srsenb ... --rf.srate=23.04e6 --rf.device_args=type=x300,sampling_rate=23.04e6,send_frame_size=8000,recv_frame_size=8000
`
But I still got some thing like this --- [INFO] [UHD RF] Tx while waiting for EOB, timed out... 47.2011 >= 43.9856. Starting new burst...

So I rebuild srsSRN in my systeam with the latest version--https://docs.srsran.com/en/latest/general/source/1_installation.html
Here is my systeam.
(base) dragon@dragon:~$ uname -a
Linux dragon 5.4.0-52-generic #57~18.04.1-Ubuntu SMP Thu Oct 15 14:04:49 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

I fixed the problem timed out... 47.2011 >= 43.9856, but I met a new one.

```
(base) dragon@dragon:/etc/srslte$ sudo srsenb
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/dragon/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
-- Detected Device: B200mini
-- Operating over USB 2.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 23.040000 MHz... 
-- Actually got clock rate 23.040000 MHz.
-- Performing timer loopback test... pass
Setting frequency: DL=2685.0 Mhz, UL=2565.0 MHz for cc_idx=0 nof_prb=50
LLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLLLLLLLLLLLLLLLLLLLOOOOOUULLLLLLLLLLLLLLLLLOOOOLLLLLLLLLLLLLLLLLLOLLLLLOOOOUULLLLLLLLLLLLLLLLLOLLLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOOOOLLLLLLLLLLLLLLLLLLLLOLLLLOOOOULLLLLLLLLLLLLLLLOLOOOLLLLLLLLLLLLLLLLLLLLOLLLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOOOconnect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
O
==== eNodeB started ===
Type <t> to view trace
LLLLLLLLLLLLLLLLLLOLLLLLLOOOOULLLLLLLLLLLLLLLLLLLLLLLOOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOOOOLLLLLLLLLLLLLLLLLLLOLLLLLOOOOULLLLLLLLLLLLLLLLLOLLLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLOOOLLLLLLLLLLLLLLLLLLLOLLLLLOOOOULLLLLLLLLLLLLLLLLLLLLOLLOOOLOLLLLLOOOOLULLLLLLLLLLLLLLLLLLLOLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLOOOLLLLLLLLLLLLLLLLOLLOOOOUULLLLLLLLLLLLLLLLLOLLOOOLLLLLLLOLLLLLOOOOULLLLLLLLLLLLLLLLLOLLLLLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOOOOLLLLLLLLLLLLLLLLLLOLLLLLLOOOOULLLLLLLLLLLLLLLLLLLOLLLLOOOLLOLLLLOOOOUULLLLLLLLLLLLLLLLLLLOOOOLLLLLLOOOOOULLLLLLLLLLLLLLLLLLLLLOLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLOOOLLLLLLLLLLLLLLLLLLOOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLOOOLLLLLLLLLLLLLLOLLLLOOOOUUULLLLLLLLLLLLLLLLLOOOOLLLLLLLLLLLLOOOOOUULLLLLLLLLLLLLLLOLLLLOOOLLLLOLLOOOOULLLLLLLLLLLLLLLLLOLLLLLLOOOOUULLLLLLLLLLLLLLLOLLOOOOLLLLLOOOOULLLLLLLLLLLLLLLLLLLLLOOOOOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLOOOOLLLLLLLLLLLLLLLLLLLLLLLLLL

```

I am confused what is the meanning of so many LLLLLOLLLLL?

thank you.

andrepuschmann, 2021-09-27:
```
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
-- Detected Device: B200mini
-- Operating over USB 2.
```
USB2 isn't enough to carry the rates. You need USB3

zhujiu, 2021-09-27:
> ```
> Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
> -- Detected Device: B200mini
> -- Operating over USB 2.
> ```
> 
> USB2 isn't enough to carry the rates. You need USB3

Thank you, @andrepuschmann  I noticed it a few miniuts ago and changed it to USB3, bit I still got this, LLLLLLOLLLLLLLL.

`(base) dragon@dragon:~$ sudo srsenb
[sudo] password for dragon: 
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /home/dragon/.config/srsran/enb.conf...
WARNING: Could not verify cpu0 scaling governor
WARNING: Could not verify cpu1 scaling governor
WARNING: Could not verify cpu2 scaling governor
WARNING: Could not verify cpu3 scaling governor

Built in Release mode using commit 89f16eed2 on branch master.

Opening 1 channels in RF device=default with args=default
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
-- Detected Device: B200mini
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 23.040000 MHz... 
-- Actually got clock rate 23.040000 MHz.
-- Performing timer loopback test... pass
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
ULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLLLL
==== eNodeB started ===
Type <t> to view trace
ULLLLLLLLLLLLLLLLULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOUUUUUULLUUULLLLLLLLLLLLLLLLLLLLLLLLUULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLUULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLUUUULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLULLLLLLLLLLLLLLLLLLLLLLULLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLOLLLLLLq
Stopping ..
UO---  exiting  ---
`

zhujiu, 2021-09-29:
> ```
> Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
> -- Detected Device: B200mini
> -- Operating over USB 2.
> ```
> 
> USB2 isn't enough to carry the rates. You need USB3

Hi, @andrepuschmann , I already changed to usb3, but I got LLLLLLLLOLLLL too.  Do you have any idea about this?

shuimoshusheng, 2021-10-14:
@sadiqzq  set cpu to performance and check if the antenna is plugged in. then retry run srsenb!

sadiqzq, 2021-10-20:
> @sadiqzq <https://github.com/sadiqzq>， you device is usrp N210 or
> B210？the firmware is matching

The usrp is NI 2920, with updated firmware N210. Yes it is matching with
the compatible version which is mentioned on Ettus website.

andrepuschmann, 2021-11-26:
Issue "solved". Closing.

closed 2021-11-26 by andrepuschmann