srsran/srsRAN_4G #668: Uplink PRB allocation

wangguixian1999, 2021-05-14:
## Issue Description ##
When there is only one UE under the srsenb, whatever uplink throughtput it is, the srsenb will give all the PRBs it can allocate to the UE (The UE only has uplink flow). Because the limit of hadware, more than one UE was not been tested.
For example, assume the maxmum uplink bandwith is 10Mbps, the uplink flow is 5Mbps on average and the PRB can be used for uplink is 40. The srsenb will give 40 prbs to the UE.

## Setup Details ##
Use release_20_04.
Set "n_prb = 50" in the enb.cof.

## Expected Behavior ##
The UE will get 20 prbs if the numbers of prb and uplink bandwith is linear.

6
## Additional Information ##
We foundthe source code of this part.
```
1 uint32_t get_required_prb_ul(const sched_ue_cell& cell, uint32_t req_bytes)
2 {
3  if (req_bytes == 0) {
4    return 0;
5   }
6  auto compute_tbs_approx = [&cell](uint32_t nof_prb) {
7    const uint32_t N_srs  = 0;
8    uint32_t       nof_re = (2 * (SRSRAN_CP_NSYMB(cell.cell_cfg->cfg.cell.cp) - 1) - N_srs) * nof_prb * SRSRAN_NRE;
9    return cqi_to_tbs_ul(cell, nof_prb, nof_re, -1).tbs_bytes;
10  };
11
12  // find nof prbs that lead to a tbs just above req_bytes
13  int                                      target_tbs = static_cast<int>(req_bytes) + 4;
14  uint32_t                                 max_prbs   = std::min(cell.tpc_fsm.max_ul_prbs(), cell.cell_cfg->nof_prb());
15  std::tuple<uint32_t, int, uint32_t, int> ret =
16    false_position_method(1U, max_prbs, target_tbs, compute_tbs_approx, [](int y) { return y == SRSRAN_ERROR; });
17  uint32_t req_prbs = std::get<2>(ret);
18  while (!srsran_dft_precoding_valid_prb(req_prbs) && req_prbs < cell.cell_cfg->nof_prb()) {
19   req_prbs++;
20  }
21  return req_prbs;
22}
```
The 18th and 19th may the reason why the srsenb allocation all prbs.

wangguixian1999, 2021-06-09:
> Hey, do you have an enb.log with MAC in Debug that you can share? Also, can you test with release 21.04?

I tested release 21.04, and the issues also existed.

wangguixian1999, 2021-06-09:
The log of MAC in debug is following. The full debug log is to big, I only copy part of it.
[enb-mac-debug-part.txt](https://github.com/srsran/srsRAN/files/6622364/enb-mac-debug-part.txt)

wangguixian1999, 2021-06-09:
Besides, we find the definition of _**bool srsran_dft_precoding_valid_prb(uint32_t nof_prb)**_ . It's in _**srsRAN / lib / src / phy / dft / dft_precoding.c**_.
[image]
It may because **_valid_prb[101]_** is not continues, srsenb will find the closest number of prb which is larger than the required prb.

If the above is currect, is this a limit of LTE protocol or a limit of srsenb?

frankist, 2021-06-09:
Hey @wangguixian1999, I am not sure I understand the issue. That valid_prb table you showed is defined for the LTE protocol.

The scheduler will allocate for a UE the number PRBs necessary to fit all of its pending UL data. The eNB knows the pending UL data through the buffer status report that the UE sends. Therefore, I don't see anything wrong with what you just described.

wangguixian1999, 2021-06-10:
Thanks, I misunderstand the LTE protocol. 
I will close the issues.

closed 2021-06-10 by wangguixian1999