srsran/srsRAN_4G #670: EPC is not accepting the attach accept from eNB

modyngs, 2021-05-17:
## Issue Description ##
Hi,
I've used SRSRAN in the case that both EPC+RAN were in the same Machine or different machine but in the same LAN network. there was all well.

So then i decided to Use Cloud for EPC and Local PC for RAN! the problem appeared! 

_**MME not receiving the Attach complete even though its seen in the trace taken on EPC host**_

Attach complete and Activate default bearer accept sent from eNB to EPC
but again EPC is sending Downlink NAS transport with Attach accept and Activate default bearer request

## Setup Details ##
Setup is as Follows:
COTS UE is Samsung A42 and SDR is USRP B210
eNB is on local pc [ local ip=192.168.1.5, and DHCP public ip from ISP=5.64.x.x]
RAN PC is core i7 6700, 16GB RAM, 128gb SSD with SrsRAN {master branch}and UHD_4.0 {master branch}
Core (SRSEPC) is on openstack VM [ local ip= 192.168.1.23, Floating ip=46.102.143.85]
with 4 vCPU,4G RAM, 20GB storage

```
   UE            =====>>>>>>     SRSLTE    ===>>>       ISP DHCP IP add.       ===>>>     OPENSTACK     ==>>>          VM
             =====>>>>>>         192.168.1.5  ===>>>     5.64.x.x            ===>>>      46.102.143.85     ==>>>   192.168.1.23
```
Please find the Attached files of CORE& RAN bellow
[srsEPC.zip](https://github.com/srsran/srsRAN/files/6495285/srsEPC.zip)

## Expected Behavior ##
I expect to have internet connection

## Actual Behaviour ##
While i dont have interenet access! even in UE i can see 4G  sign in just one Direction! but do can not ping or access any webpages.

## Steps to reproduce the problem ##
in openstack VM : ( for EPC )
sudo apt-get update && sudo apt-get -y dist-upgrade  && sudo reboot

sudo apt-get -y install build-essential cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev
git clone https://github.com/srsRAN/srsRAN.git
cd srsRAN && mkdir build && cd build 
cmake ../ 
make
make test 
sudo make install
sudo srsran_install_configs.sh user
sudo nano ~/.config/srsran/epc.conf
and change it as there is in EPC zip file (MME AND SGWU ip add. )

modyngs, 2021-05-18:
Also the same test as above with different openstack VM add.
```
UE   ==>>     SRSRAN    ===>>>       ISP DHCP IP add.       ===>>>     OPENSTACK     ==>>>          VM( SRSEPC)
     ==>>  192.168.1.5       ===>>>     5.64.x.x            ===>>>  46.102.143.85     ==>>>        10.10.10.78
```
Here is the Logs_pcaps_confs of EPC
[EPC.zip](https://github.com/srsran/srsRAN/files/6498519/EPC.zip)

modyngs, 2021-05-18:
SCTP peer address unreachable. Association: 0
this errors can be seen in LOGs

modyngs, 2021-05-18:
> ### SCTP peer address unreachable. Association: 0

This Problem Solved Bu changing MTU in both sides!

But still the problem of Not accessing to internet remained

in EPC:
[image]

modyngs, 2021-05-18:
> ### SCTP peer address unreachable. Association: 0

This Problem Solved By changing MTU in both sides!

But still the problem of Not accessing to internet remained

in EPC:
[image]

in RAN:
[image]
[image]

Here is also new Logs_PCAP_configs:
[RAN.zip](https://github.com/srsran/srsRAN/files/6499092/RAN.zip)

modyngs, 2021-05-25:
Problem Solve by setting MTU size

closed 2021-05-25 by modyngs

90r1ll4, 2024-02-29:
Hi @modyngs the UE can still not reach the internet, the documentation is not helping me, i tried the above solution, not working for me, and also can you tell me the MTU size you set to?